espressif/vscode-esp-idf-extension #840: Extension installes wrong binaries

beta-tester, 2022-11-12:
### OS

Linux

### Operating System version

Linux dev 5.15.76-v7+ #1597 SMP Fri Nov 4 12:13:17 GMT 2022 armv7l GNU/Linux

### Visual Studio Code version

1.73.1-1667960052

### ESP-IDF version

4.4 and 5.0

### Python version

2.9.x

### Doctor command output

n/a

### Extension

1.5.1

### Description

i followed https://github.com/espressif/vscode-esp-idf-extension/blob/master/docs/tutorial/install.md

- my Raspberry Pi 2B is with RaspOS Bullseye
- i installed VS Core 1.73.1 with `sudo apt install code`
- i installed all dependencies `sudo apt-get install git wget flex bison gperf python3 python3-venv cmake ninja-build ccache libffi-dev libssl-dev dfu-util libusb-1.0-0`
- i installed the Espressif IDF Extension 1.5.1.
- i run the ESP_IDF::Setup.

but when i run the ESP-IDF::Setup, it installes some binaries for `amd64`, `x86_64`, `linux64` on my Raspberry Pi that only can handle `armhf` binaries.

it happens in the normal and advanced setup mode.
i tried the 4.4 and the 5.0 release of esp-idf.

i also tried to clone the esp-idf manually and installed the tools manually to get the correct armhf binaries, but when i run the ESP-IDF::Setup in advanced mode find existing ones and selected all the existing esp-idf `/home/admin/esp/esp-idf` and tools `/home/admin/.espressif` pathes from my manual clone and installation, the setup still begin to download the tools in the `amd64` version and overrides the existing correct `armhf` ones or at least ignores them.
the result is the same i can not build/flash/monitor any example in VS Core + Espressif IDF Extension.

for example, `cmake` i installed via `apt`, but VS Core tries to execute the `cmake` installed from the esp-idf in the tools path that is not `armfh` binaries -  `Downloading CMake build system; cmake-3.24.0-linux-x86_64.tar.gz has (45941) KB`

[image]

### Debug Message

```plain
n/a
```

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

beta-tester, 2022-11-12:
here the logs i found:

the error: like in the screenshot:
[ESP-IDF-example.log](https://github.com/espressif/vscode-esp-idf-extension/files/9995853/ESP-IDF-example.log)

and the log from installation (i removes all the lines with 0%..100% progresses):
[ESP-IDF-install.log](https://github.com/espressif/vscode-esp-idf-extension/files/9995854/ESP-IDF-install.log)

beta-tester, 2022-11-12:
```bash
$ uname -a | tee -a version.txt
Linux dev 5.15.76-v7+ #1597 SMP Fri Nov 4 12:13:17 GMT 2022 armv7l GNU/Linux

$ cat /etc/os-release
PRETTY_NAME="Raspbian GNU/Linux 11 (bullseye)"
NAME="Raspbian GNU/Linux"
VERSION_ID="11"
VERSION="11 (bullseye)"
VERSION_CODENAME=bullseye
ID=raspbian
ID_LIKE=debian
HOME_URL="http://www.raspbian.org/"
SUPPORT_URL="http://www.raspbian.org/RaspbianForums"
BUG_REPORT_URL="http://www.raspbian.org/RaspbianBugs"
```

beta-tester, 2022-11-13:
a little bit more progress.
i tried again manually. clone  esp-idf repository, install esp-idf-tools, exported the environment variables and started VS Code from that environment.
i keept all the already installed esp-idf-tools in ESP-IDF Extension; ADVANCED setup untouched.

now it starts building the `hello_wold` example - but it ends up  with errors because of missing module `idf_component_manager`.
i can sucessfully build the `hello_world` project from the same export-environment outside of VS Code from the console by using `idf.py build`.
or even worse, i can build the `hello_world` project inside VS Code, when i open a  terminal and type in `idf.py build`.

- i deleted all the mess from earlier tries:
  ```bash
  rm -rf ~/.espressif/
  rm -rf ~/esp/esp-idf/
  rm -rf ~/.vscode/
  rm -rf ~/.config/Code/
  ```
- i cloend v4.4.3:
  ```bash
  git clone --branch=v4.4.3 --recurse-submodules https://github.com/espressif/esp-idf ~/esp/esp-idf/
  ```
- i installed the esp-idf tools:
  ```bash
  cd ~/esp/esp-idf/
  . ./install.sh esp32
  ```
- i run code from console with exports:
  ```bash
  cd ~/esp/esp-idf/
  . ./export.sh
  code
  ```
- by starting VS Code from that export-environment:
  - i installed the `Espressif IDF Extension`.
    [image]
  - in `ESP-IDF: Configure ESP-IDF extension`, i choosed `ADVANCED` setup.
    [image]
    - selected "Find ESP-IDF in your system", ...
      [image]
    - selected "Use existing ESP-IDF Tools"
      [image]
    - copied `60-openocd.rules` in to place
      [image]
  - i opend `ESP-IDF: Show Examples Projects` and created the hello_world project to my projects folder
    [image]
  - closed VS Code
- by starting VS Code from that export-environment the second time:
  - i enabled to trust all files and parent folders of my projects folder
    [image]
  - and tried to build the `hello_world` project - it ends up with errors:
    [image]
  - but when i open a terminal in VS Code and run idf.py build, it builds with success.
    [image]
    [image]

here the log text:
```log
 *  Executing task: cmake -G Ninja -DPYTHON_DEPS_CHECKED=1 -DESP_PLATFORM=1 -B /media/pi/data/projects/esp32/hello_world/build -S /media/pi/data/projects/esp32/hello_world 

-- Found Git: /usr/bin/git (found version "2.30.2") 
-- The C compiler identification is GNU 8.4.0
-- The CXX compiler identification is GNU 8.4.0
-- The ASM compiler identification is GNU
-- Found assembler: /home/pi/.espressif/tools/xtensa-esp32-elf/esp-2021r2-patch5-8.4.0/xtensa-esp32-elf/bin/xtensa-esp32-elf-gcc
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Check for working C compiler: /home/pi/.espressif/tools/xtensa-esp32-elf/esp-2021r2-patch5-8.4.0/xtensa-esp32-elf/bin/xtensa-esp32-elf-gcc - skipped
-- Detecting C compile features
-- Detecting C compile features - done
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Check for working CXX compiler: /home/pi/.espressif/tools/xtensa-esp32-elf/esp-2021r2-patch5-8.4.0/xtensa-esp32-elf/bin/xtensa-esp32-elf-g++ - skipped
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Project is not inside a git repository, or git repository has no commits; will not use 'git describe' to determine PROJECT_VER.
-- Building ESP-IDF components for target esp32
CMake Error at /home/pi/esp/esp-idf/tools/cmake/build.cmake:452 (message):
  /usr/bin/python: Error while finding module specification for
  'idf_component_manager.prepare_components' (ModuleNotFoundError: No module
  named 'idf_component_manager')

Call Stack (most recent call first):
  /home/pi/esp/esp-idf/tools/cmake/project.cmake:384 (idf_build_process)
  CMakeLists.txt:6 (project)

-- Configuring incomplete, errors occurred!
See also "/media/pi/data/projects/esp32/hello_world/build/CMakeFiles/CMakeOutput.log".

 *  The terminal process "/usr/bin/bash '-c', 'cmake -G Ninja -DPYTHON_DEPS_CHECKED=1 -DESP_PLATFORM=1 -B /media/pi/data/projects/esp32/hello_world/build -S /media/pi/data/projects/esp32/hello_world'" terminated with exit code: 1. 
```

beta-tester, 2022-11-13:
just now, i see the following issues:
1. the **Espressif ESP-IDF Extension** is installing some wrong binaries (e.g., `amd64`, ...).
2. the **Espressif ESP-IDF Extension** ignores already installed pre-required packages in the system (e.g., `cmake`, ...)
3. the **Espressif ESP-IDF Extension** does not setup building environment (e.g., you have to start vs code from console after you ran `export.sh`)
4. the **Espressif ESP-IDF Extension** does not build because of is missing `idf_component_manager` but `idf.py build` can build the project.

beta-tester, 2022-11-13:
here the ESP-IDF Doctor Command report:
```log
---------------------------------------------- ESP-IDF Extension for Visual Studio Code report ---------------------------------------------
OS linux arm 5.15.76-v7l+ 
System environment variable IDF_PYTHON_ENV_PATH 
 <HOMEPATH>/.espressif/python_env/idf4.4_py3.9_env 
System environment variable PATH 
 <HOMEPATH>/esp/esp-idf/components/esptool_py/esptool:<HOMEPATH>/esp/esp-idf/components/espcoredump:<HOMEPATH>/esp/esp-idf/components/partition_table:<HOMEPATH>/esp/esp-idf/components/app_update:<HOMEPATH>/.espressif/tools/xtensa-esp32-elf/esp-2021r2-patch5-8.4.0/xtensa-esp32-elf/bin:<HOMEPATH>/.espressif/tools/esp32ulp-elf/2.35_20220830/esp32ulp-elf/bin:<HOMEPATH>/.espressif/tools/openocd-esp32/v0.11.0-esp32-20220706/openocd-esp32/bin:<HOMEPATH>/.espressif/python_env/idf4.4_py3.9_env/bin:<HOMEPATH>/esp/esp-idf/tools:<HOMEPATH>/.cargo/bin:<HOMEPATH>/.local/bin:<HOMEPATH>/bin:/usr/local/sbin:/usr/local/bin:/usr/sbin:/usr/bin:/sbin:/bin:/usr/local/games:/usr/games 
System environment variable PYTHON 
 undefined 
Visual Studio Code version 1.73.1 
Visual Studio Code language en 
Visual Studio Code shell /usr/bin/bash 
ESP-IDF Extension version 1.5.1 
Workspace folder /media/pi/data/projects/esp32/hello_world 
---------------------------------------------------- Extension configuration settings ------------------------------------------------------
ESP-ADF Path (idf.espAdfPath) ${env:ADF_PATH}
ESP-IDF Path (idf.espIdfPath) <HOMEPATH>/esp/esp-idf
ESP-MDF Path (idf.espMdfPath) ${env:MDF_PATH}
ESP-Matter Path (idf.espMatterPath) ${env:ESP_MATTER_PATH}
Custom extra paths (idf.customExtraPaths) 
Custom extra vars (idf.customExtraVars) {}
Virtual env Python Path (idf.pythonBinPath) python
Serial port (idf.port) /dev/ttyUSB1
OpenOCD Configs (idf.openOcdConfigs) interface/ftdi/esp32_devkitj_v1.cfg,board/esp32-wrover.cfg
ESP-IDF Tools Path (idf.toolsPath) <HOMEPATH>/.espressif
Git Path (idf.gitPath) /usr/bin/git
-------------------------------------------------------- Configurations access -------------------------------------------------------------
Access to ESP-ADF Path (idf.espAdfPath) false
Access to ESP-IDF Path (idf.espIdfPath) true
Access to ESP-MDF Path (idf.espMdfPath) false
Access to ESP-Matter Path (idf.espMatterPath) false
Access to ESP-IDF Custom extra paths
Access to Virtual env Python Path (idf.pythonBinPath) false
Access to CMake in environment PATH true
Access to Ninja in environment PATH true
Access to ESP-IDF Tools Path (idf.toolsPath) true
-------------------------------------------------------- Configurations has spaces -------------------------------------------------------------
Spaces in system environment Path false
Spaces in ESP-ADF Path (idf.espAdfPath) false
Spaces in ESP-IDF Path (idf.espIdfPath) false
Spaces in ESP-MDF Path (idf.espMdfPath) false
Spaces in ESP-Matter Path (idf.espMatterPath) false
Spaces in ESP-IDF Custom extra paths
Spaces in Virtual env Python Path (idf.pythonBinPath) false
Spaces in ESP-IDF Tools Path (idf.toolsPath) false
----------------------------------------------------------- Executables Versions -----------------------------------------------------------
Git version 2.30.2
ESP-IDF version 4.4.3
Python version 3.9.2
Python's pip version 22.3.1
-------------------------------------------------- Python packages in idf.pythonBinPath ----------------------------------------------------
bitstring version: 3.1.9
Brotli version: 1.0.9
certifi version: 2022.9.24
cffi version: 1.15.1
charset-normalizer version: 2.1.1
click version: 8.1.3
colorama version: 0.4.6
construct version: 2.10.54
contextlib2 version: 21.6.0
cryptography version: 38.0.3
ecdsa version: 0.18.0
Flask version: 0.12.5
Flask-Compress version: 1.13
Flask-SocketIO version: 2.9.6
future version: 0.18.2
gdbgui version: 0.13.2.0
gevent version: 1.5.0
greenlet version: 2.0.1
[... 52 lines cut ...]
                "${config:idf.espIdfPath}/components/**",
                "${config:idf.espIdfPathWin}/components/**",
                "${config:idf.espAdfPath}/components/**",
                "${config:idf.espAdfPathWin}/components/**",
                "${workspaceFolder}/**"
            ],
            "browse": {
                "path": [
                    "${config:idf.espIdfPath}/components",
                    "${config:idf.espIdfPathWin}/components",
                    "${config:idf.espAdfPath}/components/**",
                    "${config:idf.espAdfPathWin}/components/**",
                    "${workspaceFolder}"
                ],
                "limitSymbolsToIncludedHeaders": false
            }
        }
    ],
    "version": 4
}
 
----------------------------------------------------------- Latest error -----------------------------------------------------------------
Latest error at Unknown error in ESP-IDF doctor command
--------------------------------------------------------------------------------------------------------------------------------------------
```

beta-tester, 2022-11-13:
i ran ADVANCED setup again and used for "Select Python version": `/usr/bin/python` instead of preselected `/home/pi/.espressif/python_env/idf4.4_py3.9_env`
now i can configure and build the `hello_world` project.

ok, only the first 3 issues left:
1. the Espressif ESP-IDF Extension is installing some wrong binaries (e.g., amd64, ...).
2. the Espressif ESP-IDF Extension ignores already installed pre-required packages in the system (e.g., cmake, ...)
3. the Espressif ESP-IDF Extension does not setup building environment (e.g., you have to start vs code from console after you ran export.sh)

brianignacio5, 2022-11-25:
There are many aspects here to consider:

- Your OS  is not listed in the `$IDF_PATH/tools/tools.json` which is used to download tools for your OS and validate existing tools against `tools.json`.
- How can you determine the extension ignores pre-required package? CMake seems to run even though the environment is not properly configured.
- The extension doesn't setup the environment because the extension has not been configured properly. If the extension is configured properly, you could open a `ESP-IDF Terminal` with env variables already configured.

I think the main issue here is that we rely on tools.json to give proper validation of tools while is not working. We need to improve this for not supported OS there.

You can try using a [json manual configuration](https://github.com/espressif/vscode-esp-idf-extension/blob/master/docs/SETUP.md#json-manual-configuration) of the extension.

beta-tester, 2022-11-28:
> Your OS is not listed in the `$IDF_PATH/tools/tools.json` which is used to download tools for your OS and validate existing tools against `tools.json`.

then i am wondering why installing `esp-idf` manually is pulling the correct binariey, but installing throught `vscode-esp-idf-extension` is pulling the wrong binaries.

> How can you determine the extension ignores pre-required package? CMake seems to run even though the environment is not properly configured.

because `vscode-esp-idf-extension` tried to execute the `amd64` version of CMake, even there is cmake package already installed with the correct `arm` architecture on the system.

brianignacio5, 2022-11-29:
Could you share the extension log ?

Windows: %USERPROFILE%\.vscode\extensions\espressif.esp-idf-extension-VERSION\esp_idf_vsc_ext.log
Linux & MacOSX: $HOME/.vscode/extensions/espressif.esp-idf-extension-VERSION/esp_idf_vsc_ext.log

Apologies I was wrong. Indeed for some reason the extension is downloading the tools from the wrong version of your OS. It is weird because based on your install log, some tools are indeed downloaded corresponding with your OS and architecture but some tools are not.

You can see that `idf.customExtraPaths` and `idf.pythonBinPath` are not configured with the right values which make me think that extension failed to set the values in `settings.json`.

beta-tester, 2022-11-29:
i added the log to this post:
https://github.com/espressif/vscode-esp-idf-extension/issues/840#issuecomment-1312564959

brianignacio5, 2022-11-30:
What is the result of `uname -m` in your OS terminal?

It seems. the CMake being called is installed incorrectly. The Tools.json in the extension is only used for really old version of ESP-IDF that didn't implemented it before. Need to Review the setup process to see what is wrong.

beta-tester, 2022-11-30:
on Raspberry Pi 4 with Raspi OS (32bit)
```bash
~ $ uname -m
armv7l
```

beta-tester, 2022-12-22:
but it is not fixed - or is it?

brianignacio5, 2023-01-04:
Could you try this [vsix installer](https://nightly.link/espressif/vscode-esp-idf-extension/actions/runs/3834819964/esp-idf-extension.vsix.zip) ?

beta-tester, 2023-01-04:
i have not much time today, so i did only a quick run on an RPi4 where
1. i just deleted the `~/.config/Code` and `~/.vscode` folder to reset VS Code environment and all the `~/esp`, `~/.espressif`, `~/.config/esp` and `~/.cache/Espressif` folders.
2. after a system reboot i downloaded your `vsix` file.
3. i unziped the content to `/tmp/test/esp-idf-extension.vsix`
4. in VS Code under `EXTENSIONS | ...` i clicked `Install from VSIX` and then selected the file `/tmp/test/esp-idf-extension.vsix/esp-idf-extension.vsix`
5. after a while the extension is installed:
   [image]
6. on the Espressif setup tab, i clicked EXPRESS. and selected `v5.0 (release version)` as ESP-IDF version to install.
7. alter click to install, i got in the log window `ninja doesn't have a compatible version for linux-armel` and the install button ist still there.
   [image]
   [esp_idf_vsc_ext.log](https://github.com/espressif/vscode-esp-idf-extension/files/10344120/esp_idf_vsc_ext.log)

brianignacio5, 2023-01-05:
Oh yes I see. I added some validation to use PATH ninja instead in this [latest vsix installer](https://nightly.link/espressif/vscode-esp-idf-extension/actions/artifacts/498861807.zip)

beta-tester, 2023-01-05:
hi, i tried your new version.
i did the same as yesterday.

but when i open the Espressif setup tab and select `v5.0 (release version)` for ESP-IDF version, the text box for `IDF_TOOLS_PATH` stays empty and the installation can not start until i fill it with `/home/pi/.espressif`.
this text box was filled in the previous version automatically. 

except of that, the installation and setup runs throught.
- i can run ESP-Doctor.
- i can use 'hello-world' from the ESP-Examples.
- i can select the target device ESP32 / ESP-WROOM-3.3V DevKit.
- i can open, edit and save the configuration of the project in the ESP-SDK Configuration Editor.
- i can build the project.
- i can select the flash method.
- i can select the flash port.
- i can flash.
- i can monitor.

well done. thank you!
when you can bring back the default path for the `ISP_TOOLS_PATH` text box, in the EXPRESS setup, it will be perfect.

brianignacio5, 2023-01-06:
Thank you for your feedback @beta-tester 

I'm not sure how the IDF_TOOLS_PATH stays empty. I can't reproduce it locally and this PR changes should not affect it. Maybe the value of `idf.toolsPath` in settings.json is empty ? The default value comes from this setting with fallback to env variable `$IDF_TOOLS_PATH` or `$HOME/.espressif`

beta-tester, 2023-01-07:
as i remember, the only difference i did was, in the previous try i closed VS Code just after the installation of the ESP Extension and after reopened VS Code, i did the Express setup of the environment.
in the last try, i continued the Express setup just after the installation of the ESP Extension without closind VS Code in between. maybe this makes a different. or i just forgot to clean something before the try.

maybe i retry it later again...

beta-tester, 2023-01-08:
> I'm not sure how the IDF_TOOLS_PATH stays empty. I can't reproduce it locally and this PR changes should not affect it. Maybe the value of idf.toolsPath in settings.json is empty ? The default value comes from this setting with fallback to env variable $IDF_TOOLS_PATH or $HOME/.espressif

i can reproduce the behavior.

- if i use the extension from your first post https://github.com/espressif/vscode-esp-idf-extension/issues/840#issuecomment-1370451273 , then the field is filled automatically.
   [image]

- if i use the extension from your second post, https://github.com/espressif/vscode-esp-idf-extension/issues/840#issuecomment-1371822655 , then the filed stays empty.
  [image]

so there must be something different in the extension.

brianignacio5, 2023-01-12:
Sorry for the delay, I think I might have fix the issue here:

https://nightly.link/espressif/vscode-esp-idf-extension/actions/artifacts/508506684.zip

Could you try see if this fix the issue ?

beta-tester, 2023-01-12:
> Could you try see if this fix the issue ?

yes, i see the issue is fixed, and no new issues found. ;-)
thank you.

(i only tested the EXPRESS setup)

beta-tester, 2023-01-13:
is the new extension already published in the regular public '_VS Code extension repository_'?
do i have to reinstall or update the manually locally installed extension to receive future updates?

brianignacio5, 2023-01-13:
The fix here is not published yet just merged to GitHub master branch. When the new release is publish the vscode will update to the new one in the marketplace automatically.

PR #859 (2023-01-13): add linux archs to platform info
## Description

Add platform in idfToolsManager.ts to manage Linux arm packages.

Fix #840

## Type of change

Please delete options that are not relevant.
- Bug fix (non-breaking change which fixes an issue)

## How has this been tested?

Manual testing of the setup wizard in Linux.

**Test Configuration**:
* ESP-IDF Version: 5.1
* OS (Windows,Linux and macOS): Linux

## Checklist
- [x] PR Self Reviewed
- [x] Applied Code formatting
- [ ] Added Documentation
- [ ] Added Unit Test
- [ ] Verified on all platforms - Windows,Linux and macOS