espressif/vscode-esp-idf-extension #842: [Feature Request]: Add New Debugging Views for Variables in "Watch"-Window

Delzye, 2022-11-16:
The new way to display structs is really nice, but it only works for local variables. If I want to watch e.g. a global variable, the old view is used (which admittedly is not very useful).
The "View Memory"-Button is also unavailable for the "Watch"-View.
 
Is It possible to also use the new way of displaying variables in the "Watch"-View, as well as adding the "view memory" button?

I wouldn't know how to properly look into some variables otherwise.

Also there is no "View Memory" button for pointers right now (second screenshot, `atBuf` points at an array inside of another array, but I cannot look at its contents, no button).

**Additional context**
 "Watch"-View still uses the old representation:
[image]
No Memory-View for a pointer to the start of an array:
[image]

brianignacio5, 2022-11-24:
Thanks for your feature request @Delzye 

Could you provide an example in c of those variables for me to reproduce ? There are a few GDB call need to happen on the debug adapter to get further information from those pointers for example. Understanding what kind of value is a pointer does help to query these values.

Delzye, 2022-11-24:
Hi @brianignacio5 
Sure, let's take these variables:
```c
    volatile struct struct_t local_a = {
      .x = 6,
      .y = 7,
      .z = 8
    };

    volatile int local_arr[] = {1,2,3,4,5};
    volatile int* local_arr_ptr = local_arr;
```
with:
```c
struct struct_t {
    int x;
    int y;
    int z;
};
```
This is the result:
[image]

Delzye, 2022-12-21:
Could this be reopened, please?
It has not been completed.

brianignacio5, 2023-01-05:
Could you try this [vsix installer](https://nightly.link/espressif/vscode-esp-idf-extension/actions/artifacts/496257414.zip) ??

PR #860 (2023-01-20): add watch vars expand point view bin data
## Description

Add expansion of complex variables in Debugger's Watch variables. Also add `View Binary data` for pointers (value of *0x######) by adding pointer address as memory reference.

Fix #842 

## Type of change

Please delete options that are not relevant.
- New feature (non-breaking change which adds functionality)

## How has this been tested?

Tested on MacOS with example values from #842

- Test A
- Test B

**Test Configuration**:
* ESP-IDF Version: 5.1
* OS (Windows,Linux and macOS): macOS

## Dependent components impacted by this PR:

- [... cut]
closed 2023-01-20 by brianignacio5