espressif/vscode-esp-idf-extension #845: [Feature Request]: How do I use the ESP Peripheral View

tionebrr, 2022-11-20:
Hello dear devs,
The documentation DEBUGGING.md mention svd Files and the ESP Peripheral View. However, how to set it up is not explained.
It would be great if this could be clarified.
Thanks !

closed 2022-11-21 by tionebrr

brianignacio5, 2022-11-25:
Hi @tionebrr sorry for late reply You mean 

```
Our extension implements a ESP Peripheral View tree view in the Run and debug view which will use the SVD file defined in the IDF Svd File Path (idf.svdFilePath) configuration setting to populate a set of peripherals registers values for the active debug session target. You could find [Espressif SVD](https://github.com/espressif/svd) files from Espressif SVD.
```

from [here](https://github.com/espressif/vscode-esp-idf-extension/blob/master/docs/DEBUGGING.md#configuration-for-visual-studio-code-debug).

If the setting is defined and file exists the Peripheral view should render. Do you think this is not clear in the docs @tionebrr  ? We could improve the docs if so.

tionebrr, 2022-11-28:
I just didn't knew where to define `idf.svdFilePath`. I think just mentionning that it is in `settings.json` could help. Thanks for the support ;)