espressif/vscode-esp-idf-extension #846: Why ask users to install cmake ninja first thing?

ramiws, 2022-11-20:
**Is your feature request related to a problem? Please describe.**
 I'm always frustrated when you ask the uses to do something but then I see that you install cmake and ninja anyways during the tools download. 
I wasted time installing them without brew, for nothing. Thanks!

**Describe the solution you'd like**
remove the request to install cmake and ninja Prerequisites

**Describe alternatives you've considered**
Platformio

brianignacio5, 2022-11-25:
Could you describe where you see this requisites ?

ramiws, 2022-11-28:
What is the first thing you see when you install the extension? have you tried ?

brianignacio5, 2022-11-29:
Something like this?
[image]

or these
[image]

closed 2022-12-21 by github-actions[bot]

ramiws, 2023-01-21:
Both having the same issue...