espressif/vscode-esp-idf-extension #847: SDK Configuration editor does not reflect Kconfig changes

brunosilvano, 2022-11-20:
### OS

MacOS

### Operating System version

macOS 10.15

### Visual Studio Code version

1.68.1

### ESP-IDF version

4.4.3

### Python version

3.9.13

### Doctor command output

N/A

### Description

After manually changing the `Kconfig.projbuild` file (e.g. adding a new param) and running the command `ESP-IDF: SDK Configuration editor (menuconfig)`, the VS Code menu won't show the changes. They can be seen when running `idf.py menuconfig` in the terminal though or when the VS Code window is reloaded (like `Developer: Reload Window` command).

### Debug Message

```plain
N/A
```

### Other Steps to Reproduce

Could not try on other OS.

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

brianignacio5, 2022-11-21:
Did you have the `SDK Configuration Editor` opened before modifying the Kconfig.projbuild file ? The process keeps running in the background and updates might not be reflected. You need to Dispose the confserver process with the `Dispose current SDK Configuration editor server process` command and re run the SDK Configuration Editor to see the changes applied.

brunosilvano, 2022-11-21:
Yes, I opened it before, then closed it, modified the file and opened it again. I see, what you suggested works, just got to manually dispose the confserver after every change. Would be nicer if this was done automatically somehow, in my opinion. Makes sense as an improvement?

brianignacio5, 2022-11-21:
The problem is reduced to how to detect these changes ? In the extension we currently have a watcher for `sdkconfig` file when it changes to update the GUI values, but the issue is that the Kconfig_menus.json in the build directory used to build the SDK Configuration is built when SDK Configuration Editor starts (without cache). There are many Kconfig files in the ESP-IDF directory and adding a watcher for each consumes too much resources in the extension. We could try to add a custom setting to watch a set of files to trigger SDK Configuration Editor re-build (which could take some time each time) although it could cause many problems if the user is making many changes in short amount of time.

brunosilvano, 2022-11-24:
Yes, I think so. Ok, I see, it is not that simple then. Not sure if the side effects of watching the files are worth it then - one could easily do many changes without being aware of that. Maybe for now would just be nice to add a note about that in the docs? Like, just so people are not wondering why the menu won't update.

PR #849 (2022-11-29): add dispose sdk config process in docs
## Description

Add dispose SDK Configuration Editor process in related docs.

Fix #847

## Type of change

Please delete options that are not relevant.
- Documentation

## How has this been tested?

No test needed. Adding documentation

**Test Configuration**:
* ESP-IDF Version: 5.1
* OS (Windows,Linux and macOS): All

## Checklist
- [x] PR Self Reviewed
- [x] Applied Code formatting
- [x] Added Documentation
- [ ] Added Unit Test
- [x] Verified on all platforms - Windows,Linux and macOS
closed 2022-11-29 by brianignacio5