espressif/vscode-esp-idf-extension #851: [Feature Request]: Add support for EXTRA_PARTITION_SUBTYPES in Partition Table Editor

IH303, 2022-11-28:
I have some custom partition subtypes. I defined them as described [here](https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-guides/partition-tables.html#extra-partition-subtypes) using the **EXTRA_PARTITION_SUBTYPES** property. 
E.g.:
```
idf_build_set_property(EXTRA_PARTITION_SUBTYPES "data, rsa, 0x9a" APPEND)
```

The *ESP-IDF Partition Table Editor* does not recognize those custom subtypes (or their names). They are not shown in the drop-down menu.

It would be great, if I could chose my custom defined subtypes from the drop-down menu instead of typing hex strings. 
Currently I also cannot add a custom subtype using hex strings for the type **data** because it says:

> When type is "data", the subtype field can be specified as "ota" (0x00), "phy" (0x01), "nvs" (0x02), "nvs_keys" (0x04), "fat" (0x81), > "spiffs" (0x82) or a range of other component-specific subtypes (0x05, 0x06, 0x80, 0x81, 0x82)"

With the support for **EXTRA_PARTITION_SUBTYPES** the user could add his own component-specific subtypes for the type **data**

In short:
Add the possibility to chose custom subtypes defined with the **EXTRA_PARTITION_SUBTYPES** property from the drop-down menu in the *ESP-IDF Partition Table Editor*

I hope it is possible for you to implement this feature. Thank you.

brianignacio5, 2023-04-18:
The problem with this is that EXTRA_PARTITION_SUBTYPES is defined in CMakeLists.txt so using a custom partition.csv can't know before the build that these types exists. 

Need to discuss with ESP-IDF colleagues about this feature.

Xordania, 2025-05-26:
Hi, just wondering if this has been looked at recently? I'm seeing the same issue on the v5.4.1 version of the IDF.

brianignacio5, 2025-05-27:
This issue is low priority for us as we have pressing bug fixes to work first.

When we have some time we can try to work on this feature request/ enhancement