espressif/vscode-esp-idf-extension #853: [Feature Request]: support Host-Generated Flash Encryption

chipweinberger, 2022-12-09:
### Backgound
Docs: https://docs.espressif.com/projects/esp-idf/en/latest/esp32s3/security/flash-encryption.html#manual-encryption

ESP-IDF has 2 flash encryption modes:
- release
- development

### Problem
**Release mode** is not supported by the VSC ESP-IDF Extension, even if we have a Host Generated key.encrypted.

### Suggested Solution
- VSC should check If 'Flash Encryption (Release)' is enabled in the menuconfig
- VSC should automatically prompt me to enter a Host Generated File (and add it as a preference)
- VSC should pre-encypt the app & bootloader firmware using `espsecure.py encrypt_flash_data --aes_xts --keyfile /path/to/key.bin --address 0x10000 --output my-app-ciphertext.bin build/my-app.bin`
- VSC should upload the encyrypted firmware

chipweinberger, 2022-12-09:
Similarly, in Release, VSC Flash fails:

`A fatal error occurred: Detecting flash size is not supported in secure download mode. Need to manually specify flash size.`

This is despite my Flash Size being specified in my Menu Config.

[image]

chipweinberger, 2023-05-17:
@brianignacio5 appreciate you taking a look at this again.

Just to provide more thoughts, I'm thinking:
- the user would have to specify an aes key file
- vsc would encrypt the binary using this key and esp-secure.py
- vsc would upload the encrypted binary

brianignacio5, 2023-05-17:
Sorry this was automatically closed by the bot, This could be a nice feature plan but a bit dangerous for beginner users so need to consider some validations as well.

Thank you for the thoughts, will think of feature action plan for this and decide with IDF team a good course of action. Will tag any work done related to this.

chipweinberger, 2023-05-17:
No problem. No rush!

chipweinberger, 2023-12-31:
I'm getting closer to production, and this feature would be very useful.

- I want to use production devices for continued development 
- production devices use a host generated flash encryption key

This means i lose the ability to push the "flash" button in vscode, which is annoying!

Perhaps I can rig up the "run custom command button" myself as a workaround.

That said, i think this feature would be very simple to implement? Like 1 day of work maximum.