espressif/vscode-esp-idf-extension #854: Windows project path not escaped

Nicell, 2022-12-12:
### OS

Windows

### Operating System version

Windows 11

### ESP-IDF version

5.0

### Python version

3.8.7

### Doctor command output

Nothing? I run the command and all it says is preparing, then nothing. I'm using an ESP32 S3 Box

### Description

From what it looks like, on pressing the build button on Windows, I get an error that looks like the path isn't properly escaped. I haven't done any special configuration and feel like I'm missing something. I'm using the esp-box examples to build.

### Debug Message

```plain
Executing task: cmake -G Ninja -DPYTHON_DEPS_CHECKED=1 -DESP_PLATFORM=1 -B c:\Users\Nick\Documents\Git\esp-box\examples\factory_demo\build -S c:\Users\Nick\Documents\Git\esp-box\examples\factory_demo 

CMake Error: The source directory "C:UsersNickDocumentsGitesp-boxexamplesfactory_demo" does not exist.
Specify --help for usage, or press the help button on the CMake GUI.
```

### Other Steps to Reproduce

1. Express installation using the VSCode plugin
2. Use the esp-box repository
3. Open the factory demo example
4. Click the build button given by the extension
5. Error by unescaped path

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

Nicell, 2022-12-29:
Not stale

Nicell, 2023-01-14:
Not stale

Gobol, 2023-02-04:
Got exactly the same problem. Sitting here for a few hours, still nothing.
ESP-IDF build works ok. Flash - ok. 
When running ESP-IDF Monitor task - (Git Bash) mingw64 shell kicks in and screws up all windows-style paths by escaping.

My Doctor log below...

[esp_idf_vsc_ext.log](https://github.com/espressif/vscode-esp-idf-extension/files/10609288/esp_idf_vsc_ext.log)

Gobol, 2023-02-04:
In the meanwhile I switched to virtualized ubuntu toolchain... works ok.

Gobol, 2023-02-09:
@radurentea, I don't really know what are you asking for... You wan't me to save *.log file as *.txt? Or to translate json to txt ?

Here comes the error dump from terminal : 

```
Executing task: C:/Users/Gobo/.espressif/python_env/idf5.0_py3.8_env/Scripts/python.exe C:\Users\Gobo\esp\esp-idf\tools\idf_size.py d:\Code\GoboHomeFirmware\hk_esp32_test\build\hello_world.map 

C:\Users\Gobo\.espressif\tools\idf-python\3.8.7\python.exe: can't open file 'C:UsersGoboespesp-idftoolsidf_size.py': [Errno 2] No such file or directory

 *  The terminal process "C:\Program Files\Git\bin\bash.exe '-c', 'C:/Users/Gobo/.espressif/python_env/idf5.0_py3.8_env/Scripts/python.exe C:\Users\Gobo\esp\esp-idf\tools\idf_size.py d:\Code\GoboHomeFirmware\hk_esp32_test\build\hello_world.map'" terminated with exit code: 2.
```

`---------------------------------------------- ESP-IDF Extension for Visual Studio Code report ---------------------------------------------
OS win32 x64 10.0.19044 
System environment variable IDF_PYTHON_ENV_PATH 
 undefined 
System environment variable PATH 
 C:\Program Files\Common Files\Oracle\Java\javapath;C:\Program Files\Oculus\Support\oculus-runtime;C:\Program Files (x86)\Common Files\Oracle\Java\javapath;C:\Windows\system32;C:\Windows;C:\Windows\System32\Wbem;C:\Windows\System32\WindowsPowerShell\v1.0\;C:\Windows\System32\OpenSSH\;C:\Program Files (x86)\Intel\Intel(R) Management Engine Components\DAL;C:\Program Files\Intel\Intel(R) Management Engine Components\DAL;C:\Program Files\NVIDIA Corporation\NVIDIA NvDLISR;C:\Program Files\Git\cmd;D:\Code\nodejs\;C:\Program Files\PuTTY\;C:\Program Files (x86)\IVI Foundation\VISA\WinNT\Bin\;C:\Program Files\IVI Foundation\VISA\Win64\Bin\;C:\Program Files (x86)\IVI Foundation\VISA\WinNT\Bin;C:\Program Files (x86)\Windows Kits\8.1\Windows Performance Toolkit\;C:\WINDOWS\system32;C:\WINDOWS;C:\WINDOWS\System32\Wbem;C:\WINDOWS\System32\WindowsPowerShell\v1.0\;C:\WINDOWS\System32\OpenSSH\;C:\Program Files (x86)\ZeroTier\One\;C:\Program Files (x86)\IVI Foundation\IVI\Bin\;C:\Program Files\IVI Foundation\IVI\Bin\;C:\Program Files (x86)\NVIDIA Corporation\PhysX\Common;C:\Program Files\dotnet\;C:\Program Files\Azure Data Studio\bin;C:<HOMEPATH>\AppData\Local\Programs\Python\Python38-32\Scripts\;C:<HOMEPATH>\AppData\Local\Programs\Python\Python38-32\;C:<HOMEPATH>\AppData\Local\Microsoft\WindowsApps;C:<HOMEPATH>\AppData\Local\Programs\Microsoft VS Code\bin;C:\Program Files (x86)\Nmap;C:<HOMEPATH>\AppData\Roaming\npm;D:\Code\SysGCC\bin;C:\SysGCC\raspberry\bin;C:<HOMEPATH>\AppData\Local\GitHubDesktop\bin;C:<HOMEPATH>\AppData\Local\Microsoft\WindowsApps;C:<HOMEPATH>\.dotnet\tools 
System environment variable PYTHON 
 undefined 
Visual Studio Code version 1.75.0 
Visual Studio Code language en 
Visual Studio Code shell C:\Program Files\Git\bin\bash.exe 
ESP-IDF Extension version 1.5.1 
Workspace folder d:\Code\GoboHomeFirmware\hk_esp32_test 
---------------------------------------------------- Extension configuration settings ------------------------------------------------------
ESP-ADF Path (idf.espAdfPath) ${env:ADF_PATH}
ESP-IDF Path (idf.espIdfPath) C:/Users/Gobo/esp/esp-idf
ESP-MDF Path (idf.espMdfPath) ${env:MDF_PATH}
ESP-Matter Path (idf.espMatterPath) ${env:ESP_MATTER_PATH}
Custom extra paths (idf.customExtraPaths) C:<HOMEPATH>\.espressif\tools\xtensa-esp-elf-gdb\11.2_20220823\xtensa-esp-elf-gdb\bin;C:<HOMEPATH>\.espressif\tools\riscv32-esp-elf-gdb\11.2_20220823\riscv32-esp-elf-gdb\bin;C:<HOMEPATH>\.espressif\tools\xtensa-esp32-elf\esp-2022r1-11.2.0\xtensa-esp32-elf\bin;C:<HOMEPATH>\.espressif\tools\xtensa-esp32s2-elf\esp-2022r1-11.2.0\xtensa-esp32s2-elf\bin;C:<HOMEPATH>\.espressif\tools\xtensa-esp32s3-elf\esp-2022r1-11.2.0\xtensa-esp32s3-elf\bin;C:<HOMEPATH>\.espressif\tools\riscv32-esp-elf\esp-2022r1-11.2.0\riscv32-esp-elf\bin;C:<HOMEPATH>\.espressif\tools\esp32ulp-elf\2.35_20220830\esp32ulp-elf\bin;C:<HOMEPATH>\.espressif\tools\cmake\3.24.0\bin;C:<HOMEPATH>\.espressif\tools\openocd-esp32\v0.11.0-esp32-20221026\openocd-esp32\bin;C:<HOMEPATH>\.espressif\tools\ninja\1.10.2;C:<HOMEPATH>\.espressif\tools\idf-exe\1.0.3;C:<HOMEPATH>\.espressif\tools\ccache\4.6.2\ccache-4.6.2-windows-x86_64;C:<HOMEPATH>\.espressif\tools\dfu-util\0.9\dfu-util-0.9-win64
Custom extra vars (idf.customExtraVars) {"OPENOCD_SCRIPTS":"C:\\Users\\Gobo\\.espressif\\tools\\openocd-esp32\\v0.11.0-esp32-20221026/openocd-esp32/share/openocd/scripts","IDF_CCACHE_ENABLE":"1"}
Virtual env Python Path (idf.pythonBinPath) C:/Users/Gobo/.espressif/python_env/idf5.0_py3.8_env/Scripts/python.exe
Serial port (idf.port) COM5
OpenOCD Configs (idf.openOcdConfigs) board/esp32-bridge.cfg
ESP-IDF Tools Path (idf.toolsPath) C:/Users/Gobo/.espressif
Git Path (idf.gitPath) C:<HOMEPATH>\.espressif\tools\idf-git\2.30.1\cmd\git.exe
-------------------------------------------------------- Configurations access -------------------------------------------------------------
Access to ESP-ADF Path (idf.espAdfPath) false
Access to ESP-IDF Path (idf.espIdfPath) true
Access to ESP-MDF Path (idf.espMdfPath) false
Access to ESP-Matter Path (idf.espMatterPath) false
Access to ESP-IDF Custom extra paths
Access to C:<HOMEPATH>\.espressif\tools\xtensa-esp-elf-gdb\11.2_20220823\xtensa-esp-elf-gdb\bin: true
Access to C:<HOMEPATH>\.espressif\tools\riscv32-esp-elf-gdb\11.2_20220823\riscv32-esp-elf-gdb\bin: true
Access to C:<HOMEPATH>\.espressif\tools\xtensa-esp32-elf\esp-2022r1-11.2.0\xtensa-esp32-elf\bin: true
[... cut ...]
---------------------------------------------------- Check ESP-IDF debug adapter requirements.txt ------------------------------------------
Check Debug AdapterPython packages Python requirements are satisfied.
----------------------------------------------------------- Latest error -----------------------------------------------------------------
Latest error at Unknown error in ESP-IDF doctor command
--------------------------------------------------------------------------------------------------------------------------------------------
`

Gobol, 2023-02-11:
@radurentea, I've changed given settings to your string, and here is the result - even worse : 

```
/usr/bin/bash: C:UsersGobo.espressifpython_envidf5.0_py3.8_envScriptspython.exe: command not found

 *  The terminal process "C:\Program Files\Git\bin\bash.exe '-c', 'C:\Users\Gobo\.espressif\python_env\idf5.0_py3.8_env\Scripts\python.exe C:\Users\Gobo\esp\esp-idf\tools\idf_size.py d:\Code\GoboHomeFirmware\hk_esp32_test\build\hello_world.map'" terminated with exit code: 127. 
```
And the Doctor says:

```
---------------------------------------------- ESP-IDF Extension for Visual Studio Code report ---------------------------------------------
OS win32 x64 10.0.19044 
System environment variable IDF_PYTHON_ENV_PATH 
 undefined 
System environment variable PATH 
 C:\Program Files\Common Files\Oracle\Java\javapath;C:\Program Files\Oculus\Support\oculus-runtime;C:\Program Files (x86)\Common Files\Oracle\Java\javapath;C:\Windows\system32;C:\Windows;C:\Windows\System32\Wbem;C:\Windows\System32\WindowsPowerShell\v1.0\;C:\Windows\System32\OpenSSH\;C:\Program Files (x86)\Intel\Intel(R) Management Engine Components\DAL;C:\Program Files\Intel\Intel(R) Management Engine Components\DAL;C:\Program Files\NVIDIA Corporation\NVIDIA NvDLISR;C:\Program Files\Git\cmd;D:\Code\nodejs\;C:\Program Files\PuTTY\;C:\Program Files (x86)\IVI Foundation\VISA\WinNT\Bin\;C:\Program Files\IVI Foundation\VISA\Win64\Bin\;C:\Program Files (x86)\IVI Foundation\VISA\WinNT\Bin;C:\Program Files (x86)\Windows Kits\8.1\Windows Performance Toolkit\;C:\WINDOWS\system32;C:\WINDOWS;C:\WINDOWS\System32\Wbem;C:\WINDOWS\System32\WindowsPowerShell\v1.0\;C:\WINDOWS\System32\OpenSSH\;C:\Program Files (x86)\ZeroTier\One\;C:\Program Files (x86)\IVI Foundation\IVI\Bin\;C:\Program Files\IVI Foundation\IVI\Bin\;C:\Program Files (x86)\NVIDIA Corporation\PhysX\Common;C:\Program Files\dotnet\;C:\Program Files\Azure Data Studio\bin;C:<HOMEPATH>\AppData\Local\Programs\Python\Python38-32\Scripts\;C:<HOMEPATH>\AppData\Local\Programs\Python\Python38-32\;C:<HOMEPATH>\AppData\Local\Microsoft\WindowsApps;C:<HOMEPATH>\AppData\Local\Programs\Microsoft VS Code\bin;C:\Program Files (x86)\Nmap;C:<HOMEPATH>\AppData\Roaming\npm;D:\Code\SysGCC\bin;C:\SysGCC\raspberry\bin;C:<HOMEPATH>\AppData\Local\GitHubDesktop\bin;C:<HOMEPATH>\AppData\Local\Microsoft\WindowsApps;C:<HOMEPATH>\.dotnet\tools 
System environment variable PYTHON 
 undefined 
Visual Studio Code version 1.75.0 
Visual Studio Code language en 
Visual Studio Code shell C:\Program Files\Git\bin\bash.exe 
ESP-IDF Extension version 1.5.1 
Workspace folder d:\Code\GoboHomeFirmware\hk_esp32_test 
---------------------------------------------------- Extension configuration settings ------------------------------------------------------
ESP-ADF Path (idf.espAdfPath) ${env:ADF_PATH}
ESP-IDF Path (idf.espIdfPath) C:<HOMEPATH>\esp\esp-idf
ESP-MDF Path (idf.espMdfPath) ${env:MDF_PATH}
ESP-Matter Path (idf.espMatterPath) ${env:ESP_MATTER_PATH}
Custom extra paths (idf.customExtraPaths) C:<HOMEPATH>\.espressif\tools\xtensa-esp-elf-gdb\11.2_20220823\xtensa-esp-elf-gdb\bin;C:<HOMEPATH>\.espressif\tools\riscv32-esp-elf-gdb\11.2_20220823\riscv32-esp-elf-gdb\bin;C:<HOMEPATH>\.espressif\tools\xtensa-esp32-elf\esp-2022r1-11.2.0\xtensa-esp32-elf\bin;C:<HOMEPATH>\.espressif\tools\xtensa-esp32s2-elf\esp-2022r1-11.2.0\xtensa-esp32s2-elf\bin;C:<HOMEPATH>\.espressif\tools\xtensa-esp32s3-elf\esp-2022r1-11.2.0\xtensa-esp32s3-elf\bin;C:<HOMEPATH>\.espressif\tools\riscv32-esp-elf\esp-2022r1-11.2.0\riscv32-esp-elf\bin;C:<HOMEPATH>\.espressif\tools\esp32ulp-elf\2.35_20220830\esp32ulp-elf\bin;C:<HOMEPATH>\.espressif\tools\cmake\3.24.0\bin;C:<HOMEPATH>\.espressif\tools\openocd-esp32\v0.11.0-esp32-20221026\openocd-esp32\bin;C:<HOMEPATH>\.espressif\tools\ninja\1.10.2;C:<HOMEPATH>\.espressif\tools\idf-exe\1.0.3;C:<HOMEPATH>\.espressif\tools\ccache\4.6.2\ccache-4.6.2-windows-x86_64;C:<HOMEPATH>\.espressif\tools\dfu-util\0.9\dfu-util-0.9-win64
Custom extra vars (idf.customExtraVars) {"OPENOCD_SCRIPTS":"C:\\Users\\Gobo\\.espressif\\tools\\openocd-esp32\\v0.11.0-esp32-20221026/openocd-esp32/share/openocd/scripts","IDF_CCACHE_ENABLE":"1"}
Virtual env Python Path (idf.pythonBinPath) C:<HOMEPATH>\.espressif\python_env\idf5.0_py3.8_env\Scripts\python.exe
Serial port (idf.port) COM5
OpenOCD Configs (idf.openOcdConfigs) board/esp32-bridge.cfg
ESP-IDF Tools Path (idf.toolsPath) C:<HOMEPATH>\.espressif
Git Path (idf.gitPath) C:<HOMEPATH>\.espressif\tools\idf-git\2.30.1\cmd\git.exe
-------------------------------------------------------- Configurations access -------------------------------------------------------------
Access to ESP-ADF Path (idf.espAdfPath) false
Access to ESP-IDF Path (idf.espIdfPath) true
Access to ESP-MDF Path (idf.espMdfPath) false
Access to ESP-Matter Path (idf.espMatterPath) false
Access to ESP-IDF Custom extra paths
Access to C:<HOMEPATH>\.espressif\tools\xtensa-esp-elf-gdb\11.2_20220823\xtensa-esp-elf-gdb\bin: true
Access to C:<HOMEPATH>\.espressif\tools\riscv32-esp-elf-gdb\11.2_20220823\riscv32-esp-elf-gdb\bin: true
Access to C:<HOMEPATH>\.espressif\tools\xtensa-esp32-elf\esp-2022r1-11.2.0\xtensa-esp32-elf\bin: true
Access to C:<HOMEPATH>\.espressif\tools\xtensa-esp32s2-elf\esp-2022r1-11.2.0\xtensa-esp32s2-elf\bin: true
Access to C:<HOMEPATH>\.espressif\tools\xtensa-esp32s3-elf\esp-2022r1-11.2.0\xtensa-esp32s3-elf\bin: true
Access to C:<HOMEPATH>\.espressif\tools\riscv32-esp-elf\esp-2022r1-11.2.0\riscv32-esp-elf\bin: true
[... cut ...]
---------------------------------------------------- Check ESP-IDF debug adapter requirements.txt ------------------------------------------
Check Debug AdapterPython packages Python requirements are satisfied.
----------------------------------------------------------- Latest error -----------------------------------------------------------------
Latest error at Unknown error in ESP-IDF doctor command
--------------------------------------------------------------------------------------------------------------------------------------------

```

brianignacio5, 2023-02-23:
Could you change the terminal you are using in vscode not use `Visual Studio Code shell C:\Program Files\Git\bin\bash.exe `?

Maybe use Powershell or cmd ? The reason why path are not escaping is because git bash works like a unix terminal not a windows one.

Gobol, 2023-03-01:
I have my default terminal in VSCode set to cmd.exe, tried powershell as well, but *ESP-IDF Size* do not recognize this setting, and still use bash from Git.

brianignacio5, 2023-03-13:
Based on your doctor command output: `Visual Studio Code shell C:\Program Files\Git\bin\bash.exe ` Maybe there is a local settings.json in your workspace folder or workspace overriding the system settings ?

rogue-shadowdancer, 2023-03-15:
same problem

> Based on your doctor command output: `Visual Studio Code shell C:\Program Files\Git\bin\bash.exe ` Maybe there is a local settings.json in your workspace folder or workspace overriding the system settings ?

I think this is the reason. but what if I do not want to just give up using bash? Or is there a setting that I can use powershell only for idf?

radurentea, 2023-03-28:
Hi @rogue-shadowdancer,

You could try to use specific workspace folder settings and in there, set the "terminal.external.windowsExec" to the powershell path.

Example:

`"terminal.external.windowsExec": "C:\\Windows\\System32\\cmd.exe"`

rogue-shadowdancer, 2023-03-29:
Hi @radurentea,

Many thanks! It looks great! Actually, I mean that I wonder if  idf extension will allow us to select the non-default terminal in the future features : )

Gobol, 2023-03-30:
For me :

`"terminal.integrated.shell.windows": "C:\\Windows\\System32\\cmd.exe",`

worked perfectly.
Set in workspace/folder **settings.json**.

Thanks.

radurentea, 2023-04-07:
> Hi @radurentea,
> 
> Many thanks! It looks great! Actually, I mean that I wonder if idf extension will allow us to select the non-default terminal in the future features : )

Hi, we could do something for sure in the future. Can you create a feature request for it and explain a bit how you would you like to be implemented?

closed 2023-04-07 by radurentea