espressif/vscode-esp-idf-extension #855: [Feature Request]: support sdkconfig.defaults instead of sdkconfig

chipweinberger, 2022-12-15:
For multiple build configs:
https://github.com/espressif/esp-idf/blob/master/examples/build_system/cmake/multi_config/README.md

In that recommended example, the `sdkconfig` file is placed in the `build` directory, and `sdkconfig.defaults` is used in the parent directory.

It seems VSC does not recognize this configuration and keeps putting a new `sdkconfig` file into the parent directory. 

I think VSC should open the `sdkconfig.defaults` file if it exists, and save to that file only, if it exists.

chipweinberger, 2022-12-15:
I think this multi-config setup with `sdkconfig.defaults` also causes the baud rate to not be found...

brianignacio5, 2023-01-09:
Could you try this [vsix installer](https://nightly.link/espressif/vscode-esp-idf-extension/actions/artifacts/503213293.zip) ?

chipweinberger, 2023-01-10:
wow very exciting! ill try it right now!!

brianignacio5, 2023-01-10:
Still need to add a way to select the sdkconfig defaults from build command, I'll be adding something like #671 using a tool build configuration file for this and dropbox to select each

chipweinberger, 2023-01-10:
Ok I tried it out.

**Works:**
- when editing the sdkconfig, VSC-ESP-IDF now loads the`sdkconfig.defaults` file, yay
- when saving the sdkconfig, VSC-ESP-IDF does not create a new `sdkconfig` file, yay

**Does not work:**
- when saving the sdkconfig, VSC-ESP-IDF does not update the the `sdkconfig.defaults`

Even without the feature fully finished, this is still a good improvement!

brianignacio5, 2023-01-10:
It shouldn't save the values in `sdkconfig.defaults` @chipweinberger . The values are saved in the `build/sdkconfig`. The idea here is to have multiple build folders each one containing a `sdkconfig` file for each config.

chipweinberger, 2023-01-10:
Yes. That makes sense of course. `build/sdkconfig` should always be updated when you save. 

In my project I have 2 configurations.

`sdkconfig.defaults`
`sdkconfig.prod`

[image]

I think somewhere in this toolbar, there should be a _sdkconfig_ selector. In my example, it would have 2 options available, **sdkconfig.defaults** and **sdkconfig.prod**.

Depending on your selection, the **SDK Configuration Editor** would save to selected file (and use it for `idf.py build`)

This is the best implementation I can think of.

chipweinberger, 2023-01-10:
Just updated my comment ^

brianignacio5, 2023-01-19:
In the PR I added the `idf.sdkconfigDefaults` configuration setting to customize the sdkconfig default files to use and added documentation to use the multiple configuration use case.

I think adding an additional UI to choose which sdkconfig files to use is not a common use case for most projects and it will make the UI more complicated. I think is better to combine several configs for the project in a file and allow to switch between these env. Something like the feature request in #671

chipweinberger, 2023-01-19:
Okay. As long as it is compatible with [Multiple Build Configurations Example](https://github.com/espressif/esp-idf/blob/master/examples/build_system/cmake/multi_config/README.md) then it is okay!

Thanks for working on this!

PR #870 (2023-01-20): add build sdkconfig support
## Description

Add support for multiple build configuration as done in [esp-idf cmake multi_config example](https://github.com/espressif/esp-idf/tree/master/examples/build_system/cmake/multi_config).

Load sdkconfig path as follows:
1) From current workspace folder root directory `CMakeLists.txt` SDKCONFIG variable else
2) From `SDKCONFIG` environment variable (either system or extension defined in `idf.customExtraVars` setting) else
3) From current workspace folder root directory `sdkconfig` file.

Also add `idf.sdkconfigDefaults` to define the set of SDKCONFIG_DE [... cut]
closed 2023-01-20 by brianignacio5