espressif/vscode-esp-idf-extension #856: esp-idf tool installation fails due to ERROR: MSys/Minggw is not supported

stsgil, 2022-12-16:
### OS

Windows

### Operating System version

Windows 10

### Visual Studio Code version

1.74.1

### ESP-IDF version

5.0

### Python version

3.10.9

### Doctor command output

Don't know, nothing happens when I run this command

### Extension

[esp_idf_vsc_ext.log](https://github.com/espressif/vscode-esp-idf-extension/files/10244344/esp_idf_vsc_ext.log)

### Description

ESP-IDF tool installation fails with, because I guess it finds MinGW64 through my git bash, which is not supported?

This is my Nth attempt at installing this stuff. Before I had MSYS installed, I was able to successfully configure/install everything, but when I tried to build my project, I got the same error.
Potentially something is wrong with my environment variable setup / I need msys after all? But in the readme for this extension, msys is not listed in the pre-requisites for Windows, I feel like you could at least provide some references to IDF documentation for those, who have no experience with this whole infrastructure.

### Debug Message

```plain
ERROR: MSys/Mingw is not supported. Please follow the getting started guide of the documentation to set up a supported environment
Command failed: "C:\Users\Stas\.espressif\tools\idf-python\3.8.7\python.exe" "C:\Users\Stas\esp\esp-idf\tools\idf_tools.py" install-python-env
```

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

stsgil, 2022-12-16:
Sorry if i sound a bit emotional. To add:
esp-idf docs say "please install idf through the IDE extension" (https://docs.espressif.com/projects/esp-idf/en/latest/esp32/get-started/index.html)
I also tried installing the esp-idf using the online installer 2.17 and then option "use existing idf" in the extension setup wizard, but I had a similar error, it failed to build mconf-idf because "no MSYS2"

stsgil, 2022-12-19:
I had some stale path variables in my vscode extension settings and i probably provided wrong IDF_PATH and IDF_TOOLS_PATH

kongzhongfeima, 2024-08-04:
same error

johnosbb, 2024-08-04:
same error installing on Windows

brianignacio5, 2024-09-09:
This error is triggered in ESP-IDF idf.py here: https://github.com/espressif/esp-idf/blob/master/tools/idf_tools.py#L3240

It seems there is an environment variable MSYSTEM in your setup that is being detected as MSys/Mingw

MartyMcFlyInTheSky, 2024-10-18:
Uninstall MSYS2. This env variable gets set under `C:\msys64\msys2_shell.cmd` which is (I believe) executed for bootstrapping.

henryfung3a27, 2025-03-25:
On Windows, setting the VSCode default terminal profile to anything other than `Git Bash` solved for me. i.e. in VSCode settings.json, comment out
```
  // "terminal.integrated.defaultProfile.windows": "Git Bash",
```