espressif/vscode-esp-idf-extension #858: App trace log‘s "Show Report" button no behavior

WangShuoran, 2022-12-24:
### OS

Windows

### Operating System version

win10

### Visual Studio Code version

1.74.2

### ESP-IDF version

4.4.3

### Python version

3.8.7(but my pc have python3.10 with environment path)

### Doctor command output

[output.txt](https://github.com/espressif/vscode-esp-idf-extension/files/10298218/output.txt)

### Extension

[output.txt](https://github.com/espressif/vscode-esp-idf-extension/files/10298220/output.txt)

### Description

When I am finish the trace and in the project have trace folder and a file was created in the folder. The file name prefix is trace.
Like this:
[image]
But when I enter the trace archives:
[image]
This click without any behavior. And I test the file is correct, because I use the file enter the powershell cammand(use idf environment):
```bash
python C:\Espressif\frameworks\esp-idf-v4.4.3\tools\esp_app_trace\logtrace_proc.py D:\mine\debug_temp\app_trace_to_host\trace\trace_1671867825325.trace D:\mine\debug_temp\app_trace_to_host\build\app_trace_to_host.elf
```
And the output is:
[image]

So I have last possible causes:
1. The logtrace_proc.py is not find
2. [extension](https://github.com/espressif/vscode-esp-idf-extension) can not know my elf file

### Debug Message

```plain
Not relevant
```

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

WangShuoran, 2022-12-24:
And I follow the document https://github.com/espressif/vscode-esp-idf-extension/blob/master/docs/tutorial/app_tracing.md which let me only open openocd, but as my test this cannot report data.

When I try open openocd and run debuuger and run the code(without breakpoint stop), and the record is recorded. I don't know why? Is my operation worry or document miss this?

brianignacio5, 2023-01-04:
Could you try this [vsix installer](https://nightly.link/espressif/vscode-esp-idf-extension/actions/artifacts/497499446.zip)

WangShuoran, 2023-01-04:
It can working, I expect it to become the official version.
[image]

And I have a little tip:
[image]
If an Trace log is 0kb(empty) when I show report it will no output like this:
[image]
But if the extension give any tip show this trace log is 0KB, I will very happy to doesn't spend time check this file.

brianignacio5, 2023-01-04:
Something like this?

[image]

Added in the same PR

PR #864 (2023-01-05): fix project elf file
## Description

Please include a summary of the change and which issue is fixed.

Fixes #858

## Type of change

Please delete options that are not relevant.
- Bug fix (non-breaking change which fixes an issue)

## How has this been tested?

Manual testing app tracing following the example tutorial in [here](https://github.com/espressif/vscode-esp-idf-extension/blob/master/docs/tutorial/app_tracing.md)

- Manual Test

**Test Configuration**:
* ESP-IDF Version: 4.4.3
* OS (Windows,Linux and macOS): macOS

## Checklist
- [x] PR Self Reviewed
- [x] Applied Code formatting
- [ [... cut]
closed 2023-01-05 by brianignacio5