espressif/vscode-esp-idf-extension #862: crosstool-ng version error after re-installing ESP-IDF with the vs-code extension

tongclement, 2023-01-01:
### OS

MacOS

### Operating System version

M1 Mac OS 12.6

### Visual Studio Code version

1.74.2 (Universal)

### ESP-IDF version

4.4.3

### Python version

3.10

### Doctor command output

```
---------------------------------------------- ESP-IDF Extension for Visual Studio Code report ---------------------------------------------
OS darwin arm64 21.6.0 
System environment variable IDF_PYTHON_ENV_PATH 
 undefined 
System environment variable PATH 
 <HOMEPATH>/.deta/bin:/usr/local/sbin:<HOMEPATH>/miniforge3/bin:<HOMEPATH>/miniforge3/condabin:/opt/homebrew/bin:/opt/homebrew/sbin:/opt/local/bin:/opt/local/sbin:/Library/Frameworks/Python.framework/Versions/3.10/bin:/usr/local/bin:/usr/bin:/bin:/usr/sbin:/sbin:/Library/Apple/usr/bin:/Library/Frameworks/Mono.framework/Versions/Current/Commands:<HOMEPATH>/Developer/flutter/bin 
System environment variable PYTHON 
 undefined 
Visual Studio Code version 1.74.2 
Visual Studio Code language en 
Visual Studio Code shell /bin/zsh 
ESP-IDF Extension version 1.5.1 
Workspace folder <HOMEPATH>/Desktop/Programming/2022/safelybuoy/experimentation/ardunio-as-a-component/ardunio-as-a-component-example 
---------------------------------------------------- Extension configuration settings ------------------------------------------------------
ESP-ADF Path (idf.espAdfPath) <HOMEPATH>/.espressif/esp-adf
ESP-IDF Path (idf.espIdfPath) <HOMEPATH>/esp/esp-idf
ESP-MDF Path (idf.espMdfPath) ${env:MDF_PATH}
ESP-Matter Path (idf.espMatterPath) ${env:ESP_MATTER_PATH}
Custom extra paths (idf.customExtraPaths) <HOMEPATH>/.espressif/tools/xtensa-esp32-elf/esp-2021r2-patch3-8.4.0/xtensa-esp32-elf/bin:<HOMEPATH>/.espressif/tools/xtensa-esp32s2-elf/esp-2021r2-patch3-8.4.0/xtensa-esp32s2-elf/bin:<HOMEPATH>/.espressif/tools/xtensa-esp32s3-elf/esp-2021r2-patch3-8.4.0/xtensa-esp32s3-elf/bin:<HOMEPATH>/.espressif/tools/riscv32-esp-elf/esp-2021r2-patch3-8.4.0/riscv32-esp-elf/bin:<HOMEPATH>/.espressif/tools/esp32ulp-elf/2.28.51-esp-20191205/esp32ulp-elf-binutils/bin:<HOMEPATH>/.espressif/tools/esp32s2ulp-elf/2.28.51-esp-20191205/esp32s2ulp-elf-binutils/bin:<HOMEPATH>/.espressif/tools/openocd-esp32/v0.11.0-esp32-20211220/openocd-esp32/bin
Custom extra vars (idf.customExtraVars) {"OPENOCD_SCRIPTS":"<HOMEPATH>/.espressif/tools/openocd-esp32/v0.11.0-esp32-20211220/openocd-esp32/share/openocd/scripts"}
Virtual env Python Path (idf.pythonBinPath) <HOMEPATH>/.espressif/python_env/idf4.3_py3.9_env/bin/python
Serial port (idf.port) /dev/cu.SLAB_USBtoUART
OpenOCD Configs (idf.openOcdConfigs) board/esp32-wrover-kit-3.3v.cfg
ESP-IDF Tools Path (idf.toolsPath) <HOMEPATH>/.espressif
Git Path (idf.gitPath) git
-------------------------------------------------------- Configurations access -------------------------------------------------------------
Access to ESP-ADF Path (idf.espAdfPath) true
Access to ESP-IDF Path (idf.espIdfPath) true
Access to ESP-MDF Path (idf.espMdfPath) false
Access to ESP-Matter Path (idf.espMatterPath) false
Access to ESP-IDF Custom extra paths
Access to <HOMEPATH>/.espressif/tools/xtensa-esp32-elf/esp-2021r2-patch3-8.4.0/xtensa-esp32-elf/bin: true
Access to <HOMEPATH>/.espressif/tools/xtensa-esp32s2-elf/esp-2021r2-patch3-8.4.0/xtensa-esp32s2-elf/bin: true
Access to <HOMEPATH>/.espressif/tools/xtensa-esp32s3-elf/esp-2021r2-patch3-8.4.0/xtensa-esp32s3-elf/bin: true
Access to <HOMEPATH>/.espressif/tools/riscv32-esp-elf/esp-2021r2-patch3-8.4.0/riscv32-esp-elf/bin: true
Access to <HOMEPATH>/.espressif/tools/esp32ulp-elf/2.28.51-esp-20191205/esp32ulp-elf-binutils/bin: true
Access to <HOMEPATH>/.espressif/tools/esp32s2ulp-elf/2.28.51-esp-20191205/esp32s2ulp-elf-binutils/bin: true
Access to <HOMEPATH>/.espressif/tools/openocd-esp32/v0.11.0-esp32-20211220/openocd-esp32/bin: true
Access to Virtual env Python Path (idf.pythonBinPath) true
Access to CMake in environment PATH true
Access to Ninja in environment PATH true
Access to ESP-IDF Tools Path (idf.toolsPath) true
-------------------------------------------------------- Configurations has spaces -------------------------------------------------------------
Spaces in system environment Path false
Spaces in ESP-ADF Path (idf.espAdfPath) false
Spaces in ESP-IDF Path (idf.espIdfPath) false
Spaces in ESP-MDF Path (idf.espMdfPath) false
[... 147 lines cut ...]

Toolchain:
  /Users/[username]/.espressif/tools/xtensa-esp32-elf/esp-2021r2-patch3-8.4.0/xtensa-esp32-elf/bin/xtensa-esp32-elf-gcc,
  crosstool-ng version esp-2021r2-patch3 doesn't match supported version
  esp-2021r2-patch5

  Please try to run 'idf.py fullclean' to solve it quickly.

  Check Getting Started documentation if the error continues.

  You can override this error and proceed with build by defining the
  IDF_MAINTAINER environment variable.
```

OR:
```
ninja: error: loading 'build.ninja': No such file or directory

 *  The terminal process "/bin/zsh '-c', 'ninja '" failed to launch (exit code: 1). 
```
```

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

dobairoland, 2023-01-03:
Notes at https://github.com/espressif/esp-idf/issues/10473 could provide more information on this issue. Let me know if any help is needed from the ESP-IDF side.

brianignacio5, 2023-01-04:
It might be related to some cache in the build directory. Try deleting the build folder and build again.

brianignacio5, 2023-01-18:
Any updates @tongclement ?

tongclement, 2023-01-18:
to be fixed

tongclement, 2023-01-18:
> It might be related to some cache in the build directory. Try deleting the build folder and build again.

I did do full cleans - it had no effect.

tongclement, 2023-01-18:
Thanks to @dobairoland for his help, he previously replied that:
https://github.com/espressif/esp-idf/issues/10473#issuecomment-1369738344
"Thanks @tongclement for your notes on this issue. I don't think that anything could be done on the ESP-IDF side. Perhaps in the linked vscode issue the developers will be able to improve this.

ESP-IDF does not change the PATH permanently. Changes to environment variables are discarded after the terminal is closed. PATH is updated by adding items to the beginning, therefore, globally set values are not affecting it. However, we cannot handle VScode-related workspace changes at this level. I'm closing this for this reason."

So maybe the can be changed from a bug report to a feature request/usability improvement: adding a one click idf updater/version changer. This might be seen as unnecessary for professionals who are very used to updating and downgrading but in my humble opinion it is a hindrance to using esp-idf (instead of the ardunio lib built on esp idf etc)

tongclement, 2023-01-18:
Thanks for all the help and apologies for the mediocre documentation on my end.

brianignacio5, 2023-01-19:
Thanks for the information @tongclement .

The issue I notice is that while you are using ESP-IDF v4.4 in the extension, the tools listed in your settings are those from the previous v4.3 which is why when ESP-IDF is trying to build is not using the expected xtensa toolchain. Did you used the extension setup wizard to install the new version? Maybe you have settings defined in your project's `.vscode/settings.json` overwriting the global settings of vscode ?

brianignacio5, 2023-01-31:
Closing this issue for lack of update from the user. Please comment here if you have any additional issues.

closed 2023-01-31 by brianignacio5