espressif/vscode-esp-idf-extension #866: ESP-IDF: Monitor QEMU device command not doing anything

Forever3402, 2023-01-05:
### OS

Linux

### Operating System version

https://hub.docker.com/r/espressif/idf

### Visual Studio Code version

1.74.2

### ESP-IDF version

5.1

### Python version

3.8.10

### Doctor command output

---------------------------------------------- ESP-IDF Extension for Visual Studio Code report ---------------------------------------------
OS linux x64 5.10.16.3-microsoft-standard-WSL2 
System environment variable IDF_PYTHON_ENV_PATH 
 /opt/esp/python_env/idf5.1_py3.8_env 
System environment variable PATH 
 /vscode/vscode-server/bin/linux-x64/e8a3071ea4344d9d48ef8a4df2c097372b0c5161/bin/remote-cli:/opt/esp/idf/components/esptool_py/esptool:/opt/esp/idf/components/espcoredump:/opt/esp/idf/components/partition_table:/opt/esp/idf/components/app_update:/opt/esp/tools/xtensa-esp-elf-gdb/12.1_20221002/xtensa-esp-elf-gdb/bin:/opt/esp/tools/riscv32-esp-elf-gdb/12.1_20221002/riscv32-esp-elf-gdb/bin:/opt/esp/tools/xtensa-esp32-elf/esp-2022r1-11.2.0/xtensa-esp32-elf/bin:/opt/esp/tools/xtensa-esp32s2-elf/esp-2022r1-11.2.0/xtensa-esp32s2-elf/bin:/opt/esp/tools/xtensa-esp32s3-elf/esp-2022r1-11.2.0/xtensa-esp32s3-elf/bin:/opt/esp/tools/riscv32-esp-elf/esp-2022r1-11.2.0/riscv32-esp-elf/bin:/opt/esp/tools/esp32ulp-elf/2.35_20220830/esp32ulp-elf/bin:/opt/esp/tools/openocd-esp32/v0.11.0-esp32-20221026/openocd-esp32/bin:/opt/esp/python_env/idf5.1_py3.8_env/bin:/opt/esp/idf/tools:/opt/qemu/bin:/opt/qemu/bin:/usr/local/sbin:/usr/local/bin:/usr/sbin:/usr/bin:/sbin:/bin 
System environment variable PYTHON 
 undefined 
Visual Studio Code version 1.74.2 
Visual Studio Code language en 
Visual Studio Code shell /usr/bin/bash 
ESP-IDF Extension version 1.5.1 
Workspace folder /workspaces 
---------------------------------------------------- Extension configuration settings ------------------------------------------------------
ESP-ADF Path (idf.espAdfPath) ${env:ADF_PATH}
ESP-IDF Path (idf.espIdfPath) /opt/esp/idf
ESP-MDF Path (idf.espMdfPath) ${env:MDF_PATH}
ESP-Matter Path (idf.espMatterPath) ${env:ESP_MATTER_PATH}
Custom extra paths (idf.customExtraPaths) 
Custom extra vars (idf.customExtraVars) {"OPENOCD_SCRIPTS":"/opt/esp/tools/openocd-esp32/v0.11.0-esp32-20221026/openocd-esp32/share/openocd/scripts","ESP_ROM_ELF_DIR":"/opt/esp/tools/esp-rom-elfs/20220823/"}
Virtual env Python Path (idf.pythonBinPath) /opt/esp/python_env/idf5.1_py3.8_env/bin/python
Serial port (idf.port) /dev/ttyUSB1
OpenOCD Configs (idf.openOcdConfigs) interface/ftdi/esp32_devkitj_v1.cfg,board/esp32-wrover.cfg
ESP-IDF Tools Path (idf.toolsPath) /opt/esp
Git Path (idf.gitPath) /usr/bin/git
-------------------------------------------------------- Configurations access -------------------------------------------------------------
Access to ESP-ADF Path (idf.espAdfPath) false
Access to ESP-IDF Path (idf.espIdfPath) true
Access to ESP-MDF Path (idf.espMdfPath) false
Access to ESP-Matter Path (idf.espMatterPath) false
Access to ESP-IDF Custom extra paths
Access to Virtual env Python Path (idf.pythonBinPath) true
Access to CMake in environment PATH true
Access to Ninja in environment PATH true
Access to ESP-IDF Tools Path (idf.toolsPath) true
-------------------------------------------------------- Configurations has spaces -------------------------------------------------------------
Spaces in system environment Path false
Spaces in ESP-ADF Path (idf.espAdfPath) false
Spaces in ESP-IDF Path (idf.espIdfPath) false
Spaces in ESP-MDF Path (idf.espMdfPath) false
Spaces in ESP-Matter Path (idf.espMatterPath) false
Spaces in ESP-IDF Custom extra paths
Spaces in Virtual env Python Path (idf.pythonBinPath) false
Spaces in ESP-IDF Tools Path (idf.toolsPath) false
----------------------------------------------------------- Executables Versions -----------------------------------------------------------
Git version 2.25.1
ESP-IDF version 5.1
Python version 3.8.10
[... 99 lines cut ...]
I have discovered the cause of the issue, and also found a workaround which would produce the expected result.

The cause of the issue is located at this function https://github.com/espressif/vscode-esp-idf-extension/blob/aa8be3ab686c33bb1c2b0d05de76ac0f19bc706a/src/extension.ts#L2961-L2975
At line 2973, the function awaits the result of `createNewIdfMonitor()`, but does not use the monitor object.
It is supposed to call `start()` on the monitor object like in https://github.com/espressif/vscode-esp-idf-extension/blob/aa8be3ab686c33bb1c2b0d05de76ac0f19bc706a/src/extension.ts#L3109-L3114

The workaround is to manually reconstruct the monitoring command and issue it to a new terminal accroding to https://github.com/espressif/vscode-esp-idf-extension/blob/aa8be3ab686c33bb1c2b0d05de76ac0f19bc706a/src/espIdf/monitor/index.ts#L62-L74
The exact command I used was `/opt/esp/python_env/idf5.1_py3.8_env/bin/python /opt/esp/idf/tools/idf_monitor.py -p socket://localhost:5555 -b 115200 --toolchain-prefix xtensa-esp32-elf- --target esp32 /workspaces/build/blink.elf`

This was very similar to the command issued by the "ESP-IDF: Monitor your device" command
`/opt/esp/python_env/idf5.1_py3.8_env/bin/python /opt/esp/idf/tools/idf_monitor.py -p /dev/ttyUSB1 -b 115200 --toolchain-prefix xtensa-esp32-elf- --target esp32 /workspaces/build/blink.elf` with the only difference being the port.

### Debug Message

```plain
None
```

### Other Steps to Reproduce

This issue was produced in Docker, but it should be irrelevant according to the souce code.

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

brianignacio5, 2023-01-06:
Could you test this [vsix installer](https://nightly.link/espressif/vscode-esp-idf-extension/actions/artifacts/496218227.zip) to see if the PR fix the issue ?

Forever3402, 2023-01-06:
I have tested your PR and it indeed fixes the issue, as well as let me interact with the monitor terminal, which I could not with my workaround.

closed 2023-01-09 by brianignacio5

muye998, 2023-03-06:
@brianignacio5 Hi, before begining, I am so sorry to ask you something maybe easy, but I tried many ways and spend much time to look for solution. So I will appreciate for your reply!

I also have some question in step7 on the [QUME guide](https://github.com/espressif/vscode-esp-idf-extension/blob/master/docs/QEMU.md), it successed to open the monitor windows in terminal, but it is only output `invalid header: 0x20746365`, what is the reason of it?
[image]

BTW, I think the guide is too simple to understand. Is there any more detailed guide documents about `QEMU`

brianignacio5, 2023-03-13:
Hi @muye998 yes indeed our QEMU only support esp32 for the moment, but we are working for supporting more chips soon!

muye998, 2023-03-16:
> Hi @muye998 yes indeed our QEMU only support esp32 for the moment, but we are working for supporting more chips soon!

thanks your reply，maybe if it will be supported in this year? @brianignacio5

PR #863 (2023-01-09): fix qemu monitor
## Description

Fix QEMU Monitor terminal init

Fix #866 

## Type of change

Please delete options that are not relevant.
- Bug fix (non-breaking change which fixes an issue)

## How has this been tested?

Manual testing of QEMU monitor start.

- QEMU start

**Test Configuration**:
* ESP-IDF Version: 5.1
* OS (Windows,Linux and macOS): macOS

## Checklist
- [x] PR Self Reviewed
- [x] Applied Code formatting
- [ ] Added Documentation
- [ ] Added Unit Test
- [x] Verified on all platforms - Windows,Linux and macOS