espressif/vscode-esp-idf-extension #872: I couldn't find the artifect following readme's "nightly build" part, could you help me with it?

fish1968, 2023-01-11:
### OS

Windows

### Operating System version

Windows10

### Visual Studio Code version

1.35

### ESP-IDF version

4.4.3

### Python version

3.9.13

### Doctor command output

No

### Extension

No

### Description

I clicked into https://github.com/espressif/vscode-esp-idf-extension/actions/workflows/release.yml and check the newest [v1.51 build](https://github.com/espressif/vscode-esp-idf-extension/actions/runs/3415946575), it tells me the build time is around 6mins, but at the artifacts part nothing available for me to download, a "-" shows there. 
Also, I want to know, do i need to use the nightly build version if I have already install the extension through vscode extension store? I guess not, right?

### Debug Message

```plain
NO
```

### Other Steps to Reproduce

I used chrome as my web browser.

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

brianignacio5, 2023-01-13:
Which version of this extension are you trying to use?

If you want to use the latest release you could just use the vscode marketplace version (which is updated to the latest release) or the Releases pages in this repository. The nightly build mentioned in the install tutorial is to install a vsix installer from the current master or any pull request in this repository.

We have yet to implement a stable nightly build mechanism.

fish1968, 2023-01-13:
I see, thanks for your reply.

closed 2023-01-13 by fish1968

brianignacio5, 2023-01-20:
I added some links in the PR #874 which should add links for the current GitHub master and PRs artifact.