espressif/vscode-esp-idf-extension #875: Cannot debug using ESP32 and JLINK

eme-wesleywong, 2023-01-18:
### OS

Linux

### Operating System version

Ubuntu 22.04

### Visual Studio Code version

1.74.3

### ESP-IDF version

4.4.3

### Python version

3.8.7

### Doctor command output

---------------------------------------------- ESP-IDF Extension for Visual Studio Code report ---------------------------------------------
OS linux x64 5.15.0-43-generic 
System environment variable IDF_PYTHON_ENV_PATH 
 undefined 
System environment variable PATH 
 <HOMEPATH>/.local/bin:<HOMEPATH>/.local/bin:/usr/local/sbin:/usr/local/bin:/usr/sbin:/usr/bin:/sbin:/bin:/usr/games:/usr/local/games:/snap/bin:/snap/bin 
System environment variable PYTHON 
 undefined 
Visual Studio Code version 1.74.3 
Visual Studio Code language en 
Visual Studio Code shell /usr/bin/bash 
ESP-IDF Extension version 1.5.1 
Workspace folder <HOMEPATH>/esp/workspace/blink 
---------------------------------------------------- Extension configuration settings ------------------------------------------------------
ESP-ADF Path (idf.espAdfPath) ${env:ADF_PATH}
ESP-IDF Path (idf.espIdfPath) <HOMEPATH>/esp/esp-idf
ESP-MDF Path (idf.espMdfPath) ${env:MDF_PATH}
ESP-Matter Path (idf.espMatterPath) ${env:ESP_MATTER_PATH}
Custom extra paths (idf.customExtraPaths) <HOMEPATH>/.espressif/tools/xtensa-esp32-elf/esp-2021r2-patch5-8.4.0/xtensa-esp32-elf/bin:<HOMEPATH>/.espressif/tools/xtensa-esp32s2-elf/esp-2021r2-patch5-8.4.0/xtensa-esp32s2-elf/bin:<HOMEPATH>/.espressif/tools/xtensa-esp32s3-elf/esp-2021r2-patch5-8.4.0/xtensa-esp32s3-elf/bin:<HOMEPATH>/.espressif/tools/riscv32-esp-elf/esp-2021r2-patch5-8.4.0/riscv32-esp-elf/bin:<HOMEPATH>/.espressif/tools/esp32ulp-elf/2.35_20220830/esp32ulp-elf/bin:<HOMEPATH>/.espressif/tools/openocd-esp32/v0.11.0-esp32-20220706/openocd-esp32/bin
Custom extra vars (idf.customExtraVars) {"OPENOCD_SCRIPTS":"<HOMEPATH>/.espressif/tools/openocd-esp32/v0.11.0-esp32-20220706/openocd-esp32/share/openocd/scripts"}
Virtual env Python Path (idf.pythonBinPath) <HOMEPATH>/.espressif/python_env/idf4.4_py3.10_env/bin/python
Serial port (idf.port) /dev/ttyUSB0
OpenOCD Configs (idf.openOcdConfigs) interface/jlink.cfg,target/esp32.cfg
ESP-IDF Tools Path (idf.toolsPath) <HOMEPATH>/.espressif
Git Path (idf.gitPath) /usr/bin/git
-------------------------------------------------------- Configurations access -------------------------------------------------------------
Access to ESP-ADF Path (idf.espAdfPath) false
Access to ESP-IDF Path (idf.espIdfPath) true
Access to ESP-MDF Path (idf.espMdfPath) false
Access to ESP-Matter Path (idf.espMatterPath) false
Access to ESP-IDF Custom extra paths
Access to <HOMEPATH>/.espressif/tools/xtensa-esp32-elf/esp-2021r2-patch5-8.4.0/xtensa-esp32-elf/bin: true
Access to <HOMEPATH>/.espressif/tools/xtensa-esp32s2-elf/esp-2021r2-patch5-8.4.0/xtensa-esp32s2-elf/bin: true
Access to <HOMEPATH>/.espressif/tools/xtensa-esp32s3-elf/esp-2021r2-patch5-8.4.0/xtensa-esp32s3-elf/bin: true
Access to <HOMEPATH>/.espressif/tools/riscv32-esp-elf/esp-2021r2-patch5-8.4.0/riscv32-esp-elf/bin: true
Access to <HOMEPATH>/.espressif/tools/esp32ulp-elf/2.35_20220830/esp32ulp-elf/bin: true
Access to <HOMEPATH>/.espressif/tools/openocd-esp32/v0.11.0-esp32-20220706/openocd-esp32/bin: true
Access to Virtual env Python Path (idf.pythonBinPath) true
Access to CMake in environment PATH true
Access to Ninja in environment PATH true
Access to ESP-IDF Tools Path (idf.toolsPath) true
-------------------------------------------------------- Configurations has spaces -------------------------------------------------------------
Spaces in system environment Path false
Spaces in ESP-ADF Path (idf.espAdfPath) false
Spaces in ESP-IDF Path (idf.espIdfPath) false
Spaces in ESP-MDF Path (idf.espMdfPath) false
Spaces in ESP-Matter Path (idf.espMatterPath) false
Spaces in ESP-IDF Custom extra paths
[... 567 lines cut ...]
2023-01-18 15:10:11,547 - Debug Adapter (WriterThread) - DEBUG - Writing: {"type": "event", "event": "output", "body": {"output": "Stopping target poller\n", "source": {}}, "seq": 26}

2023-01-18 15:10:11,547 - Gdb - DEBUG - RESULT: exit None
2023-01-18 15:10:11,547 - Debug Adapter (WriterThread) - DEBUG - Writing: {"type": "event", "event": "output", "body": {"output": "Closing the GDB process\n", "source": {}}, "seq": 27}

[Debug Adapter]
2023-01-18 15:10:11,666 - Debug Adapter (main) - DEBUG - Closing the Socket
[Debug Adapter]
2023-01-18 15:10:11,666 - Debug Adapter (WriterThread) - DEBUG - Writing: {"type": "event", "event": "output", "body": {"output": "Closing the Socket\n", "source": {}}, "seq": 28}

2023-01-18 15:10:11,666 - Debug Adapter (main) - DEBUG - Stopping of the Writer thread
2023-01-18 15:10:11,667 - Debug Adapter (main) - DEBUG - Stopping of the Reader thread
[Debug Adapter]
2023-01-18 15:10:11,667 - Debug Adapter (main) - DEBUG - Adapter is Stopped
DEBUG_ADAPTER_STOPPED
2023-01-18 15:10:11,667 - Debug Adapter (main) - CRITICAL - Debug adapter -> Extension: DEBUG_ADAPTER_STOPPED
2023-01-18 15:10:11,667 - Debug Adapter(Command Processor) - DEBUG - Processed command: disconnect
[Stopped] : ESP-IDF Debug Adapter
[OpenOCD]
[Stopped] : OpenOCD Server
```

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

brianignacio5, 2023-02-03:
It seems you already had a connection working before.

`Info : rejected 'gdb' connection, no more connections allowed`

eme-wesleywong, 2023-02-23:
@brianignacio5 ,
There is no other connection.
Below is new launch.json that I tried.
```
{
  "version": "0.2.0",
  "configurations": [

    {
      "type": "espidf",
      "name": "Launch",
      "request": "launch",
      "debugPort": 9998,
      "logLevel": 2,
      "mode": "auto",
      "gdbinitFile": "${workspaceFolder}/gdbinit",
      "verifyAppBinBeforeDebug": false,
      "tmoScaleFactor": 1,
      "env": {
        "CUSTOM_ENV_VAR": "SOME_VALUE"
      }
    }
  ]
}
```

gdbinit
```
target extended-remote :3333
set remote hardware-watchpoint-limit 2
mon reset halt
flushregs
thb app_main
c
```

If verifyAppBinBeforeDebug is false, then I got below error message.
```
[OpenOCD]
Open On-Chip Debugger  v0.11.0-esp32-20220706 (2022-07-06-15:48)
Licensed under GNU GPL v2
For bug reports, read
	http://openocd.org/doc/doxygen/bugs.html
debug_level: 2

Info : only one transport option; autoselect 'jtag'
Info : esp_usb_jtag: VID set to 0x303a and PID to 0x1001
Info : esp_usb_jtag: capabilities descriptor set to 0x2000
Warn : Transport "jtag" was already selected
Info : Listening on port 6666 for tcl connections
Info : Listening on port 4444 for telnet connections
[OpenOCD]
Info : esp_usb_jtag: serial (68:B6:B3:3B:AD:EC)
[OpenOCD]
Info : esp_usb_jtag: Device found. Base speed 40000KHz, div range 1 to 255
Info : clock speed 10000 kHz
Info : JTAG tap: esp32s3.cpu0 tap/device found: 0x120034e5 (mfg: 0x272 (Tensilica), part: 0x2003, ver: 0x1)
Info : JTAG tap: esp32s3.cpu1 tap/device found: 0x120034e5 (mfg: 0x272 (Tensilica), part: 0x2003, ver: 0x1)
[OpenOCD]
Info : [esp32s3.cpu0] Target halted, PC=0x4201805A, debug_reason=00000001
[OpenOCD]
Info : [esp32s3.cpu1] Target halted, PC=0x4201805A, debug_reason=00000000
[OpenOCD]
Info : starting gdb server for esp32s3.cpu0 on 3333
Info : Listening on port 3333 for gdb connections
[Debug Adapter]
DEBUG_ADAPTER_STARTED
2023-02-23 11:38:36,953 - Debug Adapter (main) - CRITICAL - Debug adapter -> Extension: DEBUG_ADAPTER_STARTED
[Debug Adapter]
DEBUG_ADAPTER_READY2CONNECT
2023-02-23 11:38:36,954 - Debug Adapter (main) - CRITICAL - Debug adapter -> Extension: DEBUG_ADAPTER_READY2CONNECT
[OpenOCD]
Info : accepting 'gdb' connection on tcp/3333
[OpenOCD]
Warn : No symbols for FreeRTOS!
[OpenOCD]
Info : [esp32s3.cpu0] Target halted, PC=0x403B2482, debug_reason=00000001
[OpenOCD]
Info : Set GDB target to 'esp32s3.cpu0'
[OpenOCD]
Info : Flash mapping 0: 0x10020 -> 0x3c020020, 34 KB
Info : Flash mapping 1: 0x20020 -> 0x42000020, 99 KB
[OpenOCD]
Info : [esp32s3.cpu0] Target halted, PC=0x403B2482, debug_reason=00000001
[OpenOCD]
Info : Auto-detected flash bank 'esp32s3.cpu0.flash' size 32768 KB
Info : Using flash bank 'esp32s3.cpu0.flash' size 32768 KB
[OpenOCD]
Info : [esp32s3.cpu0] Target halted, PC=0x403B2482, debug_reason=00000001
[OpenOCD]
Info : Flash mapping 0: 0x10020 -> 0x3c020020, 34 KB
Info : Flash mapping 1: 0x20020 -> 0x42000020, 99 KB
Info : Using flash bank 'esp32s3.cpu0.irom' size 100 KB
[OpenOCD]
Info : [esp32s3.cpu0] Target halted, PC=0x403B2482, debug_reason=00000001
[OpenOCD]
Info : Flash mapping 0: 0x10020 -> 0x3c020020, 34 KB
Info : Flash mapping 1: 0x20020 -> 0x42000020, 99 KB
Info : Using flash bank 'esp32s3.cpu0.drom' size 36 KB
Info : New GDB Connection: 1, Target esp32s3.cpu0, state: halted
[OpenOCD]
Warn : Prefer GDB command "target extended-remote :3333" instead of "target remote :3333"
[OpenOCD]
Info : rejected 'gdb' connection, no more connections allowed
Info : dropped 'gdb' connection
[Debug Adapter]
2023-02-23 11:38:44,457 - Debug Adapter (main) - ERROR - int() argument must be a string, a bytes-like object or a real number, not 'NoneType'
Traceback (most recent call last):
  File "/home/wesleywong/.vscode/extensions/espressif.esp-idf-extension-1.5.1/esp_debug_adapter/debug_adapter/command_processor.py", line 144, in on_launch_request
    self.da.run(start=(not self.da.state.no_debug))
  File "/home/wesleywong/.vscode/extensions/espressif.esp-idf-extension-1.5.1/esp_debug_adapter/debug_adapter/debug_adapter.py", line 902, in run
    self.on_target_stopped(rsn)
  File "/home/wesleywong/.vscode/extensions/espressif.esp-idf-extension-1.5.1/esp_debug_adapter/debug_adapter/debug_adapter.py", line 704, in on_target_stopped
    self.update_threads()
  File "/home/wesleywong/.vscode/extensions/espressif.esp-idf-extension-1.5.1/esp_debug_adapter/debug_adapter/debug_adapter.py", line 765, in update_threads
    self.thread_selected = int(gdb_resp[0])
TypeError: int() argument must be a string, a bytes-like object or a real number, not 'NoneType'
[Debug Adapter]
DEBUG_ADAPTER_STOPPED
2023-02-23 11:38:44,618 - Debug Adapter (main) - CRITICAL - Debug adapter -> Extension: DEBUG_ADAPTER_STOPPED
[Stopped] : ESP-IDF Debug Adapter
[OpenOCD]
[Stopped] : OpenOCD Server
```

But if I set verifyAppBinBeforeDebug to true, then the debug session would be stopped just after starting openocd.
The error message is "Cannot read properties of undefined (reading 'message')".

Again, I can debug normally using command line windows.

brianignacio5, 2023-02-23:
I'm a bit confused because now the esp32 is esp32s3 and only seeing the OpenOCD output not the debug adapter output. Sure there is an error on the `gdb_resp[0]` but maybe we need to increase the logLevel of debug adapter and openOCD to see the issue. Also what is the settings for this new target and is the code the same ? Steps to reproduce?

Also could you try the latest extension installer GitHub release in master [here](https://nightly.link/espressif/vscode-esp-idf-extension/workflows/ci/master/esp-idf-extension.vsix.zip) ? We have made some changes to debug adapter recently.

eme-wesleywong, 2023-02-24:
@brianignacio5 
I manage to get it work. I am not sure what is the cause of this issue.
I deleted ""gdbinitFile": "${workspaceFolder}/gdbinit"," in launch.json and then I can start debugging.
Here is the log of successful debugging with loglevel 3.
```
[OpenOCD]
Open On-Chip Debugger  v0.11.0-esp32-20220706 (2022-07-06-15:48)
Licensed under GNU GPL v2
For bug reports, read
	http://openocd.org/doc/doxygen/bugs.html
debug_level: 2

Info : only one transport option; autoselect 'jtag'
Info : esp_usb_jtag: VID set to 0x303a and PID to 0x1001
Info : esp_usb_jtag: capabilities descriptor set to 0x2000
Warn : Transport "jtag" was already selected
Info : Listening on port 6666 for tcl connections
Info : Listening on port 4444 for telnet connections
[OpenOCD]
Info : esp_usb_jtag: serial (68:B6:B3:3B:AD:EC)
Info : esp_usb_jtag: Device found. Base speed 40000KHz, div range 1 to 255
[OpenOCD]
Info : clock speed 10000 kHz
[OpenOCD]
Info : JTAG tap: esp32s3.cpu0 tap/device found: 0x120034e5 (mfg: 0x272 (Tensilica), part: 0x2003, ver: 0x1)
Info : JTAG tap: esp32s3.cpu1 tap/device found: 0x120034e5 (mfg: 0x272 (Tensilica), part: 0x2003, ver: 0x1)
[OpenOCD]
Info : [esp32s3.cpu0] Debug controller was reset.
[OpenOCD]
Info : [esp32s3.cpu0] Core was reset.
[OpenOCD]
Info : [esp32s3.cpu1] Debug controller was reset.
[OpenOCD]
Info : [esp32s3.cpu1] Core was reset.
[OpenOCD]
Info : starting gdb server for esp32s3.cpu0 on 3333
Info : Listening on port 3333 for gdb connections
[Debug Adapter]
2023-02-24 14:09:19,488 - Debug Adapter (main) - INFO - Working directory: /home/wesleywong
[Debug Adapter]
2023-02-24 14:09:19,490 - Debug Adapter (main) - INFO - Arguments: 
{   'app_flash_off': 65536,
    'board_type': None,
    'cmdfile': None,
    'core_file': (),
    'debug': 3,
    'developer_mode': 'none',
    'device_name': 'esp32s3',
    'elfpath': ('/home/wesleywong/esp/workspace/blink/build/blink.elf',),
    'log_file': '/home/wesleywong/esp/workspace/blink/debug.log',
    'log_mult_files': None,
    'log_no_debug_console': None,
    'oocd': 'openocd',
    'oocd_args': None,
    'oocd_ip': 'localhost',
    'oocd_mode': 'connect_to_instance',
    'oocd_scripts': None,
    'port': 9998,
    'postmortem': False,
    'tmo_scale_factor': 1,
    'toolchain_prefix': 'xtensa-esp32s3-elf-'}
[Debug Adapter]
DEBUG_ADAPTER_STARTED
2023-02-24 14:09:19,490 - Debug Adapter (main) - CRITICAL - Debug adapter -> Extension: DEBUG_ADAPTER_STARTED
2023-02-24 14:09:19,491 - Debug Adapter (main) - INFO - Starting. Cmd: /home/wesleywong/.vscode/extensions/espressif.esp-idf-extension-1.5.1/esp_debug_adapter/debug_adapter_main.py -d 3 -e /home/wesleywong/esp/workspace/blink/build/blink.elf -l /home/wesleywong/esp/workspace/blink/debug.log -p 9998 -dn esp32s3 -a 0x10000 -t xtensa-esp32s3-elf- -tsf 1

2023-02-24 14:09:19,491 - Debug Adapter (main) - INFO - Listening on port 9998
DEBUG_ADAPTER_READY2CONNECT
2023-02-24 14:09:19,491 - Debug Adapter (main) - CRITICAL - Debug adapter -> Extension: DEBUG_ADAPTER_READY2CONNECT
[Debug Adapter]
[... 297 lines cut ...]
 'thread-id': '1'}
[OpenOCD]
Info : rejected 'gdb' connection, no more connections allowed
Info : dropped 'gdb' connection
[Debug Adapter]
2023-02-24 14:11:20,569 - Gdb - INFO - NOTIFY: thread-group-exited {'id': 'i1'}
[Debug Adapter]
2023-02-24 14:11:21,071 - Debug Adapter (main) - ERROR - int() argument must be a string, a bytes-like object or a real number, not 'NoneType'
Traceback (most recent call last):
  File "/home/wesleywong/.vscode/extensions/espressif.esp-idf-extension-1.5.1/esp_debug_adapter/debug_adapter/command_processor.py", line 144, in on_launch_request
    self.da.run(start=(not self.da.state.no_debug))
  File "/home/wesleywong/.vscode/extensions/espressif.esp-idf-extension-1.5.1/esp_debug_adapter/debug_adapter/debug_adapter.py", line 902, in run
    self.on_target_stopped(rsn)
  File "/home/wesleywong/.vscode/extensions/espressif.esp-idf-extension-1.5.1/esp_debug_adapter/debug_adapter/debug_adapter.py", line 704, in on_target_stopped
    self.update_threads()
  File "/home/wesleywong/.vscode/extensions/espressif.esp-idf-extension-1.5.1/esp_debug_adapter/debug_adapter/debug_adapter.py", line 765, in update_threads
    self.thread_selected = int(gdb_resp[0])
TypeError: int() argument must be a string, a bytes-like object or a real number, not 'NoneType'
[Debug Adapter]
DEBUG_ADAPTER_STOPPED
2023-02-24 14:11:21,217 - Debug Adapter (main) - CRITICAL - Debug adapter -> Extension: DEBUG_ADAPTER_STOPPED
[Stopped] : ESP-IDF Debug Adapter
[OpenOCD]
[Stopped] : OpenOCD Server
```

brianignacio5, 2023-02-24:
Good that it is working @eme-wesleywong !

Probably some commands in gdbinit was not properly working. We will add more validation in the Debug Adapter to provide more details about errors here.

closed 2023-02-24 by brianignacio5