espressif/vscode-esp-idf-extension #876: ESP-IDF : Failed to init Debug Adapter!

FnxQT, 2023-01-21:
### OS

Windows

### Operating System version

Windows 11

### Visual Studio Code version

1.74.3

### ESP-IDF version

5

### Python version

3.10.8

### Doctor command output

---------------------------------------------- ESP-IDF Extension for Visual Studio Code report ---------------------------------------------
OS win32 x64 10.0.22621 
System environment variable IDF_PYTHON_ENV_PATH 
 undefined 
System environment variable PATH 
 C:\Testwell\CTC;C:\Program Files\Common Files\Oracle\Java\javapath;C:\Program Files (x86)\VMware\VMware Workstation\bin\;C:\Program Files (x86)\Common Files\Oracle\Java\javapath;C:\WINDOWS\system32;C:\WINDOWS;C:\WINDOWS\System32\Wbem;C:\WINDOWS\System32\WindowsPowerShell\v1.0\;C:\WINDOWS\System32\OpenSSH\;C:\Program Files\Intel\WiFi\bin\;C:\Program Files\Common Files\Intel\WirelessCommon\;c:\Program Files (x86)\Intel\Intel(R) Management Engine Components\DAL;c:\Program Files\Intel\Intel(R) Management Engine Components\DAL;C:\Program Files\dotnet\;C:\Program Files\usbipd-win\;C:\msys64\mingw64\bin;C:\Program Files\squishcoco;C:\Program Files (x86)\STMicroelectronics\STM32 ST-LINK Utility\ST-LINK Utility;C:<HOMEPATH>\AppData\Local\Programs\Python\Python310\Scripts\;C:<HOMEPATH>\AppData\Local\Programs\Python\Python310\;C:<HOMEPATH>\AppData\Local\Microsoft\WindowsApps;C:<HOMEPATH>\AppData\Local\Programs\Microsoft VS Code\bin;C:\OpenSSL\bin;C:\Program Files\SciTools\bin\pc-win64 
System environment variable PYTHON 
 undefined 
Visual Studio Code version 1.74.3 
Visual Studio Code language en 
Visual Studio Code shell C:\WINDOWS\System32\WindowsPowerShell\v1.0\powershell.exe 
ESP-IDF Extension version 1.5.1 
Workspace folder c:<HOMEPATH>\ESP32_Projects\Blinky\blink 
---------------------------------------------------- Extension configuration settings ------------------------------------------------------
ESP-ADF Path (idf.espAdfPath) ${env:ADF_PATH}
ESP-IDF Path (idf.espIdfPath) C:<HOMEPATH>\esp\esp-idf
ESP-MDF Path (idf.espMdfPath) ${env:MDF_PATH}
ESP-Matter Path (idf.espMatterPath) ${env:ESP_MATTER_PATH}
Custom extra paths (idf.customExtraPaths) C:<HOMEPATH>\.espressif\tools\xtensa-esp-elf-gdb\11.2_20220823\xtensa-esp-elf-gdb\bin;C:<HOMEPATH>\.espressif\tools\riscv32-esp-elf-gdb\11.2_20220823\riscv32-esp-elf-gdb\bin;C:<HOMEPATH>\.espressif\tools\xtensa-esp32-elf\esp-2022r1-11.2.0\xtensa-esp32-elf\bin;C:<HOMEPATH>\.espressif\tools\xtensa-esp32s2-elf\esp-2022r1-11.2.0\xtensa-esp32s2-elf\bin;C:<HOMEPATH>\.espressif\tools\xtensa-esp32s3-elf\esp-2022r1-11.2.0\xtensa-esp32s3-elf\bin;C:<HOMEPATH>\.espressif\tools\riscv32-esp-elf\esp-2022r1-11.2.0\riscv32-esp-elf\bin;C:<HOMEPATH>\.espressif\tools\esp32ulp-elf\2.35_20220830\esp32ulp-elf\bin;C:<HOMEPATH>\.espressif\tools\cmake\3.24.0\bin;C:<HOMEPATH>\.espressif\tools\openocd-esp32\v0.11.0-esp32-20221026\openocd-esp32\bin;C:<HOMEPATH>\.espressif\tools\ninja\1.10.2;C:<HOMEPATH>\.espressif\tools\idf-exe\1.0.3;C:<HOMEPATH>\.espressif\tools\ccache\4.6.2\ccache-4.6.2-windows-x86_64;C:<HOMEPATH>\.espressif\tools\dfu-util\0.9\dfu-util-0.9-win64
Custom extra vars (idf.customExtraVars) {"OPENOCD_SCRIPTS":"C:\\Users\\Quentin\\.espressif\\tools\\openocd-esp32\\v0.11.0-esp32-20221026/openocd-esp32/share/openocd/scripts","IDF_CCACHE_ENABLE":"1"}
Virtual env Python Path (idf.pythonBinPath) C:<HOMEPATH>\.espressif\python_env\idf5.0_py3.8_env\Scripts\python.exe
Serial port (idf.port) COM5
OpenOCD Configs (idf.openOcdConfigs) board/esp32-wrover-kit-3.3v.cfg
ESP-IDF Tools Path (idf.toolsPath) C:<HOMEPATH>\.espressif
Git Path (idf.gitPath) C:<HOMEPATH>\.espressif\tools\idf-git\2.30.1\cmd\git.exe
-------------------------------------------------------- Configurations access -------------------------------------------------------------
Access to ESP-ADF Path (idf.espAdfPath) false
Access to ESP-IDF Path (idf.espIdfPath) true
Access to ESP-MDF Path (idf.espMdfPath) false
Access to ESP-Matter Path (idf.espMatterPath) false
Access to ESP-IDF Custom extra paths
Access to C:<HOMEPATH>\.espressif\tools\xtensa-esp-elf-gdb\11.2_20220823\xtensa-esp-elf-gdb\bin: true
Access to C:<HOMEPATH>\.espressif\tools\riscv32-esp-elf-gdb\11.2_20220823\riscv32-esp-elf-gdb\bin: true
Access to C:<HOMEPATH>\.espressif\tools\xtensa-esp32-elf\esp-2022r1-11.2.0\xtensa-esp32-elf\bin: true
Access to C:<HOMEPATH>\.espressif\tools\xtensa-esp32s2-elf\esp-2022r1-11.2.0\xtensa-esp32s2-elf\bin: true
Access to C:<HOMEPATH>\.espressif\tools\xtensa-esp32s3-elf\esp-2022r1-11.2.0\xtensa-esp32s3-elf\bin: true
Access to C:<HOMEPATH>\.espressif\tools\riscv32-esp-elf\esp-2022r1-11.2.0\riscv32-esp-elf\bin: true
Access to C:<HOMEPATH>\.espressif\tools\esp32ulp-elf\2.35_20220830\esp32ulp-elf\bin: true
Access to C:<HOMEPATH>\.espressif\tools\cmake\3.24.0\bin: true
Access to C:<HOMEPATH>\.espressif\tools\openocd-esp32\v0.11.0-esp32-20221026\openocd-esp32\bin: true
Access to C:<HOMEPATH>\.espressif\tools\ninja\1.10.2: true
Access to C:<HOMEPATH>\.espressif\tools\idf-exe\1.0.3: true
Access to C:<HOMEPATH>\.espressif\tools\ccache\4.6.2\ccache-4.6.2-windows-x86_64: true
Access to C:<HOMEPATH>\.espressif\tools\dfu-util\0.9\dfu-util-0.9-win64: true
Access to Virtual env Python Path (idf.pythonBinPath) true
Access to CMake in environment PATH undefined
Access to Ninja in environment PATH undefined
Access to ESP-IDF Tools Path (idf.toolsPath) true
[... 144 lines cut ...]
        "clientID": "vscode",
        "clientName": "Visual Studio Code",
        "columnsStartAt1": true,
        "linesStartAt1": true,
        "locale": "en-us",
        "pathFormat": "path",
        "supportsArgsCanBeInterpretedByShell": true,
        "supportsInvalidatedEvent": true,
        "supportsMemoryEvent": true,
        "supportsMemoryReferences": true,
        "supportsProgressReporting": true,
        "supportsRunInTerminalRequest": true,
        "supportsVariablePaging": true,
        "supportsVariableType": true
    },
    "command": "initialize",
    "seq": 1,
    "type": "request"
}
Created gdb object of type: GdbEsp32
```

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

FnxQT, 2023-01-21:
OPENOCD LOGLEVEL4 OUTPUT:

[OpenOCD]
Open On-Chip Debugger v0.11.0-esp32-20221026 (2022-10-26-14:48)
Licensed under GNU GPL v2
For bug reports, read
	http://openocd.org/doc/doxygen/bugs.html
[OpenOCD]
debug_level: 2

[OpenOCD]
Info : Listening o[OpenOCD]
n port 6666 for tcl connections
Info : Listening on port 4444 for telnet conn[OpenOCD]
ections
[OpenOCD]
❌ Error: no device foun[OpenOCD]
❌ d
Error: unable to open ftdi device with vid 0403, pid 6010, description '*', serial '*' at bus location '*'
[OpenOCD]
❌ Error: no device found
Error: un[OpenOCD]
able to open ftdi device with vid 0403, pid 6014, description '*', serial '*' at bus location '*'
[OpenOCD]

[OpenOCD]
[Stopped] : OpenOCD Server
[Debug Adapter]
2023-01-21 15:35:30,821 - Debug Adapter (main) - DEBUG - START_TIME_2023_01_21_15-35-30
2023-01-21 15:35:30,821 - Debug Adapter (main) - DEBUG - Init errors:None
2023-01-21 15:35:30,821 - Debug Adapter (main) - INFO - Working directory: C:\Users\Quentin\AppData\Local\Programs\Microsoft VS Code
2023-01-21 15:35:30,822 - Debug Adapter (main) - INFO - Arguments: 
{   'app_flash_off': 65536,
    'board_type': None,
    'cmdfile': None,
    'core_file': (),
    'debug': 4,
    'developer_mode': 'none',
    'device_name': 'esp32',
    'elfpath': (   'c:\\Users\\Quentin\\ESP32_Projects\\Blinky\\blink\\build\\blink.elf',),
    'log_file': 'c:\\Users\\Quentin\\ESP32_Projects\\Blinky\\blink\\debug.log',
    'log_mult_files': None,
    'log_no_debug_console': None,
    'oocd': 'openocd',
    'oocd_args': None,
    'oocd_ip': 'localhost',
    'oocd_mode': 'connect_to_instance',
    'oocd_scripts': None,
    'port': 43474,
    'postmortem': False,
    'tmo_scale_factor': 1,
    'toolchain_prefix': 'xtensa-esp32-elf-'}
DEBUG_ADAPTER_STARTED
2023-01-21 15:35:30,822 - Debug Adapter (main) - CRITICAL - Debug adapter -> Extension: DEBUG_ADAPTER_STARTED
2023-01-21 15:35:30,822 - Debug Adapter (main) - INFO - Starting. Cmd: c:\Users\Quentin\.vscode\extensions\espressif.esp-idf-extension-1.5.1\esp_debug_adapter\debug_adapter_main.py -d 4 -e c:\Users\Quentin\ESP32_Projects\Blinky\blink\build\blink.elf -l c:\Users\Quentin\ESP32_Projects\Blinky\blink\debug.log -p 43474 -dn esp32 -a 0x10000 -t xtensa-esp32-elf-

[Debug Adapter]
2023-01-21 15:35:30,826 - Debug Adapter (main) - INFO - Listening on port 43474
DEBUG_ADAPTER_READY2CONNECT
2023-01-21 15:35:30,827 - Debug Adapter (main) - CRITICAL - Debug adapter -> Extension: DEBUG_ADAPTER_READY2CONNECT
[Debug Adapter]
2023-01-21 15:35:30,831 - Debug Adapter (main) - DEBUG - Got connection
2023-01-21 15:35:30,833 - Debug Adapter (WriterThread) - DEBUG - Writing: {"type": "event", "event": "output", "body": {"output": "Debug adapter -> Extension: DEBUG_ADAPTER_STARTED\n", "source": {}}, "seq": 0}

2023-01-21 15:35:30,834 - Debug Adapter (WriterThread) - DEBUG - Writing: {"type": "event", "event": "output", "body": {"output": "Starting. Cmd: c:\\Users\\Quentin\\.vscode\\extensions\\espressif.esp-idf-extension-1.5.1\\esp_debug_adapter\\debug_adapter_main.py -d 4 -e c:\\Users\\Quentin\\ESP32_Projects\\Blinky\\blink\\build\\blink.elf -l c:\\Users\\Quentin\\ESP32_Projects\\Blinky\\blink\\debug.log -p 43474 -dn esp32 -a 0x10000 -t xtensa-esp32-elf-\n\n", "source": {}}, "seq": 1}

2023-01-21 15:35:30,834 - Debug Adapter (WriterThread) - DEBUG - Writing: {"type": "event", "event": "output", "body": {"output": "Listening on port 43474\n", "source": {}}, "seq": 2}

2023-01-21 15:35:30,834 - Debug Adapter (WriterThread) - DEBUG - Writing: {"type": "event", "event": "output", "body": {"output": "Debug adapter -> Extension: DEBUG_ADAPTER_READY2CONNECT\n", "source": {}}, "seq": 3}

[... 120 lines cut ...]
2023-01-21 15:35:36,269 - Debug Adapter (WriterThread) - DEBUG - Writing: {"type": "event", "event": "output", "body": {"output": "Got json: {\n    \"arguments\": {\n        \"restart\": false\n    },\n    \"command\": \"disconnect\",\n    \"seq\": 2,\n    \"type\": \"request\"\n}\n", "source": {}}, "seq": 10}

2023-01-21 15:35:36,269 - Debug Adapter (main) - DEBUG - Adapter is Stopping...
[Debug Adapter]
2023-01-21 15:35:36,272 - Debug Adapter (WriterThread) - DEBUG - Writing: {"type": "event", "event": "output", "body": {"output": "Debug Adapter stopped\n", "source": {}}, "seq": 11}

2023-01-21 15:35:36,273 - Debug Adapter (main) - DEBUG - Stopping target poller
2023-01-21 15:35:36,274 - Debug Adapter (main) - DEBUG - Closing the GDB process
2023-01-21 15:35:36,274 - Debug Adapter (WriterThread) - DEBUG - Writing: {"type": "event", "event": "output", "body": {"output": "Adapter is Stopping...\n", "source": {}}, "seq": 12}

[Debug Adapter]
2023-01-21 15:35:36,275 - Debug Adapter (main) - DEBUG - Closing the Socket
[Debug Adapter]
2023-01-21 15:35:36,276 - Debug Adapter (WriterThread) - DEBUG - Writing: {"type": "event", "event": "output", "body": {"output": "Stopping target poller\n", "source": {}}, "seq": 13}

2023-01-21 15:35:36,276 - Debug Adapter (main) - DEBUG - Stopping of the Writer thread
2023-01-21 15:35:36,277 - Debug Adapter (WriterThread) - DEBUG - Writing: {"type": "event", "event": "output", "body": {"output": "Closing the GDB process\n", "source": {}}, "seq": 14}

2023-01-21 15:35:36,277 - Debug Adapter (main) - DEBUG - Stopping of the Reader thread
[Debug Adapter]
2023-01-21 15:35:36,277 - Debug Adapter (main) - DEBUG - Adapter is Stopped
DEBUG_ADAPTER_STOPPED
2023-01-21 15:35:36,277 - Debug Adapter (main) - CRITICAL - Debug adapter -> Extension: DEBUG_ADAPTER_STOPPED
2023-01-21 15:35:36,278 - Debug Adapter(Command Processor) - DEBUG - Processed command: disconnect
[Stopped] : ESP-IDF Debug Adapter

FnxQT, 2023-01-29:
Hello,

Yes I'm able to build, flash and monitor the device without problem using the Blinky example provided in the ESP-IDF.
I'm using an ESP-WROOM-32 / ESP32 DEVKITV1

brianignacio5, 2023-01-31:
I think the issue is your openOCD board configuration. Based on your doctor command, you are using the wrover kit one. Try using the esp32 for devkitv1.

For reference, here is a list of [openOCD boards](https://github.com/espressif/openocd-esp32/blob/master/tcl/esp-config.json) you can select with the `ESP-IDF: Select OpenOCD Board Configuration` command.

closed 2023-02-14 by brianignacio5