espressif/vscode-esp-idf-extension #878: ESP-IDF: Install ESP-Matter command error

eos1d3, 2023-01-23:
### OS

MacOS

### Operating System version

macOS 12.6.3

### Visual Studio Code version

1.74.3

### ESP-IDF version

4.4.3

### Python version

3.10.9

### Doctor command output

na

### Description

Install ESP-Matter command error. It is caused by incomplete installation of pkg-config & openssl on Intel Mac.

The installation of the package is different from M1/M2 Mac.

### Debug Message

```plain
WELCOME TO...

         █
         █
     ▄   █   ▄                                █     █
     ▀▀█████▀▀      ▄▀▀▀▄ ▄▀▀▀▄    ▄▀▀▀▀▄█  ▀▀█▀▀▀▀▀█▀▀   ▄▀▀▀▀▄    ▄▀▀
   ▀█▄       ▄█▀   █     █     █  █      █    █     █    █▄▄▄▄▄▄█  █   
     ▀█▄   ▄█▀     █     █     █  █      █    █     █    █         █   
  ▄██▀▀█   █▀▀██▄  █     █     █   ▀▄▄▄▄▀█    ▀▄▄   ▀▄▄   ▀▄▄▄▄▀   █   
 ▀▀    █   █    ▀▀

  BOOTSTRAP! Bootstrap may take a few minutes; please be patient.

Downloading and installing packages into local source directory:

  Setting up CIPD package manager...done (27.2s)
  Setting up Python environment.....[/]
ERROR at //build/config/linux/pkg_config.gni:116:17: Script returned non-zero exit code.
    pkgresult = exec_script(pkg_config_script, args, "value")
                ^----------
Current dir: /Users/xxxx/esp/esp-matter/esp-matter/connectedhomeip/connectedhomeip/.environment/gn_out/
Command: python3 /Users/xxxx/esp/esp-matter/esp-matter/connectedhomeip/connectedhomeip/build/config/linux/pkg-config.py openssl
Returned 1.
stderr:

Package openssl was not found in the pkg-config search path.
Perhaps you should add the directory containing `openssl.pc'
to the PKG_CONFIG_PATH environment variable
No package 'openssl' found
Could not run pkg-config.

See //src/crypto/BUILD.gn:84:3: whence it was called.
  pkg_config("openssl_config") {
  ^-----------------------------
See //BUILD.gn:152:7: which caused the file to be included.
      "${chip_root}/src/crypto",
      ^------------------------
['gn', 'gen', '/Users/xxxx/esp/esp-matter/esp-matter/connectedhomeip/connectedhomeip/.environment/gn_out', '--args=dir_pigweed="/Users/xxxx/esp/esp-matter/esp-matter/connectedhomeip/connectedhomeip/third_party/pigweed/repo"']

Traceback (most recent call last):
  File "/Users/xxxx/esp/esp-matter/esp-matter/connectedhomeip/connectedhomeip/third_party/pigweed/repo/pw_env_setup/py/pw_env_setup/virtualenv_setup/install.py", line 335, in install_packages
    subprocess.check_call(
  File "/usr/local/Cellar/python@3.10/3.10.9/Frameworks/Python.framework/Versions/3.10/lib/python3.10/subprocess.py", line 369, in check_call
    raise CalledProcessError(retcode, cmd)
subprocess.CalledProcessError: Command '['gn', 'gen', '/Users/xxxx/esp/esp-matter/esp-matter/connectedhomeip/connectedhomeip/.environment/gn_out', '--args=dir_pigweed="/Users/xxxx/esp/esp-matter/esp-matter/connectedhomeip/connectedhomeip/third_party/pigweed/repo"']' returned non-zero exit status 1.

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/Users/xxxx/esp/esp-matter/esp-matter/connectedhomeip/connectedhomeip/third_party/pigweed/repo/pw_env_setup/py/pw_env_setup/env_setup.py", line 857, in <module>
    sys.exit(main())
  File "/Users/xxxx/esp/esp-matter/esp-matter/connectedhomeip/connectedhomeip/third_party/pigweed/repo/pw_env_setup/py/pw_env_setup/env_setup.py", line 849, in main
    return EnvSetup(**vars(parse())).setup()
  File "/Users/xxxx/esp/esp-matter/esp-matter/connectedhomeip/connectedhomeip/third_party/pigweed/repo/pw_env_setup/py/pw_env_setup/env_setup.py", line 501, in setup
    result = step(spin)
  File "/Users/xxxx/esp/esp-matter/esp-matter/connectedhomeip/connectedhomeip/third_party/pigweed/repo/pw_env_setup/py/pw_env_setup/env_setup.py", line 661, in virtualenv
    if not virtualenv_setup.install(
  File "/Users/xxxx/esp/esp-matter/esp-matter/connectedhomeip/connectedhomeip/third_party/pigweed/repo/pw_env_setup/py/pw_env_setup/virtualenv_setup/install.py", line 370, in install
    install_packages(gn_target)
  File "/Users/xxxx/esp/esp-matter/esp-matter/connectedhomeip/connectedhomeip/third_party/pigweed/repo/pw_env_setup/py/pw_env_setup/virtualenv_setup/install.py", line 343, in install_packages
    raise subprocess.CalledProcessError(
subprocess.CalledProcessError: Command '['gn', 'gen', '/Users/xxxx/esp/esp-matter/esp-matter/connectedhomeip/connectedhomeip/.environment/gn_out', '--args=dir_pigweed="/Users/xxxx/esp/esp-matter/esp-matter/connectedhomeip/connectedhomeip/third_party/pigweed/repo"']' returned non-zero exit status 1.
```

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

eos1d3, 2023-01-23:
I just wonder why `connectedhomeip` also use openssl@1.1 instead of `openssl@3`. Both also cause problem for Mac (or just Intel Mac).

```
cd /usr/local/lib/pkgconfig
ln -s ../../Cellar/openssl@1.1/1.1.1g/lib/pkgconfig/* .
```
https://github.com/project-chip/connectedhomeip/blob/master/docs/guides/BUILDING.md#prerequisites

I think the way to install openssl is outdated. 

I manually add `PKG_CONFIG_PATH` and the problem is gone. But I did that in terminal, not using VS Code command
`export PKG_CONFIG_PATH="/usr/local/opt/openssl@3/lib/pkgconfig" `

Diegorro98, 2023-01-26:
This extension assumes that you already satisfied all the prerequisites for your OS, because it only downloads ESP-Matter, Matter and its submodules, bootstraps Matter, and installs some python packages on the ESP-IDF python virtual environment (this last feature has not been released yet).

So is your duty to correctly installing the prerequisites so then you can install ESP-matter with or without the extension without any problem.

closed 2023-01-26 by eos1d3