espressif/vscode-esp-idf-extension #879: No Fedora/Universal Support; can't select usb/com port

BitcoinIsGarbage, 2023-01-23:
### OS

Linux

### Operating System version

Fedora Kinoite 37(36?) latest version

### Visual Studio Code version

1.74.3.32010 Flatpak/AppImage VSCodium (latest)

### ESP-IDF version

5

### Python version

3.6?

### Doctor command output

ESP-IDF: command not found

### Description

1. There is no support for prerequisites on Fedora Linux despite it now being like the 2nd most popular distro (redhat) and the creator of linux uses it.
2. I have no option to change ports off /dev/ttyUSB1
3. when I go into the dev folder there is no ttyUSB even when I plug in my board

### Debug Message

```plain
N/A
```

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

brianignacio5, 2023-02-01:
Hi @BitcoinIsGarbage 

What do you mean there is no support for prerequisites ?

Did you install required Linux packages :

```
sudo apt-get install git wget flex bison gperf python3 python3-venv cmake ninja-build ccache libffi-dev libssl-dev dfu-util libusb-1.0-0
```

Can you share the extension log or other [troubleshooting information](https://github.com/espressif/vscode-esp-idf-extension#Troubleshooting) as described in the issue template ?

Have you configure the extension as mentioned in the [install tutorial](https://github.com/espressif/vscode-esp-idf-extension/blob/master/docs/tutorial/install.md) ?

brianignacio5, 2023-02-20:
Closing this issue for lack of activity.

Comment here for additional information.

closed 2023-02-20 by brianignacio5