espressif/vscode-esp-idf-extension #880: Python error on build `/usr/bin/python: Error while finding module specification for   'idf_component_manager.prepare_component`

parksj10, 2023-01-29:
### OS

MacOS

### Operating System version

macOS 12.6

### Visual Studio Code version

1.74.3

### ESP-IDF version

5.0 (dev-container)

### Python version

3.8.10

### Doctor command output

---------------------------------------------- ESP-IDF Extension for Visual Studio Code report ---------------------------------------------
OS linux arm64 5.15.49-linuxkit 
System environment variable IDF_PYTHON_ENV_PATH 
 /opt/esp/python_env/idf5.0_py3.8_env 
System environment variable PATH 
 /vscode/vscode-server/bin/linux-arm64/97dec172d3256f8ca4bfb2143f3f76b503ca0534/bin/remote-cli:/opt/esp/idf/components/esptool_py/esptool:/opt/esp/idf/components/espcoredump:/opt/esp/idf/components/partition_table:/opt/esp/idf/components/app_update:/opt/esp/tools/xtensa-esp-elf-gdb/11.2_20220823/xtensa-esp-elf-gdb/bin:/opt/esp/tools/riscv32-esp-elf-gdb/11.2_20220823/riscv32-esp-elf-gdb/bin:/opt/esp/tools/xtensa-esp32-elf/esp-2022r1-11.2.0/xtensa-esp32-elf/bin:/opt/esp/tools/xtensa-esp32s2-elf/esp-2022r1-11.2.0/xtensa-esp32s2-elf/bin:/opt/esp/tools/xtensa-esp32s3-elf/esp-2022r1-11.2.0/xtensa-esp32s3-elf/bin:/opt/esp/tools/riscv32-esp-elf/esp-2022r1-11.2.0/riscv32-esp-elf/bin:/opt/esp/tools/esp32ulp-elf/2.35_20220830/esp32ulp-elf/bin:/opt/esp/tools/openocd-esp32/v0.11.0-esp32-20221026/openocd-esp32/bin:/opt/esp/python_env/idf5.0_py3.8_env/bin:/opt/esp/idf/tools:/opt/qemu/bin:/usr/local/sbin:/usr/local/bin:/usr/sbin:/usr/bin:/sbin:/bin 
System environment variable PYTHON 
 undefined 
Visual Studio Code version 1.74.3 
Visual Studio Code language en 
Visual Studio Code shell /usr/bin/bash 
ESP-IDF Extension version 1.5.1 
Workspace folder /workspaces 
---------------------------------------------------- Extension configuration settings ------------------------------------------------------
ESP-ADF Path (idf.espAdfPath) ${env:ADF_PATH}
ESP-IDF Path (idf.espIdfPath) /opt/esp/idf
ESP-MDF Path (idf.espMdfPath) ${env:MDF_PATH}
ESP-Matter Path (idf.espMatterPath) ${env:ESP_MATTER_PATH}
Custom extra paths (idf.customExtraPaths) 
Custom extra vars (idf.customExtraVars) {"OPENOCD_SCRIPTS":"/Users/josh/.espressif/tools/openocd-esp32/v0.11.0-esp32-20220706/openocd-esp32/share/openocd/scripts"}
Virtual env Python Path (idf.pythonBinPath) /opt/esp/python_env/idf5.1_py3.8_env/bin/python
Serial port (idf.port) /dev/cu.usbserial-2101
OpenOCD Configs (idf.openOcdConfigs) interface/ftdi/esp32_devkitj_v1.cfg,target/esp32.cfg
ESP-IDF Tools Path (idf.toolsPath) /opt/esp
Git Path (idf.gitPath) /usr/bin/git
-------------------------------------------------------- Configurations access -------------------------------------------------------------
Access to ESP-ADF Path (idf.espAdfPath) false
Access to ESP-IDF Path (idf.espIdfPath) true
Access to ESP-MDF Path (idf.espMdfPath) false
Access to ESP-Matter Path (idf.espMatterPath) false
Access to ESP-IDF Custom extra paths
Access to Virtual env Python Path (idf.pythonBinPath) false
Access to CMake in environment PATH true
Access to Ninja in environment PATH true
Access to ESP-IDF Tools Path (idf.toolsPath) true
-------------------------------------------------------- Configurations has spaces -------------------------------------------------------------
Spaces in system environment Path false
Spaces in ESP-ADF Path (idf.espAdfPath) false
Spaces in ESP-IDF Path (idf.espIdfPath) false
Spaces in ESP-MDF Path (idf.espMdfPath) false
Spaces in ESP-Matter Path (idf.espMatterPath) false
Spaces in ESP-IDF Custom extra paths
Spaces in Virtual env Python Path (idf.pythonBinPath) false
Spaces in ESP-IDF Tools Path (idf.toolsPath) false
----------------------------------------------------------- Executables Versions -----------------------------------------------------------
Git version 2.25.1
ESP-IDF version 5.0
Python version Not found
Python's pip version Not found
-------------------------------------------------- Python packages in idf.pythonBinPath ----------------------------------------------------
---------------------------------------------------- Check ESP-IDF python requirements.txt -------------------------------------------------
Check ESP-IDF Python packages undefined
---------------------------------------------------- Check extension requirements.txt ------------------------------------------------------
Check Extension Python packages undefined
---------------------------------------------------- Check ESP-IDF debug adapter requirements.txt ------------------------------------------
Check Debug AdapterPython packages undefined
----------------------------------------------------------- Latest error -----------------------------------------------------------------
Latest error at Unknown error in ESP-IDF doctor command
--------------------------------------------------------------------------------------------------------------------------------------------

### Description

Seems like the ESP-IDF command is trying to use the docker images native python `/usr/bin/python` rather than the specific version `/opt/esp/python_env/idf5.0_py3.8_env/bin/python`. I can get the error to bypass by adding the following to my CMakeList.txt

```
set(PYTHON "/opt/esp/python_env/idf5.0_py3.8_env/bin/python")
```

Note: this I the dev container I used: https://hub.docker.com/layers/espressif/idf/release-v5.0/images/sha256-3a595f15ab1a634ffe67876c690dbd70cfdc9179d4803cd8ca8ad413f13a3170?context=explore

### Debug Message

```plain
CMake Error at /opt/esp/idf/tools/cmake/build.cmake:519 (message):
  /usr/bin/python: Error while finding module specification for
  'idf_component_manager.prepare_components' (ModuleNotFoundError: No module
  named 'idf_component_manager')

Call Stack (most recent call first):
  /opt/esp/idf/tools/cmake/project.cmake:440 (idf_build_process)
  CMakeLists.txt:58 (project)
```

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

brianignacio5, 2023-01-30:
Your extension setup has `idf.pythonBinPath` with value `/opt/esp/python_env/idf5.1_py3.8_env/bin/python` 

Maybe update it to `/opt/esp/python_env/idf5.0_py3.8_env/bin/python` ?

parksj10, 2023-01-31:
@brianignacio5, thanks that seems to fix it, thanks.

closed 2023-02-01 by brianignacio5

asm128, 2025-03-08:
> Your extension setup has `idf.pythonBinPath` with value `/opt/esp/python_env/idf5.1_py3.8_env/bin/python`
> 
> Maybe update it to `/opt/esp/python_env/idf5.0_py3.8_env/bin/python` ?

explain