espressif/vscode-esp-idf-extension #881: ESP-IDF compilation error problem

ailunwowo, 2023-01-30:
### OS

Windows

### Operating System version

Windows 10

### ESP-IDF version

4.4.3

### Python version

3.8.7

### Doctor command output

- Found Git: F:/Espressif/tools/idf-git/2.34.2/cmd/git.exe (found version "2.34.1.windows.1") 
-- The C compiler identification is GNU 8.4.0
-- The CXX compiler identification is GNU 8.4.0
-- The ASM compiler identification is GNU
-- Found assembler: F:/Espressif/ESP4.4_tools/tools/xtensa-esp32s3-elf/esp-2021r2-patch5-8.4.0/xtensa-esp32s3-elf/bin/xtensa-esp32s3-elf-gcc.exe
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Check for working C compiler: F:/Espressif/ESP4.4_tools/tools/xtensa-esp32s3-elf/esp-2021r2-patch5-8.4.0/xtensa-esp32s3-elf/bin/xtensa-esp32s3-elf-gcc.exe - skipped
-- Detecting C compile features       
-- Detecting C compile features - done
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Check for working CXX compiler: F:/Espressif/ESP4.4_tools/tools/xtensa-esp32s3-elf/esp-2021r2-patch5-8.4.0/xtensa-esp32s3-elf/bin/xtensa-esp32s3-elf-g++.exe - skipped
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Project is not inside a git repository, or git repository has no commits; will not use 'git describe' to determine PROJECT_VER.
-- Building ESP-IDF components for target esp32s3
Solving dependencies requirements
....CMake Error at F:/Espressif/frameworks/esp-idf-v4.4.3/esp-idf/tools/cmake/build.cmake:452 (message):
  Traceback (most recent call last):

    File "runpy.py", line 194, in _run_module_as_main
    File "runpy.py", line 87, in _run_code
    File "f:\Espressif\ESP4.4_tools\python_env\idf4.4_py3.8_env\lib\site-packages\idf_component_manager\prepare_components\__main__.py", line 6, in <module>
      main()
    File "f:\Espressif\ESP4.4_tools\python_env\idf4.4_py3.8_env\lib\site-packages\idf_component_manager\prepare_components\prepare.py", line 124, in main
      args.func(args)
    File "f:\Espressif\ESP4.4_tools\python_env\idf4.4_py3.8_env\lib\site-packages\idf_component_manager\prepare_components\prepare.py", line 27, in prepare_dep_dirs
      ComponentManager(
    File "f:\Espressif\ESP4.4_tools\python_env\idf4.4_py3.8_env\lib\site-packages\idf_component_manager\core.py", line 62, in wrapper
      return func(self, *args, **kwargs)
    File "f:\Espressif\ESP4.4_tools\python_env\idf4.4_py3.8_env\lib\site-packages\idf_component_manager\core.py", line 438, in prepare_dep_dirs
      downloaded_component_paths, downloaded_component_version_dict = download_project_dependencies(
    File "f:\Espressif\ESP4.4_tools\python_env\idf4.4_py3.8_env\lib\site-packages\idf_component_manager\dependencies.py", line 111, in download_project_dependencies
      solution = solver.solve()
    File "f:\Espressif\ESP4.4_tools\python_env\idf4.4_py3.8_env\lib\site-packages\idf_component_manager\version_solver\version_solver.py", line 34, in solve
      self.solve_manifest(manifest)
    File "f:\Espressif\ESP4.4_tools\python_env\idf4.4_py3.8_env\lib\site-packages\idf_component_manager\version_solver\version_solver.py", line 53, in solve_manifest
      self.solve_component(requirement)
    File "f:\Espressif\ESP4.4_tools\python_env\idf4.4_py3.8_env\lib\site-packages\idf_component_manager\version_solver\version_solver.py", line 57, in solve_component
      name=requirement.name, spec=requirement.version_spec, target=self.requirements.target)
    File "f:\Espressif\ESP4.4_tools\python_env\idf4.4_py3.8_env\lib\site-packages\idf_component_tools\manifest\manifest.py", line 195, in version_spec
      return self.source.normalize_spec(self._version_spec)
    File "f:\Espressif\ESP4.4_tools\python_env\idf4.4_py3.8_env\lib\site-packages\idf_component_tools\sources\git.py", line 196, in normalize_spec
      commit_id = self._client.get_commit_id_by_ref(self.git_repo, self.cache_path(), ref)
    File "f:\Espressif\ESP4.4_tools\python_env\idf4.4_py3.8_env\lib\site-packages\idf_component_tools\git_client.py", line 41, in wrapper
      return func(self, *args, **kwargs)
    File "f:\Espressif\ESP4.4_tools\python_env\idf4.4_py3.8_env\lib\site-packages\idf_component_tools\git_client.py", line 72, in wrapper
      return func(self, *args, **kwargs)
    File "f:\Espressif\ESP4.4_tools\python_env\idf4.4_py3.8_env\lib\site-packages\idf_component_tools\git_client.py", line 159, in get_commit_id_by_ref
      ref = self.run(['ls-remote', '--exit-code', 'origin', 'HEAD'], cwd=bare_path)[:40]
    File "f:\Espressif\ESP4.4_tools\python_env\idf4.4_py3.8_env\lib\site-packages\idf_component_tools\git_client.py", line 183, in run
      raise GitCommandError(

  idf_component_tools.git_client.GitCommandError: 'git ls-remote --exit-code
  origin HEAD' failed with exit code 128

  fatal: unable to access 'https://github.com/espressif/esp-rainmaker.git/':
  OpenSSL SSL_read: Connection was reset, errno 10054

Call Stack (most recent call first):
  F:/Espressif/frameworks/esp-idf-v4.4.3/esp-idf/tools/cmake/project.cmake:384 (idf_build_process)
  CMakeLists.txt:13 (project)

-- Configuring incomplete, errors occurred!
See also "F:/speak/watering_demo/build/CMakeFiles/CMakeOutput.log".

 *  终端进程“C:\WINDOWS\System32\WindowsPowerShell\v1.0\powershell.exe -Command cmake -G Ninja -DPYTHON_DEPS_CHECKED=1 -DESP_PLATFORM=1 -B f:\speak\watering_demo\build -S f:\speak\watering_demo -B f:\speak\watering_demo\build -S f:\speak\watering_demo”已终止，退出代码: 1。 

### Description

Regarding this error report, the strange problem is that this project is a project I created half a month ago, and the compilation was correct at that time, but now I report an error when I compile it, and it will report an error as long as the clone project is not cloned. No error will be reported, why is this?

### Debug Message

```plain
ninja: error: loading 'build.ninja': 系统找不到指定的文件。

 *  终端进程“C:\WINDOWS\System32\WindowsPowerShell\v1.0\powershell.exe -Command ninja ”已终止，退出代码: 1。
```

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

brianignacio5, 2023-01-30:
Seems the component manager is trying to access GitHub and having an SSL connection issue. Maybe some dependencies are updated and need to check the connection is working ?

ailunwowo, 2023-01-30:
How to check?

brianignacio5, 2023-01-31:
Is really hard to know without understanding more about your current project. I would think is related to components listed in your project and the version reference. For some reason idf_component_manager is calling esp-matter and failing because it doesn't have access to GitHub.

The only thing I am sure is that you are having a network issue to access GitHub.

ailunwowo, 2023-01-31:
Ok, I probably understand, but I reinstalled the esp-idf environment, why it still fails to run

brianignacio5, 2023-01-31:
Did you deleted the build folder? There is probably some cache files there.

ailunwowo, 2023-01-31:
Yes, because I have to rebuild the project, but the new project also reports an error

ailunwowo, 2023-01-31:
Currently working again because I reinstalled the esp-idf environment and tools in vscode

brianignacio5, 2023-01-31:
Then probably something changed in the python packages in the virtual environment. Anyway glad the issue is fixed.

closed 2023-01-31 by brianignacio5

ailunwowo, 2023-01-31:
> Then probably something changed in the python packages in the virtual environment. Anyway glad the issue is fixed.

Anyway, thank you for your serious answer today