espressif/vscode-esp-idf-extension #886: virtualenv error install init

perdomocarlos, 2023-02-12:
### OS

Linux

### Operating System version

Ubuntu 22.04

### Visual Studio Code version

1.74.2

### ESP-IDF version

4.4

### Python version

3.10

### Doctor command output

Collecting virtualenv
  Using cached virtualenv-20.19.0-py3-none-any.whl (8.7 MB)
Collecting platformdirs<4,>=2.4
  Using cached platformdirs-3.0.0-py3-none-any.whl (14 kB)
Collecting distlib<1,>=0.3.6
  Using cached distlib-0.3.6-py2.py3-none-any.whl (468 kB)
Collecting filelock<4,>=3.4.1
  Using cached filelock-3.9.0-py3-none-any.whl (9.7 kB)
Installing collected packages: distlib, platformdirs, filelock, virtualenv
Successfully installed distlib-0.3.6 filelock-3.9.0 platformdirs-3.0.0 virtualenv-20.19.0
  WARNING: The script virtualenv is installed in '/home/carlosf/.local/bin' which is not on PATH.
  Consider adding this directory to PATH or, if you prefer to suppress this warning, use --no-warn-script-location.

/usr/bin/python: No module named virtualenv
Command failed: "/usr/bin/python" -m virtualenv "/home/carlosf/.espressif/python_env/idf4.4_py3.10_env" -p "/usr/bin/python"
/usr/bin/python: No module named virtualenv
Command failed: "/usr/bin/python" -m virtualenv "/home/carlosf/.espressif/python_env/idf4.4_py3.10_env" -p "/usr/bin/python"
/usr/bin/python: No module named virtualenv

### Description

When start to install virtual python environment, take this:
Collecting virtualenv
  Using cached virtualenv-20.19.0-py3-none-any.whl (8.7 MB)
Collecting platformdirs<4,>=2.4
  Using cached platformdirs-3.0.0-py3-none-any.whl (14 kB)
Collecting distlib<1,>=0.3.6
  Using cached distlib-0.3.6-py2.py3-none-any.whl (468 kB)
Collecting filelock<4,>=3.4.1
  Using cached filelock-3.9.0-py3-none-any.whl (9.7 kB)
Installing collected packages: distlib, platformdirs, filelock, virtualenv
Successfully installed distlib-0.3.6 filelock-3.9.0 platformdirs-3.0.0 virtualenv-20.19.0
  WARNING: The script virtualenv is installed in '/home/carlosf/.local/bin' which is not on PATH.
  Consider adding this directory to PATH or, if you prefer to suppress this warning, use --no-warn-script-location.

/usr/bin/python: No module named virtualenv
Command failed: "/usr/bin/python" -m virtualenv "/home/carlosf/.espressif/python_env/idf4.4_py3.10_env" -p "/usr/bin/python"
/usr/bin/python: No module named virtualenv
Command failed: "/usr/bin/python" -m virtualenv "/home/carlosf/.espressif/python_env/idf4.4_py3.10_env" -p "/usr/bin/python"
/usr/bin/python: No module named virtualenv

### Debug Message

```plain
Collecting virtualenv
  Using cached virtualenv-20.19.0-py3-none-any.whl (8.7 MB)
Collecting platformdirs<4,>=2.4
  Using cached platformdirs-3.0.0-py3-none-any.whl (14 kB)
Collecting distlib<1,>=0.3.6
  Using cached distlib-0.3.6-py2.py3-none-any.whl (468 kB)
Collecting filelock<4,>=3.4.1
  Using cached filelock-3.9.0-py3-none-any.whl (9.7 kB)
Installing collected packages: distlib, platformdirs, filelock, virtualenv
Successfully installed distlib-0.3.6 filelock-3.9.0 platformdirs-3.0.0 virtualenv-20.19.0
  WARNING: The script virtualenv is installed in '/home/carlosf/.local/bin' which is not on PATH.
  Consider adding this directory to PATH or, if you prefer to suppress this warning, use --no-warn-script-location.

/usr/bin/python: No module named virtualenv
Command failed: "/usr/bin/python" -m virtualenv "/home/carlosf/.espressif/python_env/idf4.4_py3.10_env" -p "/usr/bin/python"
/usr/bin/python: No module named virtualenv
Command failed: "/usr/bin/python" -m virtualenv "/home/carlosf/.espressif/python_env/idf4.4_py3.10_env" -p "/usr/bin/python"
/usr/bin/python: No module named virtualenv
```

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

moefear85, 2023-02-12:
Your python dependencies are not setup correctly (probably not following official instructions). You need to:
`pip install virtualenv`

moefear85, 2023-02-12:
you need to add the folder `/home/carlosf/.local/bin` to your path

brianignacio5, 2023-02-22:
Closing this issue for lack of response from posting user.

Make sure to follow the install steps as described in the readme, like the [prerequisites](https://github.com/espressif/vscode-esp-idf-extension#prerequisites)

Comment here if you still have issues.

closed 2023-02-22 by brianignacio5