espressif/vscode-esp-idf-extension #887: Can't compile any project using esp-idf from master branch since issue #813

moefear85, 2023-02-12:
### OS

Linux

### Operating System version

Ubuntu 22.10

### Visual Studio Code version

1.75.1

### ESP-IDF version

5.1

### Python version

3.10.7

### Doctor command output

---------------------------------------------- ESP-IDF Extension for Visual Studio Code report ---------------------------------------------
OS linux x64 5.19.0-31-generic 
System environment variable IDF_PYTHON_ENV_PATH 
 undefined 
System environment variable PATH 
 <HOMEPATH>/.local/bin:<HOMEPATH>/.local/bin:/usr/local/sbin:/usr/local/bin:/usr/sbin:/usr/bin:/sbin:/bin:/usr/games:/usr/local/games:/snap/bin:<HOMEPATH>/.local/bin:<HOMEPATH>/software/espressif/tools/xtensa-lx106-elf/bin:<HOMEPATH>/scripts:/media/moefear/BACKUP/scripts:<HOMEPATH>/.local/bin:<HOMEPATH>/software/espressif/tools/xtensa-lx106-elf/bin:<HOMEPATH>/scripts:/media/moefear/BACKUP/scripts 
System environment variable PYTHON 
 undefined 
Visual Studio Code version 1.75.1 
Visual Studio Code language en 
Visual Studio Code shell /usr/bin/bash 
ESP-IDF Extension version 1.5.1 
Workspace folder <HOMEPATH>/temp/v5.1_12-02-23_bb25d6a 
---------------------------------------------------- Extension configuration settings ------------------------------------------------------
ESP-ADF Path (idf.espAdfPath) <HOMEPATH>/source/esp-adf
ESP-IDF Path (idf.espIdfPath) <HOMEPATH>/source/esp-idf
ESP-MDF Path (idf.espMdfPath) ${env:MDF_PATH}
ESP-Matter Path (idf.espMatterPath) ${env:ESP_MATTER_PATH}
Custom extra paths (idf.customExtraPaths) <HOMEPATH>/software/espressif/tools/xtensa-esp-elf-gdb/11.2_20220823/xtensa-esp-elf-gdb/bin:<HOMEPATH>/software/espressif/tools/riscv32-esp-elf-gdb/11.2_20220823/riscv32-esp-elf-gdb/bin:<HOMEPATH>/software/espressif/tools/xtensa-esp32-elf/esp-2022r1-11.2.0/xtensa-esp32-elf/bin:<HOMEPATH>/software/espressif/tools/xtensa-esp32s2-elf/esp-2022r1-11.2.0/xtensa-esp32s2-elf/bin:<HOMEPATH>/software/espressif/tools/xtensa-esp32s3-elf/esp-2022r1-11.2.0/xtensa-esp32s3-elf/bin:<HOMEPATH>/software/espressif/tools/riscv32-esp-elf/esp-2022r1-11.2.0/riscv32-esp-elf/bin:<HOMEPATH>/software/espressif/tools/esp32ulp-elf/2.35_20220830/esp32ulp-elf/bin:<HOMEPATH>/software/espressif/tools/cmake/3.24.0/bin:<HOMEPATH>/software/espressif/tools/openocd-esp32/v0.11.0-esp32-20221026/openocd-esp32/bin:<HOMEPATH>/software/espressif/tools/ninja/1.10.2:<HOMEPATH>/software/espressif/tools/esp-rom-elfs/20220823
Custom extra vars (idf.customExtraVars) {}
Virtual env Python Path (idf.pythonBinPath) <HOMEPATH>/software/espressif/python_env/idf5.1_py3.10_env/bin/python
Serial port (idf.port) /dev/ttyUSB1
OpenOCD Configs (idf.openOcdConfigs) interface/ftdi/esp32_devkitj_v1.cfg,target/esp32.cfg
ESP-IDF Tools Path (idf.toolsPath) <HOMEPATH>/software/espressif
Git Path (idf.gitPath) git
-------------------------------------------------------- Configurations access -------------------------------------------------------------
Access to ESP-ADF Path (idf.espAdfPath) false
Access to ESP-IDF Path (idf.espIdfPath) true
Access to ESP-MDF Path (idf.espMdfPath) false
Access to ESP-Matter Path (idf.espMatterPath) false
Access to ESP-IDF Custom extra paths
Access to <HOMEPATH>/software/espressif/tools/xtensa-esp-elf-gdb/11.2_20220823/xtensa-esp-elf-gdb/bin: true
Access to <HOMEPATH>/software/espressif/tools/riscv32-esp-elf-gdb/11.2_20220823/riscv32-esp-elf-gdb/bin: true
Access to <HOMEPATH>/software/espressif/tools/xtensa-esp32-elf/esp-2022r1-11.2.0/xtensa-esp32-elf/bin: true
Access to <HOMEPATH>/software/espressif/tools/xtensa-esp32s2-elf/esp-2022r1-11.2.0/xtensa-esp32s2-elf/bin: true
Access to <HOMEPATH>/software/espressif/tools/xtensa-esp32s3-elf/esp-2022r1-11.2.0/xtensa-esp32s3-elf/bin: true
Access to <HOMEPATH>/software/espressif/tools/riscv32-esp-elf/esp-2022r1-11.2.0/riscv32-esp-elf/bin: true
Access to <HOMEPATH>/software/espressif/tools/esp32ulp-elf/2.35_20220830/esp32ulp-elf/bin: true
Access to <HOMEPATH>/software/espressif/tools/cmake/3.24.0/bin: true
Access to <HOMEPATH>/software/espressif/tools/openocd-esp32/v0.11.0-esp32-20221026/openocd-esp32/bin: true
Access to <HOMEPATH>/software/espressif/tools/ninja/1.10.2: true
Access to <HOMEPATH>/software/espressif/tools/esp-rom-elfs/20220823: true
Access to Virtual env Python Path (idf.pythonBinPath) true
Access to CMake in environment PATH true
Access to Ninja in environment PATH true
Access to ESP-IDF Tools Path (idf.toolsPath) true
-------------------------------------------------------- Configurations has spaces -------------------------------------------------------------
Spaces in system environment Path false
[... 166 lines cut ...]

CMake Error at /home/moefear/source/esp-idf/tools/cmake/project.cmake:348 (__project):
  The CMAKE_ASM_COMPILER:

    xtensa-esp32-elf-gcc

  is not a full path and was not found in the PATH.

  Tell CMake where to find the compiler by setting either the environment
  variable "ASM" or the CMake cache entry CMAKE_ASM_COMPILER to the full path
  to the compiler, or to the compiler name if it is in the PATH.
Call Stack (most recent call first):
  CMakeLists.txt:8 (project)

-- Warning: Did not find file Compiler/-ASM
-- Configuring incomplete, errors occurred!
See also "/home/moefear/temp/v5.1_12-02-23_bb25d6a/build/CMakeFiles/CMakeOutput.log".
See also "/home/moefear/temp/v5.1_12-02-23_bb25d6a/build/CMakeFiles/CMakeError.log".

 *  The terminal process "/usr/bin/bash '-c', 'cmake -G Ninja -DPYTHON_DEPS_CHECKED=1 -DESP_PLATFORM=1 -B /home/moefear/temp/v5.1_12-02-23_bb25d6a/build -S /home/moefear/temp/v5.1_12-02-23_bb25d6a'" terminated with exit code: 1.
```

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

moefear85, 2023-02-12:
update: I'm now also getting this error even if I use latest esp-idf release/v5.0 branch, when I download and install esp-idf manually, then point the extension to it, even though I'm doing this exactly the same way I always do. It's not happening when I fully install esp-idf from within the extension.

moefear85, 2023-02-12:
update: I'm now also getting this error even if I use latest esp-idf release/v5.0 branch, when I download and install esp-idf manually, then point the extension to it, even though I'm doing this exactly the same way I always do. It's not happening when I fully install esp-idf from within the extension.

And again, when I try menuconfig and build from the command line after sourcing, it works. So the problem is definately within the extension itself, not with my manual install. However, I've realized, when I click on menuconfig within the extension instead of directly hitting build, I get the following error message:

`
[SDK Configuration Editor]
---------------------------ERROR--------------------------

Error: spawn /home/moefear/software/espressif/python_env/idf5.1_py3.10_env/bin/python ENOENT
	at ChildProcess._handle.onexit (node:internal/child_process:283:19)
	at onErrorNT (node:internal/child_process:478:16)
	at process.processTicksAndRejections (node:internal/process/task_queues:83:21)
-----------------------END OF ERROR-----------------------
`

brianignacio5, 2023-02-13:
So as you mentioned, settings.json can be saved in multiples locations: (Global VSCode settings aka User Settings, Workspace settings and Workspace Folder settings.json). Look into [multiple projects doc](https://github.com/espressif/vscode-esp-idf-extension/blob/master/docs/MULTI_PROJECTS.md). The extension will always use from workspace folder > workspace > user settings. Settings can be saved in different locations depending on `idf.saveScope` configuration setting value. The doctor command is useful to know which settings are being used by the extension right now.

Related to the toolchain, it seems the extension setup fails to find the xtensa gcc toolchain. Try opening an ESP-IDF Terminal with `Open ESP-IDF Terminal` extension command and run `which xtensa-esp32-elf-gcc` to see what it finds.

moefear85, 2023-02-19:
Irrespective which settings VSCode is using, the settings.json file created on a new project, is supposed to point to the  right python out of the box, but it isn't, so this is most likely something to fix in the extension itself, especially since it was never an issue in the past and only started after upgrading esp-idf.

As for the toolchain, `which xtensa-esp32-elf-gcc` doesn't output anything. If I open any standard command prompt, I also don't get any output. But if I source `export.sh` then I get:
`/home/moefear/software/espressif/tools/xtensa-esp32-elf/esp-2022r1-11.2.0/xtensa-esp32-elf/bin/xtensa-esp32-elf-gcc`

moefear85, 2023-02-19:
Also, note that the above provided doctor output is from previous extension/idf-installs. The current one, with which I am still getting the mentioned errors, is as follows:

```
---------------------------------------------- ESP-IDF Extension for Visual Studio Code report ---------------------------------------------
OS linux x64 5.19.0-31-generic 
System environment variable IDF_PYTHON_ENV_PATH 
 undefined 
System environment variable PATH 
 <HOMEPATH>/.local/bin:<HOMEPATH>/.local/bin:/usr/local/sbin:/usr/local/bin:/usr/sbin:/usr/bin:/sbin:/bin:/usr/games:/usr/local/games:/snap/bin:<HOMEPATH>/.local/bin:<HOMEPATH>/software/espressif/tools/xtensa-lx106-elf/bin:<HOMEPATH>/scripts:/media/moefear/BACKUP/scripts:<HOMEPATH>/.local/bin:<HOMEPATH>/software/espressif/tools/xtensa-lx106-elf/bin:<HOMEPATH>/scripts:/media/moefear/BACKUP/scripts 
System environment variable PYTHON 
 undefined 
Visual Studio Code version 1.75.1 
Visual Studio Code language en 
Visual Studio Code shell /usr/bin/bash 
ESP-IDF Extension version 1.5.1 
Workspace folder <HOMEPATH>/temp/esp-idf-v5.1_13-02-23_bb25d6a 
---------------------------------------------------- Extension configuration settings ------------------------------------------------------
ESP-ADF Path (idf.espAdfPath) <HOMEPATH>/source/esp-adf
ESP-IDF Path (idf.espIdfPath) <HOMEPATH>/source/esp-idf
ESP-MDF Path (idf.espMdfPath) ${env:MDF_PATH}
ESP-Matter Path (idf.espMatterPath) ${env:ESP_MATTER_PATH}
Custom extra paths (idf.customExtraPaths) 
Custom extra vars (idf.customExtraVars) {}
Virtual env Python Path (idf.pythonBinPath) <HOMEPATH>/software/espressif/python_env/idf5.1_py3.10_env/bin/python
Serial port (idf.port) /dev/ttyUSB1
OpenOCD Configs (idf.openOcdConfigs) interface/ftdi/esp32_devkitj_v1.cfg,target/esp32.cfg
ESP-IDF Tools Path (idf.toolsPath) <HOMEPATH>/software/espressif
Git Path (idf.gitPath) /usr/bin/git
-------------------------------------------------------- Configurations access -------------------------------------------------------------
Access to ESP-ADF Path (idf.espAdfPath) false
Access to ESP-IDF Path (idf.espIdfPath) true
Access to ESP-MDF Path (idf.espMdfPath) false
Access to ESP-Matter Path (idf.espMatterPath) false
Access to ESP-IDF Custom extra paths
Access to Virtual env Python Path (idf.pythonBinPath) true
Access to CMake in environment PATH true
Access to Ninja in environment PATH true
Access to ESP-IDF Tools Path (idf.toolsPath) true
-------------------------------------------------------- Configurations has spaces -------------------------------------------------------------
Spaces in system environment Path false
Spaces in ESP-ADF Path (idf.espAdfPath) false
Spaces in ESP-IDF Path (idf.espIdfPath) false
Spaces in ESP-MDF Path (idf.espMdfPath) false
Spaces in ESP-Matter Path (idf.espMatterPath) false
Spaces in ESP-IDF Custom extra paths
Spaces in Virtual env Python Path (idf.pythonBinPath) false
Spaces in ESP-IDF Tools Path (idf.toolsPath) false
----------------------------------------------------------- Executables Versions -----------------------------------------------------------
Git version 2.37.2
ESP-IDF version 5.1.0
Python version 3.10.7
Python's pip version 23.0
-------------------------------------------------- Python packages in idf.pythonBinPath ----------------------------------------------------
bitstring version: 4.0.1
CacheControl version: 0.12.11
certifi version: 2022.12.7
cffi version: 1.15.1
charset-normalizer version: 3.0.1
click version: 8.1.3
colorama version: 0.4.6
construct version: 2.10.68
contextlib2 version: 21.6.0
cryptography version: 39.0.1
ecdsa version: 0.18.0
esp-coredump version: 1.4.2
esp-idf-kconfig version: 1.0.0
esp-idf-monitor version: 0.1.0
esptool version: 4.5.dev2
freertos-gdb version: 1.0.2
future version: 0.18.3
[... 48 lines cut ...]
                "${config:idf.espIdfPath}/components/**",
                "${config:idf.espIdfPathWin}/components/**",
                "${config:idf.espAdfPath}/components/**",
                "${config:idf.espAdfPathWin}/components/**",
                "${workspaceFolder}/**"
            ],
            "browse": {
                "path": [
                    "${config:idf.espIdfPath}/components",
                    "${config:idf.espIdfPathWin}/components",
                    "${config:idf.espAdfPath}/components/**",
                    "${config:idf.espAdfPathWin}/components/**",
                    "${workspaceFolder}"
                ],
                "limitSymbolsToIncludedHeaders": false
            }
        }
    ],
    "version": 4
}
 
----------------------------------------------------------- Latest error -----------------------------------------------------------------
Latest error at Unknown error in ESP-IDF doctor command
--------------------------------------------------------------------------------------------------------------------------------------------
````

moefear85, 2023-02-19:
here are my user settings. I don't see anything odd:

`
{
    "workbench.startupEditor": "none",
    "idf.telemetry": false,
    "telemetry.telemetryLevel": "off",
    "explorer.confirmDelete": false,
    "idf.espIdfPath": "/home/moefear/source/esp-idf",
    "idf.toolsPath": "/home/moefear/software/espressif",
    "idf.customExtraPaths": "",
    "idf.customExtraVars": "{}",
    "idf.gitPath": "/usr/bin/git",
    "cmake.configureOnOpen": false,
    "files.autoSave": "afterDelay",
    "security.workspace.trust.enabled": false,
    "security.workspace.trust.untrustedFiles": "open",
    "task.allowAutomaticTasks": "on",
    "extensions.ignoreRecommendations": true,
    "editor.wordWrap": "on",
    "debug.console.wordWrap": false
}
`

moefear85, 2023-02-19:
also I've deleted the entire code and extensions folder. Then i reinstalled the extension and tried again. It simply isn't working with the master branch. Is this on purpose? It only works with express install.

brianignacio5, 2023-02-24:
I'm not sure what is happening here. I need the latest information of your setup.

First, the latest doctor command output shows that extension and debug adapter python packages have some error. Maybe try to install them with the `ESP-IDF: Install ESP-IDF Python Packages`.

Also, what is the current build output ? Seems that "idf.customExtraPaths": "", is empty. Are you using a docker container here? Tools should be listed here but they are not shown in the doctor command output, and based on your `which xtensa-esp32-elf-gcc` doesn't seem to be globally installed.

What about the project's directory `.vscode/settings.json` ?

moefear85, 2023-02-25:
not using docker. i think the python errors are irrelevant, but i'll try to reconnfigure that. the important question is, do you have any intuition, as to why the extension isn't saving the espressif folder's path and not finding the compilers until the build folder is created? I'm not sure what you mean by "tools" but if you mean the compilers in the espressif folder, then that is basically the question, why isn't the extension noting their path despite them existing, and despite a manual build from the commandline is possible. I think something subtle changed in the extension, such as inadvertently hardcoded paths, because my espressif folder doesn't have a dot in the name.

brianignacio5, 2023-02-27:
In my opinion the reason the build is failing and toolchain are not found is because the environment PATH variable is not set properly. I think is related to your `idf.customExtraPaths` being empty, which should not be the case. You can see that your doctor command doesn't do any ESP-IDF tool validation because of this.

moefear85, 2023-03-01:
that can't be the case, since the compilers are not found using the PATH, rather through IDF_TOOLS_PATH. So that wouldn't explain why it works from the command line and then from the IDE after the build folder exists.

Moreover, `echo $PATH` gives `/home/moefear/.local/bin:/usr/local/sbin:/usr/local/bin:/usr/sbin:/usr/bin:/sbin:/bin:/usr/games:/usr/local/games:/snap/bin:/home/moefear/.local/bin:/home/moefear/software/espressif/tools/xtensa-lx106-elf/bin:/home/moefear/scripts:/media/moefear/BACKUP/scripts`. I don't see anything wrong with that.
`echo $IDF_PATH` gives: `/home/moefear/source/esp-idf` and `echo $IDF_TOOLS_PATH` gives `/home/moefear/software/espressif`. Both are correct and are valid soft links to the actual folders simply have the version, download date, and commit number as an extension, but are visible in the above logs, and there are no spaces or illegal characters in them.

moefear85, 2023-03-01:
Moreover, I just upgraded the esp-idf extension. Now when trying to use a preexisting setup, I get the following error message:
```
/bin/sh: 1: undefined: not found
Command failed: "undefined" -m pip install --upgrade --constraint "/home/moefear/.vscode/extensions/espressif.esp-idf-extension-1.6.0/espidf.constraints.txt" --no-warn-script-location  -r "/home/moefear/.vscode/extensions/espressif.esp-idf-extension-1.6.0/requirements.txt"
/bin/sh: 1: undefined: not found
```
So now it seems the extension isn't finding/using python, even though again, I didn't change anything in my system and python works fine from the command line. I don't know what shell the xtension uses behind the scenes, but it seems the extension is setting that up incorrectly. From the command line, everything works normally. Just not in the command line within visual studio that is created/setup by the extension itself.

I think this is an issue that is not showing up on whatever PC you guys use to develop the extension, because you have some subtle non-standard environment that is masking the bug in your case. Even a fresh virtual PC install faces the same error.

moefear85, 2023-03-01:
moreover, all prerequisites are met. Moreover, the mistake should have been caught by the doctor command itself, otherwise the doctor command is not properly setup to detect incorrect environment or PATH or any other variables. as for `idf.customExtraPaths` always was empty before, and never created any problems. If it should not be empty, then what exactly should go there, or what is there by default or in your system?

closed 2023-03-16 by brianignacio5