espressif/vscode-esp-idf-extension #889: Linux and Windows have different behavior related to Colorization of Code

Uvamosk, 2023-02-15:
### OS

Linux

### Operating System version

Linux Mint 21.1

### Visual Studio Code version

1.75.1

### ESP-IDF version

4.4.4

### Python version

3.8.7

### Doctor command output

No Data to provide

### Description

I have two fresh installs of VScode, ESP-IDF and C/C++ extension.
One is on Linux Mint, The other is in windows.
I am unable to get the Linux Version to match the windows versions colorization.
Both of these copies of VScode, ESP-IDF and C/C++ extension are fresh installs. 
 The only thing I have done in the windows version is add "C_Cpp.enhancedColorization": "enabled" to the settings.json inside of the .vscode folder of the project.
 When I add this line the windows version colorizes as I have always known it to look.
 When I add this line to the Linux version nothing happens and all defines and variables stay white.
 
 I have spent hours upon hours trying to figure out why the Linux version will not look like the windows version. 
 Nothing seems to change the color of the Variables or Defines even if I use other themes, 3rd party themes or try to manually add something to settings.json. This was all tested on a 3rd copy of Linux, Vscode, IDF and C/C++ extension I have setup for trying to figure out this problem.
 
 I can't even figure out if it is a Vscode issue or if the IDF causes this issue, Or what it is.
 
 The attached photo shows what I am describing. 
 The right side is Windows, The Left Side is Linux 
 The only difference between these two VScode, IDF, C/C++ installs is adding "C_Cpp.enhancedColorization": "enabled" to the settings.json inside of the .vscode folder of the project on the Windows machine.
 
 
[image]

### Debug Message

```plain
No debug data
```

### Other Steps to Reproduce

Install VSCode, ESP-IDF, C/C++ extension on windows. Create a project using Examples with in VScode/ESP-IDF
All variables will be White along with Defines. 
Add "C_Cpp.enhancedColorization": "enabled"  to the Settings.json inside of the projects .vscode folder.
save
Re open project and Variables and defines will be colorized.

Follow same steps on Linux
Nothing will happen

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

Uvamosk, 2023-02-15:
This is insane, I have found all my issues are caused by Symbolic Links in Linux.
I had created Links to my DEV folders on the Desktop.
Click Link on Desktop Path Bar = /home/user01/Desktop/MyProjects  I then right click a project and open with VS code and  all my variable and define issues exist.

Instead of clicking desktop link I go to the folder directly the Path Bar = /home/user01/MyProjects I then right click a project and open with Vs code and everything works as expected.... 

I have no understanding as to why this is a issue.. If someone could explain Please

brianignacio5, 2023-02-23:
Hi @Uvamosk what is the content of your `vscode/c_cpp_properties.json` file ? This seems to be related to the C/C++ extension itself and not much we can do on our side besides making sure that both configuration files are correct.

Uvamosk, 2023-02-23:
I apologize, I didn't close this.
It is the extension. They confirmed it. It has something to do with it
symbolic links in linux.