espressif/vscode-esp-idf-extension #890: [Terminal] terminal shows hieroglyphs when baud rate is 460800

chipweinberger, 2023-02-17:
### OS

MacOS

### Operating System version

macOS 12.6.3

### Visual Studio Code version

1.75.1

### ESP-IDF version

v4.4.3

### Python version

Python 3.9.9

### Doctor command output

charlesweinberger@Chips-Macbook-Pro-2 jamcorder_app % idf.py doctor
Executing action: doctor
ninja: error: unknown target 'doctor', did you mean 'monitor'?
command "doctor" is not known to idf.py and is not a Ninja target

### Description

Is there any way to have VSC automatically change BAUD?

### Debug Message

```plain
idf_monitor.py -p /dev/cu.SLAB_USBtoUART -b 460800 --toolchain-prefix xtensa-esp32s3-elf- --target esp32s3 /V
olumes/User/MBP-Google-Drive/jamcorder/firmware/jamcorder-firmware/jamcorder_app/build/jJamApp.elf
--- idf_monitor on /dev/cu.SLAB_USBtoUART 460800 ---
--- Quit: Ctrl+] | Menu: Ctrl+T | Help: Ctrl+T followed by Ctrl+H ---
xx�xxxx��x��x��x��x�x�xx�������x��xx�������x�����������x�x�xxx��x�x�x��x�xx�x���������������x�x�x��x�x�x���x��xx��xxxxx�x��x�x�xx�x�x�x�x���xx�x��xxx�xx�xx�xxxx�x�xx�x�xx�x�x�xxx�xx�x��xxx�xxxx����xxxxx��xx�x�x�xxx��x��x�xx��x�x�xxxx��x��x�x��x�x�x�x�xx�x���x����xxxx���xx�x�xxx�x���x��xxx�x�x�x�xx�x���x�������x�������x�xxx�x���x�x�xx��x�x�x�xx�x���x�������xx����x�xxx�x���x��x���x��xx�x�x�x�x�x��x�������x���x���x�x����x����������x�xxx�x����xx��x�x�x�xxx�x�xx�������x������x�xxxx�xxx���x��x�x�xxxx�x�x�x�x�xxx�x�x�x�x�x�xxx�xxx�xx�x�x�x���xx�xxxx�xxx��x�x�x�x���x�x��x�xx�x�x�x�x�xx�x�xx�x��x�x�x�x�x�x�x�x�xxx�xxx�x��x�xxx�x�x�x�x��x�xxx�x���x�x�x�x�x�x�x��x�xxx�xx�x�x���xxxxxx�x��x�x�xx��x�x�x�xxxx�xxx��x�xx���x�x�x�xx�����x�x�x��xI (44) boot: ESP-IDF v4.4.3-220-gd65ef269d7 2nd stage bootloader
I (44) boot: compile time 05:22:35
I (44) boot: chip revision: 0
I (45) boot.esp32s3: Boot SPI Speed : 80MHz
I (46) boot.esp32s3: SPI Mode       : DIO
I (47) boot.esp32s3: SPI Flash Size : 8MB
I (48) boot: Enabling RNG early entropy source...
I (50) boot: Partition Table:
```

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

moefear85, 2023-02-19:
Though not a bug, I'd also like this feature. But how would the extension know when the target chip has changed baudrate? There is no definite implementation for this. During flashing, esptool and the target chip communicate with each other in-band and agree to switch baudrates simultaneously. But during booting, then the 1st stage bootloader would have to send a specially formatted statement that idf-monitor can then detect and parse so it knows to change the port's baudrate and to what value to change it to. Although the actual value can be coded in advance, the monitor would still have to guess at which exact moment the 1st stage bootloader has ended, then reconfigure the port at the right moment. It can't do this retroactively, since the data would already have physically arrived garbled, meaning the esp would also have to wait some delay or expect a signal from the monitor before moving to stage 2.
One would modify stage 2 instead, which is easier, but this would still only work in cases the esp is reset by the monitor, not if reset by a button press or software resets, since the monitor wouldn't know that the esp has moved back to the boot baudrate. This means the esp would have to be programmed to also notify the monitor then wait a delay before actually resetting.

chipweinberger, 2023-02-19:
Yes i agree It's not a simple problem.

I know idf.py uses baud rate negotiation. 

If the device reboots, the panic handler could print a known string at the current baud before it reboots.

chipweinberger, 2023-02-19:
Also, perhaps I should file this in esp-idf?

idf.py monitor is implemented there.

moefear85, 2023-02-19:
yeah i guess, but as a feature request

igrr, 2023-02-19:
You can disable ROM bootloader output using either an eFuse (permanently) or a strapping pin (temporarily). This will eliminate the garbage on UART.

chipweinberger, 2023-02-20:
@igrr Appreciate the suggestion. 

Would love a solution that allows me to see the logs. 

I already re-log the reset reason at a later time, which is an improvement. 

Is there nothing that can be done? I imagine most OS can switch baud quickly.

closed 2023-03-14 by github-actions[bot]