espressif/vscode-esp-idf-extension #891: unable to start debugging,the value of miDebuggerPath is invalid

poopoofoo, 2023-02-17:
### OS

Windows

### Operating System version

win11

### Visual Studio Code version

newest

### ESP-IDF version

4.4.3

### Python version

3.8.7

### Doctor command output

1: (66) LaunchOptions{"name":"ESP32-S3 DEBUG","type":"cppdbg","request":"launch","MIMode":"gdb","miDebuggerPath":"D:\\esp\\.espressif\\tools\\xtensa-esp-elf-gdb\\11.2_20220823\\xtensa-esp-elf-gdb\\bin\\xtensa-esp32s3-elf-gdb.exe\r\nD:\\esp\\.espressif\\tools\\xtensa-esp32s3-elf\\esp-2022r1-11.2.0\\xtensa-esp32s3-elf\\bin\\xtensa-esp32s3-elf-gdb.exe","program":"C:\\Users\\Dell\\blink\\build\\blink.elf","cwd":"C:\\Users\\Dell\\blink","environment":[{"name":"PATH","value":"D:\\esp\\.espressif\\tools\\xtensa-esp-elf-gdb\\11.2_20220823\\xtensa-esp-elf-gdb\\bin;D:\\esp\\.espressif\\tools\\riscv32-esp-elf-gdb\\11.2_20220823\\riscv32-esp-elf-gdb\\bin;D:\\esp\\.espressif\\tools\\xtensa-esp32-elf\\esp-2022r1-11.2.0\\xtensa-esp32-elf\\bin;D:\\esp\\.espressif\\tools\\xtensa-esp32s2-elf\\esp-2022r1-11.2.0\\xtensa-esp32s2-elf\\bin;D:\\esp\\.espressif\\tools\\xtensa-esp32s3-elf\\esp-2022r1-11.2.0\\xtensa-esp32s3-elf\\bin;D:\\esp\\.espressif\\tools\\riscv32-esp-elf\\esp-2022r1-11.2.0\\riscv32-esp-elf\\bin;D:\\esp\\.espressif\\tools\\esp32ulp-elf\\2.35_20220830\\esp32ulp-elf\\bin;D:\\esp\\.espressif\\tools\\cmake\\3.24.0\\bin;D:\\esp\\.espressif\\tools\\openocd-esp32\\v0.11.0-esp32-20221026\\openocd-esp32\\bin;D:\\esp\\.espressif\\tools\\ninja\\1.10.2;D:\\esp\\.espressif\\tools\\idf-exe\\1.0.3;D:\\esp\\.espressif\\tools\\ccache\\4.6.2\\ccache-4.6.2-windows-x86_64;D:\\esp\\.espressif\\tools\\dfu-util\\0.9\\dfu-util-0.9-win64"}],"setupCommands":[{"text":"target remote :3333"},{"text":"set remote hardware-watchpoint-limit 2"},{"text":"mon reset halt"},{"text":"thb app_main"},{"text":"flushregs"}],"externalConsole":false,"logging":{"engineLogging":true},"__configurationTarget":6,"__sessionId":"7f2ecfa3-2113-4128-a244-e41972379344"}
1: (110) Send Event AD7MessageEvent

### Description

when i start debug ，the miDebuggerPath show two path stick together like this:
"miDebuggerPath":"D:\\esp\\.espressif\\tools\\xtensa-esp-elf-gdb\\11.2_20220823\\xtensa-esp-elf-gdb\\bin\\xtensa-esp32s3-elf-gdb.exe\r\nD:\\esp\\.espressif\\tools\\xtensa-esp32s3-elf\\esp-2022r1-11.2.0\\xtensa-esp32s3-elf\\bin\\xtensa-esp32s3-elf-gdb.exe"

### Debug Message

```plain
xx
```

### Other Steps to Reproduce

xx

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

brianignacio5, 2023-02-24:
I tried to reproduce your issue in Windows but couldn't reproduce using latest esp-idf master.

Could you try using the nightly version of this extension [here](https://nightly.link/espressif/vscode-esp-idf-extension/workflows/ci/master/esp-idf-extension.vsix.zip) ? 

Also please share the required troubleshooting information such as the extension and doctor command output. To understand your setup and points of error they might exists.

brianignacio5, 2023-03-16:
Just need to update the `miDebuggerPath` in your launch.json to:

`"miDebuggerPath":"D:\esp\.espressif\tools\xtensa-esp-elf-gdb\11.2_20220823\xtensa-esp-elf-gdb\bin\xtensa-esp32s3-elf-gdb.exe"`

or

`"miDebuggerPath":"D:\esp\.espressif\tools\xtensa-esp32s3-elf\esp-2022r1-11.2.0\xtensa-esp32s3-elf\bin\xtensa-esp32s3-elf-gdb.exe"`

closed 2023-03-16 by brianignacio5