espressif/vscode-esp-idf-extension #896: Cannot select Use existing setup with ESP-IDF v5.0

csmith-morningstar, 2023-02-23:
### OS

Windows

### Operating System version

Windows 10

### Visual Studio Code version

1.75.1

### ESP-IDF version

5.0.1

### Python version

3.8.7

### Doctor command output

---------------------------------------------- ESP-IDF Extension for Visual Studio Code report ---------------------------------------------
OS win32 x64 10.0.19044 
System environment variable IDF_PYTHON_ENV_PATH 
 undefined 
System environment variable PATH 
 C:\Program Files\Microsoft\jdk-17.0.6.10-hotspot\bin;C:\Program Files (x86)\Intel\Intel(R) Management Engine Components\iCLS\;C:\Program Files\Intel\Intel(R) Management Engine Components\iCLS\;C:\WINDOWS\system32;C:\WINDOWS;C:\WINDOWS\System32\Wbem;C:\WINDOWS\System32\WindowsPowerShell\v1.0\;C:\WINDOWS\System32\OpenSSH\;C:\Program Files (x86)\Intel\Intel(R) Management Engine Components\DAL;C:\Program Files\Intel\Intel(R) Management Engine Components\DAL;C:\Program Files (x86)\Intel\Intel(R) Management Engine Components\IPT;C:\Program Files\Intel\Intel(R) Management Engine Components\IPT;C:\Program Files\Intel\WiFi\bin\;C:\Program Files\Common Files\Intel\WirelessCommon\;C:<HOMEPATH>\AppData\Local\Programs\Python\Python310;C:<HOMEPATH>\AppData\Local\Programs\Python\Python310\Scripts\;C:\Program Files\Git\cmd;C:\Program Files\PuTTY\;C:\Program Files (x86)\PDFtk\bin\;C:\Program Files\CMake\bin;C:\msys64\mingw64\bin;C:\Program Files\Docker\Docker\resources\bin;C:\tools\kotlinc\bin;C:<HOMEPATH>\AppData\Local\Programs\Python\Python310\Scripts\;C:<HOMEPATH>\AppData\Local\Programs\Python\Python310\;C:<HOMEPATH>\AppData\Local\Microsoft\WindowsApps;C:<HOMEPATH>\AppData\Local\Programs\Microsoft VS Code\bin; 
System environment variable PYTHON 
 undefined 
Visual Studio Code version 1.75.1 
Visual Studio Code language en 
Visual Studio Code shell C:\WINDOWS\System32\WindowsPowerShell\v1.0\powershell.exe 
ESP-IDF Extension version 1.5.1 
Workspace folder c:<HOMEPATH>\Devel\esp-firmware\app 
---------------------------------------------------- Extension configuration settings ------------------------------------------------------
ESP-ADF Path (idf.espAdfPath) ${env:ADF_PATH}
ESP-IDF Path (idf.espIdfPath) C:/Espressif/frameworks/esp-idf-v5.0.1/
ESP-MDF Path (idf.espMdfPath) ${env:MDF_PATH}
ESP-Matter Path (idf.espMatterPath) ${env:ESP_MATTER_PATH}
Custom extra paths (idf.customExtraPaths) C:\Espressif\tools\xtensa-esp-elf-gdb\11.2_20220823\xtensa-esp-elf-gdb\bin;C:\Espressif\tools\riscv32-esp-elf-gdb\11.2_20220823\riscv32-esp-elf-gdb\bin;C:\Espressif\tools\xtensa-esp32-elf\esp-2022r1-11.2.0\xtensa-esp32-elf\bin;C:\Espressif\tools\xtensa-esp32s2-elf\esp-2022r1-11.2.0\xtensa-esp32s2-elf\bin;C:\Espressif\tools\xtensa-esp32s3-elf\esp-2022r1-11.2.0\xtensa-esp32s3-elf\bin;C:\Espressif\tools\riscv32-esp-elf\esp-2022r1-11.2.0\riscv32-esp-elf\bin;C:\Espressif\tools\esp32ulp-elf\2.35_20220830\esp32ulp-elf\bin;C:\Espressif\tools\cmake\3.24.0\bin;C:\Espressif\tools\openocd-esp32\v0.11.0-esp32-20221026\openocd-esp32\bin;C:\Espressif\tools\ninja\1.10.2;C:\Espressif\tools\idf-exe\1.0.3;C:\Espressif\tools\ccache\4.6.2\ccache-4.6.2-windows-x86_64;C:\Espressif\tools\dfu-util\0.9\dfu-util-0.9-win64
Custom extra vars (idf.customExtraVars) {"OPENOCD_SCRIPTS":"C:\\Espressif\\tools\\openocd-esp32\\v0.11.0-esp32-20221026/openocd-esp32/share/openocd/scripts","IDF_CCACHE_ENABLE":"1"}
Virtual env Python Path (idf.pythonBinPath) C:/Espressif/python_env/idf5.0_py3.8_env/Scripts/python.exe
Serial port (idf.port) COM1
OpenOCD Configs (idf.openOcdConfigs) board/flyswatter2.cfg
ESP-IDF Tools Path (idf.toolsPath) C:\Espressif
Git Path (idf.gitPath) C:/Espressif/tools/idf-git/2.34.2/cmd/git.exe
-------------------------------------------------------- Configurations access -------------------------------------------------------------
Access to ESP-ADF Path (idf.espAdfPath) false
Access to ESP-IDF Path (idf.espIdfPath) true
Access to ESP-MDF Path (idf.espMdfPath) false
Access to ESP-Matter Path (idf.espMatterPath) false
Access to ESP-IDF Custom extra paths
Access to C:\Espressif\tools\xtensa-esp-elf-gdb\11.2_20220823\xtensa-esp-elf-gdb\bin: true
Access to C:\Espressif\tools\riscv32-esp-elf-gdb\11.2_20220823\riscv32-esp-elf-gdb\bin: true
Access to C:\Espressif\tools\xtensa-esp32-elf\esp-2022r1-11.2.0\xtensa-esp32-elf\bin: true
Access to C:\Espressif\tools\xtensa-esp32s2-elf\esp-2022r1-11.2.0\xtensa-esp32s2-elf\bin: true
Access to C:\Espressif\tools\xtensa-esp32s3-elf\esp-2022r1-11.2.0\xtensa-esp32s3-elf\bin: true
Access to C:\Espressif\tools\riscv32-esp-elf\esp-2022r1-11.2.0\riscv32-esp-elf\bin: true
Access to C:\Espressif\tools\esp32ulp-elf\2.35_20220830\esp32ulp-elf\bin: true
Access to C:\Espressif\tools\cmake\3.24.0\bin: true
Access to C:\Espressif\tools\openocd-esp32\v0.11.0-esp32-20221026\openocd-esp32\bin: true
Access to C:\Espressif\tools\ninja\1.10.2: true
Access to C:\Espressif\tools\idf-exe\1.0.3: true
Access to C:\Espressif\tools\ccache\4.6.2\ccache-4.6.2-windows-x86_64: true
Access to C:\Espressif\tools\dfu-util\0.9\dfu-util-0.9-win64: true
Access to Virtual env Python Path (idf.pythonBinPath) true
Access to CMake in environment PATH undefined
Access to Ninja in environment PATH undefined
Access to ESP-IDF Tools Path (idf.toolsPath) true
[... 124 lines cut ...]

I tried installing ESP-IDF v4.4.4 (which I had previously uninstalled) and the option showed back up, but only v4.4.4. Uninstalling both versions, and re-installing only v5.0.1 caused the option to disappear again.

I was able to build and flash code by manually setting the following settings:

```
	"idf.espIdfPathWin": "C:/Espressif/frameworks/esp-idf-v5.0.1/",
	"idf.pythonBinPathWin": "C:/Espressif/python_env/idf5.0_py3.8_env/Scripts/python.exe",
	"idf.customExtraPaths": "C:\\Espressif\\tools\\xtensa-esp-elf-gdb\\11.2_20220823\\xtensa-esp-elf-gdb\\bin;C:\\Espressif\\tools\\riscv32-esp-elf-gdb\\11.2_20220823\\riscv32-esp-elf-gdb\\bin;C:\\Espressif\\tools\\xtensa-esp32-elf\\esp-2022r1-11.2.0\\xtensa-esp32-elf\\bin;C:\\Espressif\\tools\\xtensa-esp32s2-elf\\esp-2022r1-11.2.0\\xtensa-esp32s2-elf\\bin;C:\\Espressif\\tools\\xtensa-esp32s3-elf\\esp-2022r1-11.2.0\\xtensa-esp32s3-elf\\bin;C:\\Espressif\\tools\\riscv32-esp-elf\\esp-2022r1-11.2.0\\riscv32-esp-elf\\bin;C:\\Espressif\\tools\\esp32ulp-elf\\2.35_20220830\\esp32ulp-elf\\bin;C:\\Espressif\\tools\\cmake\\3.24.0\\bin;C:\\Espressif\\tools\\openocd-esp32\\v0.11.0-esp32-20221026\\openocd-esp32\\bin;C:\\Espressif\\tools\\ninja\\1.10.2;C:\\Espressif\\tools\\idf-exe\\1.0.3;C:\\Espressif\\tools\\ccache\\4.6.2\\ccache-4.6.2-windows-x86_64;C:\\Espressif\\tools\\dfu-util\\0.9\\dfu-util-0.9-win64",
	"idf.customExtraVars": "{\"OPENOCD_SCRIPTS\":\"C:\\\\Espressif\\\\tools\\\\openocd-esp32\\\\v0.11.0-esp32-20221026/openocd-esp32/share/openocd/scripts\",\"IDF_CCACHE_ENABLE\":\"1\"}",
```

### Debug Message

```plain
N/A
```

### Other Steps to Reproduce

Install ESP-IDF v5.0+ and try to configure with Use existingsetup

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

brianignacio5, 2023-02-24:
Seems there is an issue with your Python packages. The extension log might provide more information about the error.

Could you try installing additional extension packages with the `Install ESP-IDF Python Packages` command.

csmith-morningstar, 2023-02-24:
> Seems there is an issue with your Python packages. The extension log might provide more information about the error.
> 
> Could you try installing additional extension packages with the `Install ESP-IDF Python Packages` command.

I tried running the command and got

```
ERROR: Can not perform a '--user' install. User site-packages are not visible in this virtualenv.
Command failed: "C:/Espressif/python_env/idf5.0_py3.8_env/Scripts/python.exe" -m pip install --user virtualenv
ERROR: Can not perform a '--user' install. User site-packages are not visible in this virtualenv
```

Here is the log message

```
{"user":true,"message":"Open a folder first.","stack":"Error: PRECHECK_FAILED\n    at \\dist\\extension.js:1:4934\n    at Array.forEach (<anonymous>)\n    at Function.perform (\\dist\\extension.js:1:4887)\n    at \\dist\\extension.js:1:1226400\n    at \\dist\\extension.js:1:1209446\n    at s.h (\\out\\vs\\workbench\\api\\node\\extensionHostProcess.js:96:108016)\n    at s.$executeContributedCommand (\\out\\vs\\workbench\\api\\node\\extensionHostProcess.js:96:108744)\n    at l.N (\\out\\vs\\workbench\\api\\node\\extensionHostProcess.js:106:11239)\n    at l.M (\\out\\vs\\workbench\\api\\node\\extensionHostProcess.js:106:10957)\n    at l.H (\\out\\vs\\workbench\\api\\node\\extensionHostProcess.js:106:10044)\n    at l.G (\\out\\vs\\workbench\\api\\node\\extensionHostProcess.js:106:9022)\n    at \\out\\vs\\workbench\\api\\node\\extensionHostProcess.js:106:7810\n    at f.invoke (\\out\\vs\\workbench\\api\\node\\extensionHostProcess.js:63:145)\n    at g.deliver (\\out\\vs\\workbench\\api\\node\\extensionHostProcess.js:63:2029)\n    at d.fire (\\out\\vs\\workbench\\api\\node\\extensionHostProcess.js:63:1667)\n    at p.fire (\\out\\vs\\workbench\\api\\node\\extensionHostProcess.js:72:14861)\n    at \\out\\vs\\workbench\\api\\node\\extensionHostProcess.js:122:15815\n    at f.invoke (\\out\\vs\\workbench\\api\\node\\extensionHostProcess.js:63:145)\n    at g.deliver (\\out\\vs\\workbench\\api\\node\\extensionHostProcess.js:63:2029)\n    at d.fire (\\out\\vs\\workbench\\api\\node\\extensionHostProcess.js:63:1667)\n    at p.fire (\\out\\vs\\workbench\\api\\node\\extensionHostProcess.js:72:14861)\n    at MessagePortMain.<anonymous> (\\out\\vs\\workbench\\api\\node\\extensionHostProcess.js:122:13941)\n    at MessagePortMain.emit (node:events:526:28)\n    at Object.MessagePortMain._internalPort.emit (node:electron/js2c/utility_init:5:364)","level":"error","timestamp":"2023-02-24T16:41:35.963Z"}
{"message":"Command failed: \"C:/Espressif/python_env/idf5.0_py3.8_env/Scripts/python.exe\" -m pip install --user virtualenv\nERROR: Can not perform a '--user' install. User site-packages are not visible in this virtualenv.\r\n","stack":"Error: Command failed: \"C:/Espressif/python_env/idf5.0_py3.8_env/Scripts/python.exe\" -m pip install --user virtualenv\nERROR: Can not perform a '--user' install. User site-packages are not visible in this virtualenv.\r\n\n    at ChildProcess.exithandler (node:child_process:408:12)\n    at ChildProcess.emit (node:events:526:28)\n    at maybeClose (node:internal/child_process:1092:16)\n    at Process.ChildProcess._handle.onexit (node:internal/child_process:302:5)","level":"error","timestamp":"2023-02-24T16:41:55.352Z"}
{"user":true,"message":"Command failed: \"C:/Espressif/python_env/idf5.0_py3.8_env/Scripts/python.exe\" -m pip install --user virtualenv\nERROR: Can not perform a '--user' install. User site-packages are not visible in this virtualenv.\r\n","stack":"Error: Command failed: \"C:/Espressif/python_env/idf5.0_py3.8_env/Scripts/python.exe\" -m pip install --user virtualenv\nERROR: Can not perform a '--user' install. User site-packages are not visible in this virtualenv.\r\n\n    at ChildProcess.exithandler (node:child_process:408:12)\n    at ChildProcess.emit (node:events:526:28)\n    at maybeClose (node:internal/child_process:1092:16)\n    at Process.ChildProcess._handle.onexit (node:internal/child_process:302:5)","level":"error","timestamp":"2023-02-24T16:41:55.353Z"}
```

brianignacio5, 2023-03-16:
The `Install ESP-IDF Python Packages` should be already be fixed in master and latest release. Could you take a look?

brianignacio5, 2023-04-03:
Closing this issue for lack of response. Comment here for any additional issues.

closed 2023-04-03 by brianignacio5