espressif/vscode-esp-idf-extension #897: Command Palette commands not working.

bslisowski, 2023-02-23:
### OS

MacOS

### Operating System version

macOS Monterey Version 12.5.1

### Visual Studio Code version

1.75.1

### ESP-IDF version

5.0

### Python version

3.10.8

### Doctor command output

---------------------------------------------- ESP-IDF Extension for Visual Studio Code report ---------------------------------------------
OS darwin arm64 21.6.0 
System environment variable IDF_PYTHON_ENV_PATH 
 undefined 
System environment variable PATH 
 /opt/homebrew/bin:/opt/homebrew/sbin:/usr/local/bin:/usr/bin:/bin:/usr/sbin:/sbin:/Library/Apple/usr/bin:<HOMEPATH>/Library/Android/sdk/emulator:<HOMEPATH>/Library/Android/sdk/platform-tools:<HOMEPATH>/Documents/Flutter/flutter/bin 
System environment variable PYTHON 
 undefined 
Visual Studio Code version 1.75.1 
Visual Studio Code language en 
Visual Studio Code shell /bin/zsh 
ESP-IDF Extension version 1.5.1 
Workspace folder <HOMEPATH>/Documents/testing/vendor_server 
---------------------------------------------------- Extension configuration settings ------------------------------------------------------
ESP-ADF Path (idf.espAdfPath) ${env:ADF_PATH}
ESP-IDF Path (idf.espIdfPath) <HOMEPATH>/esp/esp-idf
ESP-MDF Path (idf.espMdfPath) ${env:MDF_PATH}
ESP-Matter Path (idf.espMatterPath) ${env:ESP_MATTER_PATH}
Custom extra paths (idf.customExtraPaths) <HOMEPATH>/.espressif/tools/xtensa-esp-elf-gdb/11.2_20220823/xtensa-esp-elf-gdb/bin:<HOMEPATH>/.espressif/tools/riscv32-esp-elf-gdb/11.2_20220823/riscv32-esp-elf-gdb/bin:<HOMEPATH>/.espressif/tools/xtensa-esp32-elf/esp-2022r1-11.2.0/xtensa-esp32-elf/bin:<HOMEPATH>/.espressif/tools/xtensa-esp32s2-elf/esp-2022r1-11.2.0/xtensa-esp32s2-elf/bin:<HOMEPATH>/.espressif/tools/xtensa-esp32s3-elf/esp-2022r1-11.2.0/xtensa-esp32s3-elf/bin:<HOMEPATH>/.espressif/tools/riscv32-esp-elf/esp-2022r1-11.2.0/riscv32-esp-elf/bin:<HOMEPATH>/.espressif/tools/esp32ulp-elf/2.35_20220830/esp32ulp-elf/bin:<HOMEPATH>/.espressif/tools/cmake/3.24.0/CMake.app/Contents/bin:<HOMEPATH>/.espressif/tools/openocd-esp32/v0.11.0-esp32-20220706/openocd-esp32/bin:<HOMEPATH>/.espressif/tools/ninja/1.10.2
Custom extra vars (idf.customExtraVars) {"OPENOCD_SCRIPTS":"<HOMEPATH>/.espressif/tools/openocd-esp32/v0.11.0-esp32-20220706/openocd-esp32/share/openocd/scripts"}
Virtual env Python Path (idf.pythonBinPath) <HOMEPATH>/.espressif/python_env/idf5.0_py3.10_env/bin/python
Serial port (idf.port) /dev/cu.usbserial-1110
OpenOCD Configs (idf.openOcdConfigs) interface/ftdi/esp32_devkitj_v1.cfg,board/esp32-wrover.cfg
ESP-IDF Tools Path (idf.toolsPath) <HOMEPATH>/.espressif
Git Path (idf.gitPath) /usr/bin/git
-------------------------------------------------------- Configurations access -------------------------------------------------------------
Access to ESP-ADF Path (idf.espAdfPath) false
Access to ESP-IDF Path (idf.espIdfPath) true
Access to ESP-MDF Path (idf.espMdfPath) false
Access to ESP-Matter Path (idf.espMatterPath) false
Access to ESP-IDF Custom extra paths
Access to <HOMEPATH>/.espressif/tools/xtensa-esp-elf-gdb/11.2_20220823/xtensa-esp-elf-gdb/bin: true
Access to <HOMEPATH>/.espressif/tools/riscv32-esp-elf-gdb/11.2_20220823/riscv32-esp-elf-gdb/bin: true
Access to <HOMEPATH>/.espressif/tools/xtensa-esp32-elf/esp-2022r1-11.2.0/xtensa-esp32-elf/bin: true
Access to <HOMEPATH>/.espressif/tools/xtensa-esp32s2-elf/esp-2022r1-11.2.0/xtensa-esp32s2-elf/bin: true
Access to <HOMEPATH>/.espressif/tools/xtensa-esp32s3-elf/esp-2022r1-11.2.0/xtensa-esp32s3-elf/bin: true
Access to <HOMEPATH>/.espressif/tools/riscv32-esp-elf/esp-2022r1-11.2.0/riscv32-esp-elf/bin: true
Access to <HOMEPATH>/.espressif/tools/esp32ulp-elf/2.35_20220830/esp32ulp-elf/bin: true
Access to <HOMEPATH>/.espressif/tools/cmake/3.24.0/CMake.app/Contents/bin: true
Access to <HOMEPATH>/.espressif/tools/openocd-esp32/v0.11.0-esp32-20220706/openocd-esp32/bin: true
Access to <HOMEPATH>/.espressif/tools/ninja/1.10.2: true
Access to Virtual env Python Path (idf.pythonBinPath) true
Access to CMake in environment PATH undefined
Access to Ninja in environment PATH undefined
Access to ESP-IDF Tools Path (idf.toolsPath) true
-------------------------------------------------------- Configurations has spaces -------------------------------------------------------------
Spaces in system environment Path undefined
Spaces in ESP-ADF Path (idf.espAdfPath) undefined
Spaces in ESP-IDF Path (idf.espIdfPath) undefined
Spaces in ESP-MDF Path (idf.espMdfPath) undefined
Spaces in ESP-Matter Path (idf.espMatterPath) undefined
Spaces in ESP-IDF Custom extra paths
Spaces in Virtual env Python Path (idf.pythonBinPath) undefined
Spaces in ESP-IDF Tools Path (idf.toolsPath) undefined
----------------------------------------------------------- Executables Versions -----------------------------------------------------------
Git version undefined
ESP-IDF version undefined
Python version undefined
Python's pip version undefined
-------------------------------------------------- Python packages in idf.pythonBinPath ----------------------------------------------------
---------------------------------------------------- Check ESP-IDF python requirements.txt -------------------------------------------------
Check ESP-IDF Python packages undefined
---------------------------------------------------- Check extension requirements.txt ------------------------------------------------------
Check Extension Python packages undefined
---------------------------------------------------- Check ESP-IDF debug adapter requirements.txt ------------------------------------------
Check Debug AdapterPython packages undefined
----------------------------------------------------------- Latest error -----------------------------------------------------------------
Latest error at Unknown error in ESP-IDF doctor command
--------------------------------------------------------------------------------------------------------------------------------------------

### Extension

[esp_idf_vsc_ext.log](https://github.com/espressif/vscode-esp-idf-extension/files/10816249/esp_idf_vsc_ext.log)

### Description

Some command palette prompts do nothing, including CMakeLists.txt Editor and Open Partition Table Editor UI.

### Debug Message

```plain
None
```

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

brianignacio5, 2023-02-24:
It is a bit weird, it seems that the command `which cmake` is failing. I'm not sure why?

If you open any terminal, what happens when you run `which cmake` ?

Also when you try to build for example, what is the output ?

bslisowski, 2023-02-24:
Both which cmake and build seem to work fine.

/Users/admin/.espressif/tools/cmake/3.24.0/CMake.app/Contents/bin/cmake

Executing task: ninja  

[1/9] Performing build step for 'bootloader'
[1/1] cd /Users/admin/Documents/testing/vendor_server/build/bootloader/esp-idf/esptool_py && /Users/admin/.espressif/python_env/idf5.0_py3.10_env/bin/python /Users/admin/esp/esp-idf/components/partition_table/check_sizes.py --offset 0x8000 bootloader 0x1000 /Users/admin/Documents/testing/vendor_server/build/bootloader/bootloader.bin
Bootloader binary size 0x66f0 bytes. 0x910 bytes (8%) free.
[6/7] Generating binary image from built executable
esptool.py v4.4
Creating esp32 image...
Merged 25 ELF sections
Successfully created esp32 image.
Generated /Users/admin/Documents/testing/vendor_server/build/vendor_server.bin
[7/7] cd /Users/admin/Documents/testing/vendor_server/build/esp-idf/esptool_py && /Users/.../partition-table.bin /Users/admin/Documents/testing/vendor_server/build/vendor_server.bin
vendor_server.bin binary size 0x1047c0 bytes. Smallest app partition is 0x177000 bytes. 0x72840 bytes (31%) free.

brianignacio5, 2023-02-28:
So which command palette commands are not working ? 

Have you check the tutorials about CMakeLists.txt Editor and Open Partition Table Editor ?

damonearl, 2023-03-01:
Just installed a fresh install on MacOS as well, none of the command palettes are working can't even run the configuration.

On Ventura 13.2.1 and VSCode 1.76.0 and ESP-IDF 1.6.0

Just gonna use my other machine until this is resolved.

eos1d3, 2023-03-03:
On Mac, if you never install ESP-IDF extension before, it can never start and will pop up a windows saying starting error.

Once you manually install ESP-IDF extension 1.51, and then upgrade to 1.6.0, it will start.

BUT, there are many many different other errors. For example, when creating a project, the Create button never response at all. I click 10 times and the screen never change.

And when building, there are different kinds of errors that I cannot remember. 

So, quickly install ESP-IDF extension 1.5.1. This happens to my two mac!

dizcza, 2023-03-04:
Same for me. ESP config command palette does not and clicking on any esp-related buttons on the bottom panel raises the errors:
```
command 'espIdf.selectPort' not found
command 'espIdf.createIdfTerminal' not found
```
As @eos1d3 suggested, I switched back to v1.5.1. But it means I'll have this issue again https://github.com/espressif/vscode-esp-idf-extension/issues/823. Hope you guys fix it soon.

SEIDO-JP, 2023-03-05:
I was just wanted to confirm - the same problem for me. Macbook Air M2, newest system (Ventura 13.2.1), latest VSCode and extension 1.6.0 not working. You can install it, but cannot use any command from the command palette. But 1.5.1 is working fine.

barjonas, 2023-03-09:
I can confirm a similar experience on Windows 11. My first contact with this extension was 1.6.0. I could not find a single command which would work, so it was impossible to follow the installation instructions. Downgraded to 1.5.1, everything seems to be working as documented.

brianignacio5, 2023-03-17:
Could you try this [vsix installer](https://nightly.link/espressif/vscode-esp-idf-extension/actions/artifacts/601852097.zip) to see if the issue is fixed ?

PR #909 (2023-04-04): catch cmakelists txt error sdkconfig file access
## Description

Catch CMakeLists.txt file access error reading IDF Target on extension activation.

Fixes #897 
Fix #905 
Fix #907
Fix #911 

## Type of change

Please delete options that are not relevant.
- Bug fix (non-breaking change which fixes an issue)

## How has this been tested?

Manual testing

**Test Configuration**:
* ESP-IDF Version: 5.0
* OS (Windows,Linux and macOS): macOS Ventura

## Checklist
- [x] PR Self Reviewed
- [x] Applied Code formatting
- [ ] Added Documentation
- [ ] Added Unit Test
- [ ] Verified on all platforms - Win [... cut]
closed 2023-04-04 by brianignacio5