espressif/vscode-esp-idf-extension #903: [Feature Request]: Allowing only one monitor terminal window

SilverBut, 2023-03-05:
**Is your feature request related to a problem? Please describe.**
[image]

When developing projects, this 'Build, Flash and Monitor' button is helpful. However, for each time I clicked it, a new monitor window appears, instead of reusing old ones. It's easy to became this after dozens of compiling, which is quite annoying.

[image]

**Describe the solution you'd like & alternatives you've considered**

When create monitor window, check if it already exists before creation, like how ESP-IDF Size/Flash works.

https://github.com/espressif/vscode-esp-idf-extension/blob/d2ca138efad73249cd12dd2b9656695d476d36d3/src/espIdf/size/idfSizePanel.ts#L32-L35

And if already exists, close it, or reuse it to run monitor command.

If necessary, only do the check when called automatically via "Build, Flash and Monitor" button, so users can still create multiple monitor window manually.

SilverBut, 2023-03-25:
not stale. still need this feature.

SilverBut, 2023-04-11:
Any maintainer?

brianignacio5, 2023-04-26:
@SilverBut Could you try this [vsix installer](https://nightly.link/espressif/vscode-esp-idf-extension/actions/artifacts/666045721.zip) ?

PR #951 (2023-04-28): add idf monitorBaudRate setting
## Description

Add `idf.monitorBaudRate` configuration setting to define IDF Monitor baud rate in the extension commands.

Delete existing IDF Monitor when re executing the IDF Monitor.

Fixes #926
Fix #903

## Type of change

- New feature (non-breaking change which adds functionality)

## Steps to test this pull request

Provide a list of steps to test changes in this PR and required output

1. Click on "settings.json" and update `idf.monitorBaudRate` to new value
2. Execute `ESP-IDF: Monitor your device` command to start IDF Monitor.
3. Observe results. Chec [... cut]
closed 2023-04-28 by brianignacio5