espressif/vscode-esp-idf-extension #904: "/usr/bin/python3 -m pip" is not valid. (ERROR_INVALID_PIP)

stolk, 2023-03-05:
### OS

Linux

### Operating System version

Ubuntu 22.04

### Visual Studio Code version

1.76.0

### ESP-IDF version

5.0.1

### Python version

Python 3.10.6

### Doctor command output

idf_versions.txt has (1) KB
idf_versions.txt progress: 100.00% (0.10 / 0.10) KB
idf_versions.txt has (1) KB
idf_versions.txt progress: 100.00% (0.10 / 0.10) KB
"/usr/bin/python3 -m pip" is not valid. (ERROR_INVALID_PIP)

### Extension

{"message":"Command failed: \"/usr/bin/python3\" -m pip --version\n/usr/bin/python3: No module named pip\n","stack":"Error: Command failed: \"/usr/bin/python3\" -m pip --version\n/usr/bin/python3: No module named pip\n\n\tat ChildProcess.exithandler (node:child_process:408:12)\n\tat ChildProcess.emit (node:events:526:28)\n\tat maybeClose (node:internal/child_process:1092:16)\n\tat Socket.<anonymous> (node:internal/child_process:451:11)\n\tat Socket.emit (node:events:526:28)\n\tat Pipe.<anonymous> (node:net:687:12)","level":"error","timestamp":"2023-03-05T22:56:55.317Z"}
{"user":true,"message":"Command failed: \"/usr/bin/python3\" -m pip --version\n/usr/bin/python3: No module named pip\n","stack":"Error: Command failed: \"/usr/bin/python3\" -m pip --version\n/usr/bin/python3: No module named pip\n\n\tat ChildProcess.exithandler (node:child_process:408:12)\n\tat ChildProcess.emit (node:events:526:28)\n\tat maybeClose (node:internal/child_process:1092:16)\n\tat Socket.<anonymous> (node:internal/child_process:451:11)\n\tat Socket.emit (node:events:526:28)\n\tat Pipe.<anonymous> (node:net:687:12)","level":"error","timestamp":"2023-03-05T22:56:55.318Z"}
{"user":true,"message":"\"/usr/bin/python3 -m pip\" is not valid. (ERROR_INVALID_PIP)","stack":"Error: \"/usr/bin/python3 -m pip\" is not valid. (ERROR_INVALID_PIP)\n\tat /dist/extension.js:1:235480\n\tat Generator.next (<anonymous>)\n\tat s (/dist/extension.js:1:228885)","level":"error","timestamp":"2023-03-05T22:56:55.320Z"}

### Description

Failed to configure.

### Debug Message

```plain
"/usr/bin/python3 -m pip" is not valid. (ERROR_INVALID_PIP)
```

### Other Steps to Reproduce

Download extension and configure 5.0.1 using /usr/bin/python3 from Ubuntu OS 22.04

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

ulexxander, 2023-03-07:
Also installed prerequisites and got the same error on `Linux Mint 21.1`

It seems that ESP IDF requires `pip`, but it is not listed in prerequisites, therefore was not installed.
```sh
$ /usr/bin/python3 -m pip
/usr/bin/python3: No module named pip
```

Googled how to install it and found this solution:
https://linuxize.com/post/how-to-install-pip-on-ubuntu-20.04/#installing-pip-for-python-3
```sh
$ sudo apt install python3-pip
```

Now command `/usr/bin/python3 -m pip` does not fail and installation succeeds.

Not sure however if there is any better way of installing `pip`, not familiar with Python.

This probably should be included in docs, right?

radurentea, 2023-03-15:
PIP is a package manager for Python packages.
It's strange because it should be included with Python versions 3.4 and higher.

dobairoland, 2023-03-15:
> PIP is a package manager for Python packages.
> It's strange because it should be included with Python versions 3.4 and higher.

There are some distributions which like to separate them, e.g. Debian/Ubuntu. On these systems `python3-pip` and `python3-venv` needs to be additionally installed.

brianignacio5, 2023-03-17:
I think this is a ESP-IDF documentation issue itself. Closing this issue and requested the documentation team to update these linux requirements.

closed 2023-03-17 by brianignacio5

hayschan, 2023-06-10:
I am running Windows 11 Pro. When I install ESP-IDF through Visual Studio Code (VSCode), I also encounter the same issue about invalid `pip`.

```python
"C:\Users\username\.espressif\tools\idf-python\3.8.7\python.exe -m pip" is not valid. (ERROR_INVALID_PIP)
```

Here is the detailed error log:

```python
Installing C:\Users\username\.espressif\tools\idf-git\2.30.1 ...
Warning File C:\Users\username\.espressif\tools\idf-git\2.30.1\cmd\git.exe
                                        already exists and was not updated.
Warning File C:\Users\username\.espressif\tools\idf-git\2.30.1\dev\_._
                                        already exists and was not updated.
Warning File C:\Users\username\.espressif\tools\idf-git\2.30.1\etc\bash.bash_logout
                                        already exists and was not updated.
Warning File C:\Users\username\.espressif\tools\idf-git\2.30.1\etc\bash.bashrc
                                        already exists and was not updated.
Warning File C:\Users\username\.espressif\tools\idf-git\2.30.1\etc\fstab
                                        already exists and was not updated.
Warning File C:\Users\username\.espressif\tools\idf-git\2.30.1\etc\gitattributes
                                        already exists and was not updated.
Warning File C:\Users\username\.espressif\tools\idf-git\2.30.1\etc\msystem
                                        already exists and was not updated.
Warning File C:\Users\username\.espressif\tools\idf-git\2.30.1\etc\nsswitch.conf
                                        already exists and was not updated.
Warning File C:\Users\username\.espressif\tools\idf-git\2.30.1\etc\profile
                                        already exists and was not updated.
Extracted C:\Users\username\.espressif\tools\idf-git\2.30.1 ...
Using existing C:\Users\username\.espressif\tools\idf-python\3.8.7
"C:\Users\username\.espressif\tools\idf-python\3.8.7\python.exe -m pip" is not valid. (ERROR_INVALID_PIP)
```

brianignacio5, 2023-06-12:
It seems there is an issue with your `C:\Users\username\.espressif\tools\idf-python` tool regarding. You can try to make sure that pip is installed correctly in this location or delete the directory and try the setup again.