espressif/vscode-esp-idf-extension #905: ESP-IDF extension install failed on macOS Ventura : command not found

FnxQT, 2023-03-08:
### OS

MacOS

### Operating System version

macOS 13.2.1

### Visual Studio Code version

1.76

### ESP-IDF version

4.1

### Python version

3.11.2

### Doctor command output

The doctor command does not event work

### Extension

Empty

### Description

I freshly installed the ESP-IDF extension on VS Code, I would like to configure it using the command palette and the "ESP-IDF: Configure ESP-IDF extension" command but it results in an error : command espIdf.setup.start not found.

If I try to run the doctor command, then I get a similar error message : command 'espIdf.doctorCommand' not found

It must be noted that I already have the ESP-IDF v4.1 installed in ~/esp/esp-idf/ and I have successfully compiled projects with it. I don't know if it may cause a problem.

### Debug Message

```plain
I have no debug message, since no command are found the extension has not generated anything.
```

### Other Steps to Reproduce

Freshly installed ESP-IDF Extension on macOS Ventura and no command are found.

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

FnxQT, 2023-03-09:
I switched to v.1.5.1 and it worked. Don't know why.

HamletLeon, 2023-03-09:
I had to downgrade to the 1.5.0. The "ESP-IDF: Configure ESP-IDF extension" didnt start on the 1.5.1 (Windows), it presents a "restart your windows" all the time.

dafeng1905, 2023-03-09:
same problem, downgrade works

anastazya, 2023-03-10:
> I switched to v.1.5.1 and it worked. Don't know why.

I can confirm ending up here after the same issues with ver. 1.6.1 on MacOSX Ventura. 1.5.1 seems to work.

brianignacio5, 2023-03-13:
Can you share the extension log ?

Check log file which can be obtained from:
Windows: %USERPROFILE%\.vscode\extensions\espressif.esp-idf-extension-VERSION\esp_idf_vsc_ext.log
Linux & MacOSX: $HOME/.vscode/extensions/espressif.esp-idf-extension-VERSION/esp_idf_vsc_ext.log

ChrisJones79, 2023-03-13:
@brianignacio5 I'm having the same issue. Here is my log.

[esp_idf_vsc_ext.log](https://github.com/espressif/vscode-esp-idf-extension/files/10958479/esp_idf_vsc_ext.log)

noisysoil, 2023-03-14:
Reported for Linux too, downgrading to 1.5.1 works: https://github.com/espressif/vscode-esp-idf-extension/issues/907

brianignacio5, 2023-03-17:
Could you try this [vsix installer](https://nightly.link/espressif/vscode-esp-idf-extension/actions/artifacts/601852097.zip) to see if the issue is fixed ?

fheckl, 2023-03-27:
> Could you try this [vsix installer](https://nightly.link/espressif/vscode-esp-idf-extension/actions/artifacts/601852097.zip) to see if the issue is fixed ?

I had the same issue on macos 13.2.1. After reading the solutions here I also downgraded to 1.5.1 and configured everything. That version works fine. 
Then I tried your manual installer but unfortunately it did not fix the issue.
So back to 1.5.1 for me.

ChrisJones79, 2023-03-27:
@brianignacio5 it worked for me. All commands I've tried work.

PR #909 (2023-04-04): catch cmakelists txt error sdkconfig file access
## Description

Catch CMakeLists.txt file access error reading IDF Target on extension activation.

Fixes #897 
Fix #905 
Fix #907
Fix #911 

## Type of change

Please delete options that are not relevant.
- Bug fix (non-breaking change which fixes an issue)

## How has this been tested?

Manual testing

**Test Configuration**:
* ESP-IDF Version: 5.0
* OS (Windows,Linux and macOS): macOS Ventura

## Checklist
- [x] PR Self Reviewed
- [x] Applied Code formatting
- [ ] Added Documentation
- [ ] Added Unit Test
- [ ] Verified on all platforms - Win [... cut]
closed 2023-04-04 by brianignacio5