espressif/vscode-esp-idf-extension #907: Extension v1.6.0 'Command not found' error - but works with extension v1.5.1.

noisysoil, 2023-03-14:
### OS

Linux

### Operating System version

Kubuntu 22.10

### Visual Studio Code version

1.76.1

### ESP-IDF version

5.0.1

### Python version

3.10.7

### Doctor command output

Doctor fails to launch with 'Command not found'.

### Extension

<root project directory> is my system's path to projects.

{"message":"Cannot access filePath: <root project directory>/Arduino/XIAO-ESP32-test/CMakeLists.txt","stack":"Error: ENOENT: no such file or directory, access '<root project directory>>

### Description

'Command not found' when invoking _any_ ESP-IDF extension command.

### Debug Message

```plain
Same as VSCode's extension error as outlined above:

<root project directory> is my system's path to projects.

{"message":"Cannot access filePath: <root project directory>/Arduino/XIAO-ESP32-test/CMakeLists.txt","stack":"Error: ENOENT: no such file or directory, access '<root project directory>>
```

### Other Steps to Reproduce

Following the 'CMakeLists.tst' not found error, I added one to the project and the extension still fails to run any command.

I have also tried re-installing the extension, removing config, etc.  However, v1.5.1 of the extension has no problems with any installation config situation.

The issue has been mentioned by other users, e.g. : [Command not found](https://esp32.com/viewtopic.php?f=40&t=32466&sid=31525d87af52ba5132639a11c8a5b0f3).

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

pastacolsugo, 2023-03-14:
I'm also getting `command not found` when invoking any command.

OS
Mac OS 12.6.2

Operating System version
Mac OS 12.6.2 (21G320)

Visual Studio Code version
1.76.1

noisysoil, 2023-03-14:
> I'm also getting `command not found` when invoking any command.
> 
> OS Mac OS 12.6.2
> 
> Operating System version Mac OS 12.6.2 (21G320)
> 
> Visual Studio Code version 1.76.1
> 
> EDIT: solved by downgrading to `v1.5.1`

Thanks for the report - it seems this bug is universal: [Similar issue on MacOS]( https://github.com/espressif/vscode-esp-idf-extension/issues/905)

yurriiy, 2023-03-15:
**All commands fails with message 'Command not found'**
ESP-IDF version 5.0.1
Windows 11 Pro 22000.1696
Visual Studio Code version 1.76.1

brianignacio5, 2023-03-17:
Could you try this [vsix installer](https://nightly.link/espressif/vscode-esp-idf-extension/actions/artifacts/601852097.zip) to see if the issue is fixed ?

parksj10, 2023-03-17:
@brianignacio5, tried your vsix, still getting the following errors 

configure ESP-IDF extension
```
command 'espIdf.setup.start' not found
```

and

when trying to create project from extension template:

```
command 'espIdf.createFiles' not found
```

brianignacio5, 2023-03-17:
Can you share the extension log or other related troubleshoot information?

Windows: %USERPROFILE%\.vscode\extensions\espressif.esp-idf-extension-VERSION\esp_idf_vsc_ext.log
Linux & MacOSX: $HOME/.vscode/extensions/espressif.esp-idf-extension-VERSION/esp_idf_vsc_ext.log

Same error as before or different one ?

windorey, 2023-03-17:
Also issue for me.

Clicking on anything from the ESP-IDF plug-in or running commands result in `command 'espIdf.x' not found`

ESP-IDF version 5.0.1
Ubuntu 20.04
VSCode version 1.76

brianignacio5, 2023-03-17:
Have you tried the new vsix installer @windorey @parksj10 ?

Could you try this [vsix installer](https://nightly.link/espressif/vscode-esp-idf-extension/actions/artifacts/601852097.zip) to see if the issue is fixed ?

Can you share the extension log or other related troubleshoot information?

Windows: %USERPROFILE%\.vscode\extensions\espressif.esp-idf-extension-VERSION\esp_idf_vsc_ext.log
Linux & MacOSX: $HOME/.vscode/extensions/espressif.esp-idf-extension-VERSION/esp_idf_vsc_ext.log

Same error as before or different one ?

parksj10, 2023-03-19:
@brianignacio5 the latest `VSIX` seems to work(ish). It was able to create a template project, but froze on `Creating ESP-IDF Project...` dialog until I clicked `Cancel` after which it opened up and seems to be init'd. Here's the log

[esp_idf_vsc_ext.log.zip](https://github.com/espressif/vscode-esp-idf-extension/files/11009977/esp_idf_vsc_ext.log.zip)

bartoszwalicki, 2023-03-19:
Hi. I have the same problem on MacOS and Linux. What I figured out is that you can't run any command when you are on any other project than esp-idf based. When you open any esp-idf project, thank you can run any command without any problem.

brianignacio5, 2023-03-20:
Did you try the latest vsix installer I share before @bartoszwalicki ?

I'll take a look @parksj10

bartoszwalicki, 2023-03-24:
@brianignacio5 tried vsix installer that you shared. Everything working.

a2800276, 2023-03-28:
It seems that extension is looking for CMakeList.txt for some reason:
> 
> `{"message":"Cannot access filePath: /home/ssc/code/some-non-esp-idf-app/CMakeLists.txt","stack":"Error: ENOENT: no such file or directory`

I was trying to play with the VSCode extension yesterday and was banging my head against the wall.

I think the above error is central: starting vscode with the IDF extension  (1.6.0) installed needs to be started in an IDF project directory initially. This is annoying, e.g. if you want to run "Create Project" or "Create Component"

It seems to me, that a workaround is `touch CMakeLists.txt`. If a file with that name is available in the current directory, the IDF extension (1.6.0) seems to load normally.

brianignacio5, 2023-03-28:
Have you tried the Installer above to see if the issue is fixed ?

a2800276, 2023-03-28:
It seems to have fixed the problem. I don't really use vs code though, so I have no idea what I'm doing. Thanks! Details:

- uninstalled existing plugin
- removed `~/.vscode/extensions/espressif.esp-idf-extension-1.6.0`
- moved the downloaded zip to an empty dir, unzipped it
- started `code` in that directory
- installed the unzipped vsix package using "Extensions -> ... -> Install from vsix"
- F1->configure : Works!

repeated the above with the 1.6.0 network install, last step fails with "Command not found". Uninstalling the vsix only via the Extension manager does repoduce the original problem, presumably because the orphaned `~/.vscode/extensions/espressif.esp-idf-extension-1.6.0` still has a valid setup.

closed 2023-04-04 by brianignacio5

embedded-solutions-art, 2024-03-26:
> It seems to have fixed the problem. I don't really use vs code though, so I have no idea what I'm doing. Thanks! Details:
>
> * uninstalled existing plugin
> * removed [...]

That is the only correct guide that works for me. Thanks a lot!

PR #909 (2023-04-04): catch cmakelists txt error sdkconfig file access
## Description

Catch CMakeLists.txt file access error reading IDF Target on extension activation.

Fixes #897 
Fix #905 
Fix #907
Fix #911 

## Type of change

Please delete options that are not relevant.
- Bug fix (non-breaking change which fixes an issue)

## How has this been tested?

Manual testing

**Test Configuration**:
* ESP-IDF Version: 5.0
* OS (Windows,Linux and macOS): macOS Ventura

## Checklist
- [x] PR Self Reviewed
- [x] Applied Code formatting
- [ ] Added Documentation
- [ ] Added Unit Test
- [ ] Verified on all platforms - Win [... cut]