espressif/vscode-esp-idf-extension #911: ESP-IDF: commands fail in Visual Studio Code

shan-mcarthur, 2023-03-19:
### OS

Windows

### Operating System version

Windows 10 v 22H2

### Visual Studio Code version

1.76.2

### ESP-IDF version

5.0

### Python version

3.8.7

### Doctor command output

command 'espldf.doctorCommand' not found
[image]

### Extension

{"message":"Cannot access filePath: c:\\iot-dev\\iot-dev\\ESP32-OLED-Arduino\\CMakeLists.txt","stack":"Error: ENOENT: no such file or directory, access 'c:\\iot-dev\\iot-dev\\ESP32-OLED-Arduino\\CMakeLists.txt'\n\tat Object.accessSync (node:fs:250:3)\n\tat e.accessSync (node:electron/js2c/asar_bundle:5:8277)\n\tat k (\\dist\\extension.js:1:6328)\n\tat \\dist\\extension.js:1:8128\n\tat B (\\dist\\extension.js:1:8307)\n\tat \\dist\\extension.js:1:37265\n\tat Generator.next (<anonymous>)\n\tat \\dist\\extension.js:1:36354\n\tat new Promise (<anonymous>)\n\tat d (\\dist\\extension.js:1:36099)\n\tat g (\\dist\\extension.js:1:37214)\n\tat \\dist\\extension.js:1:1222971\n\tat Generator.next (<anonymous>)\n\tat \\dist\\extension.js:1:1219342\n\tat new Promise (<anonymous>)\n\tat Ne (\\dist\\extension.js:1:1219087)\n\tat ct (\\dist\\extension.js:1:1219976)\n\tat Ui.db (\\out\\vs\\workbench\\api\\node\\extensionHostProcess.js:108:14833)\n\tat Ui.cb (\\out\\vs\\workbench\\api\\node\\extensionHostProcess.js:108:14532)\n\tat \\out\\vs\\workbench\\api\\node\\extensionHostProcess.js:108:12550\n\tat async f.n (\\out\\vs\\workbench\\api\\node\\extensionHostProcess.js:94:6818)\n\tat async f.m (\\out\\vs\\workbench\\api\\node\\extensionHostProcess.js:94:6781)\n\tat async f.l (\\out\\vs\\workbench\\api\\node\\extensionHostProcess.js:94:6238)","level":"error","timestamp":"2023-03-19T23:41:23.355Z"}
{"message":"Cannot access filePath: c:\\iot-dev\\iot-dev\\ESP32-OLED-Arduino\\CMakeLists.txt","stack":"Error: ENOENT: no such file or directory, access 'c:\\iot-dev\\iot-dev\\ESP32-OLED-Arduino\\CMakeLists.txt'\n\tat Object.accessSync (node:fs:250:3)\n\tat e.accessSync (node:electron/js2c/asar_bundle:5:8277)\n\tat k (\\dist\\extension.js:1:6328)\n\tat \\dist\\extension.js:1:8128\n\tat B (\\dist\\extension.js:1:8307)\n\tat \\dist\\extension.js:1:37265\n\tat Generator.next (<anonymous>)\n\tat \\dist\\extension.js:1:36354\n\tat new Promise (<anonymous>)\n\tat d (\\dist\\extension.js:1:36099)\n\tat g (\\dist\\extension.js:1:37214)\n\tat \\dist\\extension.js:1:1222971\n\tat Generator.next (<anonymous>)\n\tat \\dist\\extension.js:1:1219342\n\tat new Promise (<anonymous>)\n\tat Ne (\\dist\\extension.js:1:1219087)\n\tat ct (\\dist\\extension.js:1:1219976)\n\tat Ui.db (\\out\\vs\\workbench\\api\\node\\extensionHostProcess.js:108:14833)\n\tat Ui.cb (\\out\\vs\\workbench\\api\\node\\extensionHostProcess.js:108:14532)\n\tat \\out\\vs\\workbench\\api\\node\\extensionHostProcess.js:108:12550\n\tat async f.n (\\out\\vs\\workbench\\api\\node\\extensionHostProcess.js:94:6818)\n\tat async f.m (\\out\\vs\\workbench\\api\\node\\extensionHostProcess.js:94:6781)\n\tat async f.l (\\out\\vs\\workbench\\api\\node\\extensionHostProcess.js:94:6238)","level":"error","timestamp":"2023-03-19T23:42:43.422Z"}

### Description

I have followed all of the setup instructions, including carefully reviewing these instructions:  https://docs.espressif.com/projects/esp-idf/en/latest/esp32/get-started/windows-setup.html

I can get the hello world example to compile from the powershell command line.  I just can't get the visual studio extension working properly.

### Debug Message

```plain
n/a
```

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

brianignacio5, 2023-03-20:
Could you try this [vsix installer](https://nightly.link/espressif/vscode-esp-idf-extension/actions/artifacts/601852097.zip) to see if the issue is fixed ?

shan-mcarthur, 2023-03-21:
Thank you Brian, The extension works beautifully now!

PR #909 (2023-04-04): catch cmakelists txt error sdkconfig file access
## Description

Catch CMakeLists.txt file access error reading IDF Target on extension activation.

Fixes #897 
Fix #905 
Fix #907
Fix #911 

## Type of change

Please delete options that are not relevant.
- Bug fix (non-breaking change which fixes an issue)

## How has this been tested?

Manual testing

**Test Configuration**:
* ESP-IDF Version: 5.0
* OS (Windows,Linux and macOS): macOS Ventura

## Checklist
- [x] PR Self Reviewed
- [x] Applied Code formatting
- [ ] Added Documentation
- [ ] Added Unit Test
- [ ] Verified on all platforms - Win [... cut]
closed 2023-03-21 by shan-mcarthur