espressif/vscode-esp-idf-extension #919: vscode esp idf extension fails on Debian 12 (Bookworm) due to Python being externally managed

s-marios, 2023-03-30:
### OS

Linux

### Operating System version

Debian Bookworm

### Visual Studio Code version

1.76.2

### ESP-IDF version

5.0.1

### Python version

3.11.2

### Doctor command output

[no output produced - the following is from the extension log file]
{"tag":"DOCTOR COMMAND","user":true,"level":"warn","message":"{\"configurationSettings\":{\"espAdfPath\":\"${env:ADF_PATH}\",\"espIdfPath\":\"<HOMEPATH>/esp/esp-idf\",\"espMdfPath\":\"${env:MDF_PATH}\",\"espMatterPath\":\"${env:ESP_MATTER_PATH}\",\"customExtraPaths\":\"<HOMEPATH>/.espressif/tools/xtensa-esp-elf-gdb/11.2_20220823/xtensa-esp-elf-gdb/bin:<HOMEPATH>/.espressif/tools/riscv32-esp-elf-gdb/11.2_20220823/riscv32-esp-elf-gdb/bin:<HOMEPATH>/.espressif/tools/xtensa-esp32-elf/esp-2022r1-11.2.0/xtensa-esp32-elf/bin:<HOMEPATH>/.espressif/tools/xtensa-esp32s2-elf/esp-2022r1-11.2.0/xtensa-esp32s2-elf/bin:<HOMEPATH>/.espressif/tools/xtensa-esp32s3-elf/esp-2022r1-11.2.0/xtensa-esp32s3-elf/bin:<HOMEPATH>/.espressif/tools/riscv32-esp-elf/esp-2022r1-11.2.0/riscv32-esp-elf/bin:<HOMEPATH>/.espressif/tools/esp32ulp-elf/2.35_20220830/esp32ulp-elf/bin:<HOMEPATH>/.espressif/tools/openocd-esp32/v0.11.0-esp32-20221026/openocd-esp32/bin\",\"customExtraVars\":{\"OPENOCD_SCRIPTS\":\"<HOMEPATH>/.espressif/tools/openocd-esp32/v0.11.0-esp32-20221026/openocd-esp32/share/openocd/scripts\"},\"pythonBinPath\":\"<HOMEPATH>/.espressif/python_env/idf5.0_py3.10_env/bin/python\",\"pythonPackages\":[],\"serialPort\":\"/dev/ttyACM0\",\"openOcdConfigs\":[\"interface/ftdi/esp32_devkitj_v1.cfg\",\"board/esp32-wrover.cfg\"],\"toolsPath\":\"<HOMEPATH>/.espressif\",\"systemEnvPath\":\"<HOMEPATH>/.cargo/bin:/usr/local/bin:/usr/bin:/bin:/usr/local/games:/usr/games:/snap/bin\",\"gitPath\":\"git\"},\"configurationAccess\":{\"espIdfPath\":true,\"espAdfPath\":false,\"espMdfPath\":false,\"espMatterPath\":false,\"espIdfToolsPaths\":{\"<HOMEPATH>/.espressif/tools/xtensa-esp-elf-gdb/11.2_20220823/xtensa-esp-elf-gdb/bin\":true,\"<HOMEPATH>/.espressif/tools/riscv32-esp-elf-gdb/11.2_20220823/riscv32-esp-elf-gdb/bin\":true,\"<HOMEPATH>/.espressif/tools/xtensa-esp32-elf/esp-2022r1-11.2.0/xtensa-esp32-elf/bin\":true,\"<HOMEPATH>/.espressif/tools/xtensa-esp32s2-elf/esp-2022r1-11.2.0/xtensa-esp32s2-elf/bin\":true,\"<HOMEPATH>/.espressif/tools/xtensa-esp32s3-elf/esp-2022r1-11.2.0/xtensa-esp32s3-elf/bin\":true,\"<HOMEPATH>/.espressif/tools/riscv32-esp-elf/esp-2022r1-11.2.0/riscv32-esp-elf/bin\":true,\"<HOMEPATH>/.espressif/tools/esp32ulp-elf/2.35_20220830/esp32ulp-elf/bin\":true,\"<HOMEPATH>/.espressif/tools/openocd-esp32/v0.11.0-esp32-20221026/openocd-esp32/bin\":true},\"pythonBinPath\":true,\"cmakeInEnv\":true,\"ninjaInEnv\":true,\"toolsPath\":true},\"configurationSpacesValidation\":{\"customExtraPaths\":{\"<HOMEPATH>/.espressif/tools/xtensa-esp-elf-gdb/11.2_20220823/xtensa-esp-elf-gdb/bin\":false,\"<HOMEPATH>/.espressif/tools/riscv32-esp-elf-gdb/11.2_20220823/riscv32-esp-elf-gdb/bin\":false,\"<HOMEPATH>/.espressif/tools/xtensa-esp32-elf/esp-2022r1-11.2.0/xtensa-esp32-elf/bin\":false,\"<HOMEPATH>/.espressif/tools/xtensa-esp32s2-elf/esp-2022r1-11.2.0/xtensa-esp32s2-elf/bin\":false,\"<HOMEPATH>/.espressif/tools/xtensa-esp32s3-elf/esp-2022r1-11.2.0/xtensa-esp32s3-elf/bin\":false,\"<HOMEPATH>/.espressif/tools/riscv32-esp-elf/esp-2022r1-11.2.0/riscv32-esp-elf/bin\":false,\"<HOMEPATH>/.espressif/tools/esp32ulp-elf/2.35_20220830/esp32ulp-elf/bin\":false,\"<HOMEPATH>/.espressif/tools/openocd-esp32/v0.11.0-esp32-20221026/openocd-esp32/bin\":false},\"espAdfPath\":false,\"espIdfPath\":false,\"espMatterPath\":false,\"espMdfPath\":false,\"gitPath\":false,\"pythonBinPath\":false,\"toolsPath\":false,\"systemEnvPath\":false},\"debugAdapterRequirements\":{},\"espIdfVersion\":{\"output\":\"v5.0.1\\n\",\"result\":\"5.0.1\"},\"extensionRequirements\":{},\"gitVersion\":{\"output\":\"git version 2.39.2\\n\",\"result\":\"2.39.2\"},\"idfCheckRequirements\":{},\"latestError\":{\"killed\":false,\"code\":1,\"signal\":null,\"cmd\":\"<HOMEPATH>/.espressif/python_env/idf5.0_py3.10_env/bin/python -m pip list --format json\"},\"pipVersion\":{\"result\":\"Not found\"},\"pythonPackages\":{},\"pythonVersion\":{\"output\":\"Python 3.11.2\\n\",\"result\":\"3.11.2\"},\"systemInfo\":{\"architecture\":\"x64\",\"envPath\":\"<HOMEPATH>/.cargo/bin:/usr/local/bin:/usr/bin:/bin:/usr/local/games:/usr/games:/snap/bin\",\"extensionVersion\":\"1.6.0\",\"language\":\"en\",\"shell\":\"/usr/bin/bash\",\"platform\":\"linux\",\"systemName\":\"6.1.0-5-amd64\",\"vscodeVersion\":\"1.76.2\"},\"workspaceFolder\":\"<HOMEPATH>/playground/gpio\",\"projectConfigurations\":{}}","timestamp":"2023-03-30T01:37:42.305Z"}

### Extension

{"message":"Open On-Chip Debugger v0.11.0-esp32-20221026 (2022-10-26-14:47)\nLicensed under GNU GPL v2\nFor bug reports, read\n\thttp://openocd.org/doc/doxygen/bugs.html\n","stack":"Error: Open On-Chip Debugger v0.11.0-esp32-20221026 (2022-10-26-14:47)\nLicensed under GNU GPL v2\nFor bug reports, read\n\thttp://openocd.org/doc/doxygen/bugs.html\n\n\tat /dist/extension.js:1:9835\n\tat ChildProcess.exithandler (node:child_process:399:7)\n\tat ChildProcess.emit (node:events:526:28)\n\tat maybeClose (node:internal/child_process:1092:16)\n\tat Socket.<anonymous> (node:internal/child_process:451:11)\n\tat Socket.emit (node:events:526:28)\n\tat Pipe.<anonymous> (node:net:687:12)","level":"error","timestamp":"2023-03-30T01:32:34.223Z"}
{"message":"Command failed: \"/usr/bin/python\" -m pip install --user virtualenv\nerror: externally-managed-environment\n\n× This environment is externally managed\n╰─> To install Python packages system-wide, try apt install\n    python3-xyz, where xyz is the package you are trying to\n    install.\n    \n    If you wish to install a non-Debian-packaged Python package,\n    create a virtual environment using python3 -m venv path/to/venv.\n    Then use path/to/venv/bin/python and path/to/venv/bin/pip. Make\n    sure you have python3-full installed.\n    \n    If you wish to install a non-Debian packaged Python application,\n    it may be easiest to use pipx install xyz, which will manage a\n    virtual environment for you. Make sure you have pipx installed.\n    \n    See /usr/share/doc/python3.11/README.venv for more information.\n\nnote: If you believe this is a mistake, please contact your Python installation or OS distribution provider. You can override this, at the risk of breaking your Python installation or OS, by passing --break-system-packages.\nhint: See PEP 668 for the detailed specification.\n","stack":"Error: Command failed: \"/usr/bin/python\" -m pip install --user virtualenv\nerror: externally-managed-environment\n\n× This environment is externally managed\n╰─> To install Python packages system-wide, try apt install\n    python3-xyz, where xyz is the package you are trying to\n    install.\n    \n    If you wish to install a non-Debian-packaged Python package,\n    create a virtual environment using python3 -m venv path/to/venv.\n    Then use path/to/venv/bin/python and path/to/venv/bin/pip. Make\n    sure you have python3-full installed.\n    \n    If you wish to install a non-Debian packaged Python application,\n    it may be easiest to use pipx install xyz, which will manage a\n    virtual environment for you. Make sure you have pipx installed.\n    \n    See /usr/share/doc/python3.11/README.venv for more information.\n\nnote: If you believe this is a mistake, please contact your Python installation or OS distribution provider. You can override this, at the risk of breaking your Python installation or OS, by passing --break-system-packages.\nhint: See PEP 668 for the detailed specification.\n\n\tat ChildProcess.exithandler (node:child_process:408:12)\n\tat ChildProcess.emit (node:events:526:28)\n\tat maybeClose (node:internal/child_process:1092:16)\n\tat Socket.<anonymous> (node:internal/child_process:451:11)\n\tat Socket.emit (node:events:526:28)\n\tat Pipe.<anonymous> (node:net:687:12)","level":"error","timestamp":"2023-03-30T01:32:48.901Z"}
{"user":true,"message":"Command failed: \"/usr/bin/python\" -m pip install --user virtualenv\nerror: externally-managed-environment\n\n× This environment is externally managed\n╰─> To install Python packages system-wide, try apt install\n    python3-xyz, where xyz is the package you are trying to\n    install.\n    \n    If you wish to install a non-Debian-packaged Python package,\n    create a virtual environment using python3 -m venv path/to/venv.\n    Then use path/to/venv/bin/python and path/to [... cut ...]
note: If you believe this is a mistake, please contact your Python installation or OS distribution provider. You can override this, at the risk of breaking your Python installation or OS, by passing --break-system-packages.
hint: See PEP 668 for the detailed specification.
```

### Other Steps to Reproduce

1. Install Debian Bookworm
2. Install vscode
3. (Attempt to) install the vscode esp idf extension

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

Spacefish, 2023-03-31:
I had the same issue, it tries to install venv via the system pip.. But afterwads sets up a venv to install all the remaining packages required..

The first command fails with the error which aborts the whole process.

The idea in newer python version is to install pip/pipx via the system package manager of the system typically. And use an venv for all pip operations.

Suggested fix:
If Python Version >= 3.11:
check whether venv is already installed if yes, just skip the outside venv pip install and directly go to seting up the venv. Otherwise show an error that the user should install venv / all other globally required python packages via their systems package manager.
I think i will do a PR for such a behaviour..

To get it working on my PC i just added the flags to force install via pip ( --break-system-packages ) to the VS Extensions code b.t.w.. Not the best solution.

B.t.w. the whole thing was introduced by this: https://packaging.python.org/en/latest/specifications/externally-managed-environments/

As a temporary workarround, you might also want to just rename the `EXTERNALLY-MANAGED` marker File on your PC.

s-marios, 2023-04-18:
So now I have to "bump" this issue to avoid having the bot closing this? So be it.

"Bump".

Elijahg, 2023-04-21:
I am also getting the same issue.

brianignacio5, 2023-04-28:
Added the fix in this [vsix installer](https://nightly.link/espressif/vscode-esp-idf-extension/actions/artifacts/670362036.zip)

PR #956 (2023-05-04): remove virtualenv install from extension
## Description

Remove virtualenv install from extension setup process.

Fixes #919 
Fix #949

## Type of change

- Bug fix (non-breaking change which fixes an issue)

## Steps to test this pull request

1. Click on "Configure ESP-IDF extension" command.
2. Execute setup process using Python 3.11.
3. Observe results. Install setup should work.

## How has this been tested?

Manual test of setup install

**Test Configuration**:
* ESP-IDF Version: 5.0
* OS (Windows,Linux and macOS): macOS

## Checklist
- [ ] PR Self Reviewed
- [ ] Applied Code formatting
 [... cut]
closed 2023-05-04 by brianignacio5