espressif/vscode-esp-idf-extension #926: Console and monitor baud not showing in menuconfig UI

felixcollins, 2023-04-04:
### OS

Windows

### Operating System version

Win 10

### Visual Studio Code version

1.77.0

### ESP-IDF version

5.0

### Python version

3.8.7

### Doctor command output

```
---------------------------------------------- ESP-IDF Extension for Visual Studio Code report ---------------------------------------------
OS win32 x64 10.0.19044 
System environment variable IDF_PYTHON_ENV_PATH 
 undefined 
System environment variable PATH 
 C:\Python310\Scripts\;C:\Python310\;C:\Program Files\Eclipse Foundation\jdk-8.0.302.8-hotspot\bin;C:\Program Files\NVIDIA GPU Computing Toolkit\CUDA\v10.0\include;C:\Program Files\NVIDIA GPU Computing Toolkit\CUDA\v10.0\extras\CUPTI\libx64;C:\Program Files\NVIDIA GPU Computing Toolkit\CUDA\v10.0\bin;C:\Program Files\NVIDIA GPU Computing Toolkit\CUDA\v10.0\libnvvp;C:\Program Files\NVIDIA GPU Computing Toolkit\CUDA\v10.1\include;C:\Program Files\NVIDIA GPU Computing Toolkit\CUDA\v10.1\extras\CUPTI\libx64;C:\Program Files\NVIDIA GPU Computing Toolkit\CUDA\v10.1\bin;C:\Program Files\NVIDIA GPU Computing Toolkit\CUDA\v10.1\libnvvp;C:\Windows\system32;C:\Windows;C:\Windows\System32\Wbem;C:\Windows\System32\WindowsPowerShell\v1.0\;C:\Windows\System32\OpenSSH\;C:\Program Files\Microsoft SQL Server\130\Tools\Binn\;C:\Program Files\Microsoft SQL Server\Client SDK\ODBC\170\Tools\Binn\;C:\Program Files\Microsoft SQL Server\Client SDK\ODBC\130\Tools\Binn\;C:\Program Files (x86)\Microsoft SQL Server\140\Tools\Binn\;C:\Program Files\Microsoft SQL Server\140\Tools\Binn\;C:\Program Files\Microsoft SQL Server\140\DTS\Binn\;C:\Program Files (x86)\Microsoft SQL Server\150\DTS\Binn\;C:\Program Files\Intel\WiFi\bin\;C:\Program Files\Common Files\Intel\WirelessCommon\;C:\Program Files\PuTTY\;C:\Program Files (x86)\NVIDIA Corporation\PhysX\Common;C:\Program Files\NVIDIA Corporation\Nsight Compute 2019.4.0\;C:\flutter\bin;C:\Program Files\Android\Android Studio\jre\bin;C:<HOMEPATH>\AppData\Local\Android\Sdk\platform-tools;C:\WINDOWS\system32;C:\WINDOWS;C:\WINDOWS\System32\Wbem;C:\WINDOWS\System32\WindowsPowerShell\v1.0\;C:\WINDOWS\System32\OpenSSH\;C:\Program Files\Git\cmd;C:\ProgramData\chocolatey\bin;C:\MinGW\bin;C:\Program Files\nodejs\;C:\Program Files\dotnet\;C:\Program Files\TortoiseHg\;C:<HOMEPATH>\AppData\Local\Microsoft\WindowsApps;C:<HOMEPATH>\AppData\Local\Programs\Microsoft VS Code\bin;C:<HOMEPATH>\.dotnet\tools;C:<HOMEPATH>\AppData\Local\Microsoft\WindowsApps;C:<HOMEPATH>\AppData\Roaming\Pub\Cache\bin;C:<HOMEPATH>\AppData\Local\GitHubDesktop\bin;C:<HOMEPATH>\AppData\Roaming\npm;C:<HOMEPATH>\.dotnet\tools 
System environment variable PYTHON 
 undefined 
Visual Studio Code version 1.77.0 
Visual Studio Code language en 
Visual Studio Code shell C:\WINDOWS\System32\cmd.exe 
ESP-IDF Extension version 1.6.0 
Workspace folder c:<HOMEPATH>\source\Thor\fw-sensor-hub 
---------------------------------------------------- Extension configuration settings ------------------------------------------------------
ESP-ADF Path (idf.espAdfPath) C:<HOMEPATH>\.espressif\esp-adf
ESP-IDF Path (idf.espIdfPath) C:<HOMEPATH>\esp5\esp-idf
ESP-MDF Path (idf.espMdfPath) ${env:MDF_PATH}
ESP-Matter Path (idf.espMatterPath) ${env:ESP_MATTER_PATH}
Custom extra paths (idf.customExtraPaths) c:<HOMEPATH>\esp5\esp-tools\tools\xtensa-esp-elf-gdb\11.2_20220823\xtensa-esp-elf-gdb\bin;c:<HOMEPATH>\esp5\esp-tools\tools\riscv32-esp-elf-gdb\11.2_20220823\riscv32-esp-elf-gdb\bin;c:<HOMEPATH>\esp5\esp-tools\tools\xtensa-esp32-elf\esp-2022r1-11.2.0\xtensa-esp32-elf\bin;c:<HOMEPATH>\esp5\esp-tools\tools\xtensa-esp32s2-elf\esp-2022r1-11.2.0\xtensa-esp32s2-elf\bin;c:<HOMEPATH>\esp5\esp-tools\tools\xtensa-esp32s3-elf\esp-2022r1-11.2.0\xtensa-esp32s3-elf\bin;c:<HOMEPATH>\esp5\esp-tools\tools\riscv32-esp-elf\esp-2022r1-11.2.0\riscv32-esp-elf\bin;c:<HOMEPATH>\esp5\esp-tools\tools\esp32ulp-elf\2.35_20220830\esp32ulp-elf\bin;c:<HOMEPATH>\esp5\esp-tools\tools\cmake\3.24.0\bin;c:<HOMEPATH>\esp5\esp-tools\tools\openocd-esp32\v0.11.0-esp32-20221026\openocd-esp32\bin;c:<HOMEPATH>\esp5\esp-tools\tools\ninja\1.10.2;c:<HOMEPATH>\esp5\esp-tools\tools\idf-exe\1.0.3;c:<HOMEPATH>\esp5\esp-tools\tools\ccache\4.6.2\ccache-4.6.2-windows-x86_64;c:<HOMEPATH>\esp5\esp-tools\tools\dfu-util\0.9\dfu-util-0.9-win64
Custom extra vars (idf.customExtraVars)
    OPENOCD_SCRIPTS: c:<HOMEPATH>\esp5\esp-tools\tools\openocd-esp32\v0.11.0-esp32-20221026/openocd-esp32/share/openocd/scripts
    IDF_CCACHE_ENABLE: 1
Virtual env Python Path (idf.pythonBinPath) c:<HOMEPATH>\esp5\esp-tools\python_env\idf5.0_py3.8_env\Scripts\python.exe
Serial port (idf.port) COM7
OpenOCD Configs (idf.openOcdConfigs) interface/ftdi/tumpa-lite.cfg,target/esp32.cfg
ESP-IDF Tools Path (idf.toolsPath) c:<HOMEPATH>\esp5\esp-tools
Git Path (idf.gitPath) C:<HOMEPATH>\.espressif\tools\idf-git\2.30.1\cmd\git.exe
-------------------------------------------------------- Configurations access -------------------------------------------------------------
Access to ESP-ADF Path (idf.espAdfPath) true
Access to ESP-IDF Path (idf.espIdfPath) true
Access to ESP-MDF Path (idf.espMdfPath) false
Access to ESP-Matter Path (idf.espMatterPath) false
Access to ESP-IDF Custom extra paths
Access to c:<HOMEPATH>\esp5\esp-tools\tools\xtensa-esp-elf-gdb\11.2_20220823\xtensa-esp-elf-gdb\bin: true
Access to c:<HOMEPATH>\esp5\esp-tools\tools\riscv32-esp-elf-gdb\11.2_20220823\riscv32-esp-elf-gdb\bin: true
Access to c:<HOMEPATH>\esp5\esp-tools\tools\xtensa-esp32-elf\esp-2022r1-11.2.0\xtensa-esp32-elf\bin: true
Access to c:<HOMEPATH>\esp5\esp-tools\tools\xtensa-esp32s2-elf\esp-2022r1-11.2.0\xtensa-esp32s2-elf\bin: true
Access to c:<HOMEPATH>\esp5\esp-tools\tools\xtensa-esp32s3-elf\esp-2022r1-11.2.0\xtensa-esp32s3-elf\bin: true
Access to c:<HOMEPATH>\esp5\esp-tools\tools\riscv32-esp-elf\esp-2022r1-11.2.0\riscv32-esp-elf\bin: true
Access to c:<HOMEPATH>\esp5\esp-tools\tools\esp32ulp-elf\2.35_20220830\esp32ulp-elf\bin: true
Access to c:<HOMEPATH>\esp5\esp-tools\tools\cmake\3.24.0\bin: true
Access to c:<HOMEPATH>\esp5\esp-tools\tools\openocd-esp32\v0.11.0-esp32-20221026\openocd-esp32\bin: true
Access to c:<HOMEPATH>\esp5\esp-tools\tools\ninja\1.10.2: true
Access to c:<HOMEPATH>\esp5\esp-tools\tools\idf-exe\1.0.3: true
Access to c:<HOMEPATH>\esp5\esp-tools\tools\ccache\4.6.2\ccache-4.6.2-windows-x86_64: true
Access to c:<HOMEPATH>\esp5\esp-tools\tools\dfu-util\0.9\dfu-util-0.9-win64: true
Access to Virtual env Python Path (idf.pythonBinPath) true
[... 141 lines cut ...]
### Extension

Nothing relevant

### Description

When I run the config gui I can not find the options to change log output baud and monitor baud. 

If I edit the sdkconfig file directly, my changes get overwritten the next time I build (it is set back to 115200).

I thought the sdkconfig was the source of truth for a project but it seems this is not true for the vscode extension. Where are these settings saved for a esp32-vscode project?

Why does the log output and monitor baud settings not show up in the menuconfig gui?

( Sorry to raise a bug, but I asked in the forum but no one replied. https://www.esp32.com/viewtopic.php?f=40&t=32808)

### Debug Message

```plain
Not relevevant
```

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

brianignacio5, 2023-04-06:
The thing is that `CONFIG_ESPTOOLPY_MONITOR_BAUD` depends on `CONFIG_APP_BUILD_TYPE_PURE_RAM_APP` so you need to set the `build type` to `Pure RAM App (Experimental)` but for some reason is not shown in the menuconfig still.

brianignacio5, 2023-04-18:
Please take a look at https://docs.espressif.com/projects/esp-idf/en/latest/esp32/migration-guides/release-5.x/5.0/tools.html#idf-monitor

felixcollins, 2023-04-18:
Thanks Brian. At least I know I'm not going crazy! It will be great to be able to run the console at a high baud rate to reduce the logging impact on the application.

brianignacio5, 2023-04-26:
@felixcollins  Could you try this [vsix installer](https://nightly.link/espressif/vscode-esp-idf-extension/actions/artifacts/666045721.zip) ?

felixcollins, 2023-04-26:
Thanks @brianignacio5 !  I tried it. The config gui still does not show the baud. See attachment. If I edit the sdkcondig directly it gets overwritten at the next build. Where does the build get the information to overwrite? Can I edit that instead?
[image]

felixcollins, 2023-04-26:
Also...
[image]

felixcollins, 2023-04-26:
If I set to Custom UART it shows up!  Is that the fix?

felixcollins, 2023-04-26:
I set the console baud to 921600 and rebuilt my project. The monitor baud now starts as 460800. Which obviously does not work. Should I raise another issue as this one seems to be fixed?

brianignacio5, 2023-04-27:
Hi @felixcollins As mentioned in the 5.0 changes link I shared before, IDF Monitor now needs to specify the baud rate in the command instead of using GUI SDK Configuration Editor. In the vsix I shared from the PR, We added a new extension setting called `idf.monitorBaudRate` to launch IDF Monitor from extension command `Monitor your device` with the value in the setting. Could you try using this setting and see if it works ?

felixcollins, 2023-04-27:
Sorry. I see that 460800 is now the default. I added the setting and it now works at 921600 as I desired!

closed 2023-04-27 by felixcollins

AndriiFilippov, 2023-05-05:
Tested on:
OS - Windows 10
ESP-IDF: master
VSCode: 1.78.0

LGTM 👍

PR #951 (2023-04-28): add idf monitorBaudRate setting
## Description

Add `idf.monitorBaudRate` configuration setting to define IDF Monitor baud rate in the extension commands.

Delete existing IDF Monitor when re executing the IDF Monitor.

Fixes #926
Fix #903

## Type of change

- New feature (non-breaking change which adds functionality)

## Steps to test this pull request

Provide a list of steps to test changes in this PR and required output

1. Click on "settings.json" and update `idf.monitorBaudRate` to new value
2. Execute `ESP-IDF: Monitor your device` command to start IDF Monitor.
3. Observe results. Chec [... cut]