espressif/vscode-esp-idf-extension #927: [Feature Request]: Toggle Status Bar via settings.json

Delzye, 2023-04-05:
**Is your feature request related to a problem? Please describe.**
At our company, we use one VSCode Installation for multiple kinds of work, one of which is working with esp-idf.
The status bar tends to get pretty cluttered with icons, so we manually have to turn on/off the idf whenever we (don't) use it.
That is tedious and probably easily solvable.

**Describe the solution you'd like**
Add a setting in settings.json to turn off the status bar, simila to the one CMake-Tools or GitLens has.
This enables us to disable the status bar by default and enable it in our idf-project-template for that specific workspace.

**Describe alternatives you've considered**
Keeping it manual or using multiple VSCode Installations. Both is suboptimal, to put it nicely.

**Additional context**
CMake-Tools status bar setting:
[image]

Delzye, 2023-04-24:
I'm still waiting for a response from the developers.

brianignacio5, 2023-04-27:
Could you try this [vsix installer](https://nightly.link/espressif/vscode-esp-idf-extension/actions/artifacts/667838659.zip) ?

Delzye, 2023-04-27:
Hi Brian,

works like a charm, thank you very much!

closed 2023-04-27 by Delzye

AndriiFilippov, 2023-05-05:
Tested on: 
OS - Windows 10
ESP-IDF: master 
VSCode: 1.78.0

LGTM 👍

PR #953 (2023-04-28): add idf enableStatusBar setting to show status bar
## Description

Add  `idf.enableStatusBar`configuration setting to show or hide the extension status bar items. This settings can be apply as User, workspace or workspace folder setting.

Fixes #927

## Type of change

Please delete options that are not relevant.
- New feature (non-breaking change which adds functionality)

## Steps to test this pull request

Provide a list of steps to test changes in this PR and required output

1. Open `settings.json` and modify `idf.enableStatusBar` to `true` or `false`.
2. See status bar items appear and d [... cut]