espressif/vscode-esp-idf-extension #928: [Feature Request]: Option to prevent the extension from configuring on first opening of VSCode

Delzye, 2023-04-05:
**Is your feature request related to a problem? Please describe.**
We distribute one VSCode Installation among all developers, so that everyone works with the same configuration/version of everything.
When downloading the folder and starting VSCode for the first time, the extension tries to configure itself.
This is not necessary, because we already configured it correctly and leads to problems, because sometimes it misconfigures stuff.

To add context: we have mutliple IDF-Versions installed, because not every project is immediately updated (and some old ones are only maintained to keep functioning).

Once e.g. the extension configured itself for a mix of IDF 4.3 and 4.4 versions of the tools etc., even though the settings.json was correctly configured for IDF 5.0 (And the extension knew that, under "Use existing setup" it found IDF 5.0, but not 4.4 and 4.3, instead it just said IDF Version X.X).

This lead to building a project failing without any error message.

**Describe the solution you'd like**
An option in settings.json to prevent reconfiguring of the extension.

**Describe alternatives you've considered**
On any new setup, go to "Use existing setup" and hope the correct version of IDF shows up.

**Additional context**
-

brianignacio5, 2023-04-24:
Could you try this [vsix installer](https://nightly.link/espressif/vscode-esp-idf-extension/actions/artifacts/661867598.zip) ?

There is a `idf.showOnboardingOnInit` configuration setting to enable the self configuration on extension activation.

Delzye, 2023-04-24:
Hi Brian,

i just tried it in a VM and everything seems to work as expected.

Thanks for adding this!

closed 2023-04-24 by Delzye