espressif/vscode-esp-idf-extension #930: Express instal don't work

Andrysky, 2023-04-10:
### OS

Linux

### Operating System version

Linux raspberrypi 5.15.84-v8+ #1613 SMP PREEMPT Thu Jan 5 12:03:08 GMT 2023 aarch64 GNU/Linux

### Visual Studio Code version

1.77.1

### ESP-IDF version

release/v4.4

### Python version

Python 3.9.2

### Doctor command output

```
---------------------------------------------- ESP-IDF Extension for Visual Studio Code report ---------------------------------------------
OS linux arm64 5.15.84-v8+ 
System environment variable IDF_PYTHON_ENV_PATH 
 undefined 
System environment variable PATH 
 <HOMEPATH>/.vscode-server/bin/b7886d7461186a5eac768481578c1d7ca80e2d21/bin/remote-cli:/usr/local/sbin:/usr/local/bin:/usr/sbin:/usr/bin:/sbin:/bin:/usr/local/games:/usr/games 
System environment variable PYTHON 
 undefined 
Visual Studio Code version 1.77.1 
Visual Studio Code language en 
Visual Studio Code shell /bin/bash 
ESP-IDF Extension version 1.6.1 
Workspace folder No workspace folder is open 
---------------------------------------------------- Extension configuration settings ------------------------------------------------------
ESP-ADF Path (idf.espAdfPath) ${env:ADF_PATH}
ESP-IDF Path (idf.espIdfPath) ${env:IDF_PATH}
ESP-MDF Path (idf.espMdfPath) ${env:MDF_PATH}
ESP-Matter Path (idf.espMatterPath) ${env:ESP_MATTER_PATH}
Custom extra paths (idf.customExtraPaths) C:\Users\Andry\.espressif\tools\xtensa-esp-elf-gdb\11.2_20220823\xtensa-esp-elf-gdb\bin;C:\Users\Andry\.espressif\tools\riscv32-esp-elf-gdb\11.2_20220823\riscv32-esp-elf-gdb\bin;C:\Users\Andry\.espressif\tools\xtensa-esp32-elf\esp-2021r2-patch5-8.4.0\xtensa-esp32-elf\bin;C:\Users\Andry\.espressif\tools\xtensa-esp32s2-elf\esp-2021r2-patch5-8.4.0\xtensa-esp32s2-elf\bin;C:\Users\Andry\.espressif\tools\xtensa-esp32s3-elf\esp-2021r2-patch5-8.4.0\xtensa-esp32s3-elf\bin;C:\Users\Andry\.espressif\tools\riscv32-esp-elf\esp-2021r2-patch5-8.4.0\riscv32-esp-elf\bin;C:\Users\Andry\.espressif\tools\esp32ulp-elf\2.35_20220830\esp32ulp-elf\bin;C:\Users\Andry\.espressif\tools\cmake\3.23.1\bin;C:\Users\Andry\.espressif\tools\openocd-esp32\v0.11.0-esp32-20221026\openocd-esp32\bin;C:\Users\Andry\.espressif\tools\ninja\1.10.2;C:\Users\Andry\.espressif\tools\idf-exe\1.0.3;C:\Users\Andry\.espressif\tools\ccache\4.3\ccache-4.3-windows-64;C:\Users\Andry\.espressif\tools\dfu-util\0.9\dfu-util-0.9-win64
Custom extra vars (idf.customExtraVars)
    OPENOCD_SCRIPTS: C:\Users\Andry\.espressif\tools\openocd-esp32\v0.11.0-esp32-20221026/openocd-esp32/share/openocd/scripts
    IDF_CCACHE_ENABLE: 1
Virtual env Python Path (idf.pythonBinPath) python
Serial port (idf.port) /dev/ttyUSB1
OpenOCD Configs (idf.openOcdConfigs) interface/ftdi/esp32_devkitj_v1.cfg,board/esp32-wrover.cfg
ESP-IDF Tools Path (idf.toolsPath) ${env:IDF_TOOLS_PATH}
Git Path (idf.gitPath) /usr/bin/git
-------------------------------------------------------- Configurations access -------------------------------------------------------------
Access to ESP-ADF Path (idf.espAdfPath) false
Access to ESP-IDF Path (idf.espIdfPath) false
Access to ESP-MDF Path (idf.espMdfPath) false
Access to ESP-Matter Path (idf.espMatterPath) false
Access to ESP-IDF Custom extra paths
Access to C: false
Access to \Users\Andry\.espressif\tools\xtensa-esp-elf-gdb\11.2_20220823\xtensa-esp-elf-gdb\bin;C: false
Access to \Users\Andry\.espressif\tools\riscv32-esp-elf-gdb\11.2_20220823\riscv32-esp-elf-gdb\bin;C: false
Access to \Users\Andry\.espressif\tools\xtensa-esp32-elf\esp-2021r2-patch5-8.4.0\xtensa-esp32-elf\bin;C: false
Access to \Users\Andry\.espressif\tools\xtensa-esp32s2-elf\esp-2021r2-patch5-8.4.0\xtensa-esp32s2-elf\bin;C: false
Access to \Users\Andry\.espressif\tools\xtensa-esp32s3-elf\esp-2021r2-patch5-8.4.0\xtensa-esp32s3-elf\bin;C: false
Access to \Users\Andry\.espressif\tools\riscv32-esp-elf\esp-2021r2-patch5-8.4.0\riscv32-esp-elf\bin;C: false
Access to \Users\Andry\.espressif\tools\esp32ulp-elf\2.35_20220830\esp32ulp-elf\bin;C: false
Access to \Users\Andry\.espressif\tools\cmake\3.23.1\bin;C: false
Access to \Users\Andry\.espressif\tools\openocd-esp32\v0.11.0-esp32-20221026\openocd-esp32\bin;C: false
Access to \Users\Andry\.espressif\tools\ninja\1.10.2;C: false
Access to \Users\Andry\.espressif\tools\idf-exe\1.0.3;C: false
Access to \Users\Andry\.espressif\tools\ccache\4.3\ccache-4.3-windows-64;C: false
Access to \Users\Andry\.espressif\tools\dfu-util\0.9\dfu-util-0.9-win64: false
[... 58 lines cut ...]
install the vscode esp idf extension.
select EXPRESS install
[image]

set Select ESP-IDF version: release/v4.4
click install 
[image]

result:
```
idf_versions.txt has (1) KB
idf_versions.txt progress: 100.00% (0.10 / 0.10) KB
"/usr/bin/python -m pip" is not valid. (ERROR_INVALID_PIP)

```

### Debug Message

```plain
"/usr/bin/python -m pip" is not valid. (ERROR_INVALID_PIP)
```

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

jiafeng-1, 2023-04-10:
I also ran into this issue, restart the espressif IDF plugin, and then prompt to run the following code,
````sh
The following Python requirements are not satisfied:
pyparsing>=2.0.3,<2.4.0
To install the missing packages, please run "/home/jiafeng/esp/esp-idf/install.sh"
Diagnostic information:
    IDF_PYTHON_ENV_PATH: /home/jiafeng/esp/python_env/idf4.4_py3.6_env
    Python interpreter used: /home/jiafeng/esp/python_env/idf4.4_py3.6_env/bin/python
Command failed: "/home/jiafeng/esp/python_env/idf4.4_py3.6_env/bin/python" "/home/jiafeng/esp/esp-idf/tools/check_python_dependencies.py" -r "/home/jiafeng/esp/esp-idf/requirements.txt"
```

but,When I compile the project, I encounter the following error
```sh
ERROR: failed to parse /home/jiafeng/esp/esp-idf/components/esp_phy/linker.lf
Expected end of text, found 'if'  (at char 0), (line:1, col:1)
ninja: build stopped: subcommand failed.
```

Andrysky, 2023-04-10:
installing `sudo apt install python3-pip` seemed to help. but this info is not in https://github.com/espressif/vscode-esp-idf-extension/blob/master/docs/tutorial/install.md
and the plugin can set it itself once it sets up the environment., or at least say that you need to install pip

brianignacio5, 2023-04-11:
While in the [Readme's how to use](https://github.com/espressif/vscode-esp-idf-extension#how-to-use) we have a link to ESP-IDF system prerequisites it is true that python used to include pip and now it doesn't. 

Will add the remark in the docs.

closed 2023-05-03 by github-actions[bot]