espressif/vscode-esp-idf-extension #931: install enviroment with "vs code remote (ssh)" work wrong

Andrysky, 2023-04-10:
### OS

Windows

### Operating System version

Linux raspberrypi 5.15.84-v8+ #1613 SMP PREEMPT Thu Jan 5 12:03:08 GMT 2023 aarch64 GNU/Linux

### Visual Studio Code version

1.77.1

### ESP-IDF version

release/v4.4

### Python version

Python 3.9.2

### Doctor command output

```
---------------------------------------------- ESP-IDF Extension for Visual Studio Code report ---------------------------------------------
OS linux arm64 5.15.84-v8+ 
System environment variable IDF_PYTHON_ENV_PATH 
 undefined 
System environment variable PATH 
 <HOMEPATH>/.vscode-server/bin/b7886d7461186a5eac768481578c1d7ca80e2d21/bin/remote-cli:<HOMEPATH>/.local/bin:/usr/local/sbin:/usr/local/bin:/usr/sbin:/usr/bin:/sbin:/bin:/usr/local/games:/usr/games 
System environment variable PYTHON 
 undefined 
Visual Studio Code version 1.77.1 
Visual Studio Code language en 
Visual Studio Code shell /bin/bash 
ESP-IDF Extension version 1.6.1 
Workspace folder <HOMEPATH>/<my_proj>
---------------------------------------------------- Extension configuration settings ------------------------------------------------------
ESP-ADF Path (idf.espAdfPath) ${env:ADF_PATH}
ESP-IDF Path (idf.espIdfPath) <HOMEPATH>/esp/esp-idf
ESP-MDF Path (idf.espMdfPath) ${env:MDF_PATH}
ESP-Matter Path (idf.espMatterPath) ${env:ESP_MATTER_PATH}
Custom extra paths (idf.customExtraPaths) <HOMEPATH>/.espressif/tools/xtensa-esp-elf-gdb/11.2_20220823/xtensa-esp-elf-gdb/bin:<HOMEPATH>/.espressif/tools/riscv32-esp-elf-gdb/11.2_20220823/riscv32-esp-elf-gdb/bin:<HOMEPATH>/.espressif/tools/xtensa-esp32-elf/esp-2021r2-patch5-8.4.0/xtensa-esp32-elf/bin:<HOMEPATH>/.espressif/tools/xtensa-esp32s2-elf/esp-2021r2-patch5-8.4.0/xtensa-esp32s2-elf/bin:<HOMEPATH>/.espressif/tools/xtensa-esp32s3-elf/esp-2021r2-patch5-8.4.0/xtensa-esp32s3-elf/bin:<HOMEPATH>/.espressif/tools/riscv32-esp-elf/esp-2021r2-patch5-8.4.0/riscv32-esp-elf/bin:<HOMEPATH>/.espressif/tools/esp32ulp-elf/2.35_20220830/esp32ulp-elf/bin:<HOMEPATH>/.espressif/tools/cmake/3.23.1/bin:<HOMEPATH>/.espressif/tools/openocd-esp32/v0.11.0-esp32-20221026/openocd-esp32/bin:<HOMEPATH>/.espressif/tools/ninja/1.10.2
Custom extra vars (idf.customExtraVars)
    0: {
    1: "
    2: I
    3: D
    4: F
    5: _
    6: C
    7: C
    8: A
    9: C
    10: H
    11: E
    12: _
    13: E
    14: N
    15: A
    16: B
    17: L
    18: E
    19: "
    20: :
    21: "
    22: 1
    23: "
    24: }
Virtual env Python Path (idf.pythonBinPath) <HOMEPATH>/.espressif/python_env/idf4.4_py3.9_env/bin/python
Serial port (idf.port) /dev/ttyUSB0
[... 170 lines cut ...]
problem 1
extension when installing write variables to C:/Users/Andry/AppData/Roaming/Code/User/settings.json
not in /home/andry/.vscode-server/data/Machine/settings.json
because of which
a) my local environment is corrupted
b) if I connect to a remote machine from another computer, the environment will not be configured

problem 2
on a remote host arm, and the extension sets x86-64 during installation
because of this, when trying to assemble, we get
/bin/bash: line 1: /home/andry/.espressif/tools/ninja/1.10.2/ninja: cannot execute binary file: Exec format error

confirmation:
$ file /home/andry/.espressif/tools/ninja/1.10.2/ninja
/home/andry/.espressif/tools/ninja/1.10.2/ninja: ELF 64-bit LSB executable, x86-64, version 1 (SYSV), dynamically linked, interpreter /lib64/ld-linux-x86-64. so.2, for GNU/Linux 2.6.32, BuildID[sha1]=3c1def4659e39fa9a2870be5e78feea4a96e8bc3, stripped

### Debug Message

```plain
/bin/bash: line 1: /home/andry/.espressif/tools/ninja/1.10.2/ninja: cannot execute binary file: Exec format error
```

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

brianignacio5, 2023-04-11:
Maybe try to save the extension settings in the current workspace folder by selecting Workspace Folder option in the Setup Wizard ?

Also, we don't have ARM releases for Linux in 4.4.2 in the ESP-IDF for Ninja. Maybe try to install `ninja-build` system wide and update ninja build path in `idf.customExtraPaths` or add it in your system PATH and remove the x86_64 from `idf.customExtraPaths`

Andrysky, 2023-04-11:
I would like to be able to have the settings saved for the system (remote) and not for the voxspace.

vs code does not provide information that this is a remote connection? / does not provide the ability to save on a remote host and not on a local one?

yes, I installed `ninja-build` with my hands and it worked.

brianignacio5, 2023-04-18:
Closing this issue since the user has found the fix. Comment here if need further assistance

closed 2023-04-18 by brianignacio5