espressif/vscode-esp-idf-extension #934: [Feature Request]: Getting Offset for Flash

AliMoal, 2023-04-11:
I'm working on a project with OTA structure. My project has 2 OTA slots, I use first one as 3rd stage bootloader and my main program is on second one. In VS Code Extension, flashing device uses the first offset of app in partitions.csv. If I want to flash my program on second slot, I should flash that manually with Fhash binary to partition OR run flashing by CLI.
Is there any way for choosing offset for my main program regardless of partition.csv?
I mean configuring the command which will be run when I click on Flash device icon. Generally, it is good for users to select offsets for each .bin files by their owns.
Thanks.

AliMoal, 2023-04-12:
That's good but for first time I cannot flash bootloader.bin and partition_table.bin, because bootloader and partition table options are not in partition table. how can I flash them manually?

brianignacio5, 2023-04-12:
Oh I see. but why not ? You can specify a custom partition table using csv and the partitions will be flashed according to the `${project_dir}/build/flasher_args.json` offset for each partition. As long as the partitions doesn't overlap which each other the offset should be correct and the boot loader and partition table should be flashed to device which you can use later to flash a binary to OTA partition.

What is the content of `${project_dir}/build/flasher_args.json` ?

AliMoal, 2023-04-12:
That's exactly what I'm looking for. This file will be created when we build the program and with every build it will be changed. Is there any way to change it regardless of partition csv file? I mean the *app* key in this Json file.

brianignacio5, 2023-04-28:
Try the [vsix installer]() to add an option for right click on a .bin file and enter a custom hexadecimal offset value to flash on your serial device.

PR #957 (2023-05-04): add custom offset for flash binaries
## Description

Add the option for custom offset for flashing a binary to a given hexadecimal offset.

Fixes #934

## Type of change

Please delete options that are not relevant.
- New feature (non-breaking change which adds functionality)

## Steps to test this pull request

Provide a list of steps to test changes in this PR and required output

1. Right click on a binary (.bin) and  run the "Flash binary to partition..." command.
2. Select custom offset and enter as hexadecimal value (example 0x8000).
3. See the output in the ESP-IDF Output channel of the [... cut]
closed 2023-05-04 by brianignacio5