espressif/vscode-esp-idf-extension #938: ASP-ID Size Analysis does not work in 1.6.1

Gidrodinamika43, 2023-04-14:
### OS

Windows

### Operating System version

windows 10

### Visual Studio Code version

1.77.1

### ESP-IDF version

4.4.4

### Python version

3.8.7

### Doctor command output

esp32 devkit

### Extension

[esp_idf_vsc_ext.log](https://github.com/espressif/vscode-esp-idf-extension/files/11231397/esp_idf_vsc_ext.log)

### Description

ESP-ID Size Analysis works on plugin version 1.5.1. not on version 1.6.1. The tab is completely empty. the direct request of the utility on python works everywhere.

1.5.1
[image]

1.6.1
[image]

### Debug Message

```plain
none?)
```

### Other Steps to Reproduce

none

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

tshcherban, 2023-04-24:
Also not working with 5.0.1

txf-, 2023-04-24:
I can confirm that I'm seeing the same thing on Linux.

brianignacio5, 2023-04-28:
Could you try this [vsix installer](https://nightly.link/espressif/vscode-esp-idf-extension/actions/artifacts/670365106.zip) ?

radurentea, 2023-05-03:
> Could you try this [vsix installer](https://nightly.link/espressif/vscode-esp-idf-extension/actions/artifacts/670365106.zip) ?
I've tried it and it fixes the issue.
@swkang0513 could you try it as well?

PR #955 (2023-05-04): fix size gui rm util package
## Description

Fix UI of `ESP-IDF Size analysis of binaries` web view by removing deprecated `util` package

Fixes #938 

## Type of change

Please delete options that are not relevant.
- Bug fix (non-breaking change which fixes an issue)

## Steps to test this pull request

Provide a list of steps to test changes in this PR and required output

1. Click on "ESP-IDF Size analysis of binaries" command
2. Check the size analysis is shown

## How has this been tested?

Manual test in UI

**Test Configuration**:
* ESP-IDF Version: 5.0
* OS (Windows,Linux and macOS): W [... cut]