espressif/vscode-esp-idf-extension #939: Cannot install on Windows because my Windows user id has a space in it

roblatour, 2023-04-14:
### OS

Windows

### Operating System version

windows 11

### Visual Studio Code version

1.77.3

### ESP-IDF version

5.0.1

### Python version

?

### Doctor command output

.

### Extension

I tried the defauts (which include my user name)
I also tried creating a directory c:/ESP and tried using it too, it did not work

### Description

Cannot install on Windows because my Windows user id has a space in it.

I read the instructions at 
https://github.com/espressif/vscode-esp-idf-extension/blob/master/docs/tutorial/install.md
more specifically, 
NOTE: Make sure that IDF_PATH and IDF_TOOLS_PATH doesn't have any spaces to avoid any build issues.

However when I tried to change those two path in the installer set up screen, the installer ignored what I had entered and threw an error include a path with my user name (which has a space in it).

Setting up a new user on my computer just for the sake of this one piece of software is a non starter.  

If there is another work-around please advise, and I would also too suggest that the document at the above link be updated too.

### Debug Message

```plain
.
```

### Other Steps to Reproduce

.

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

brianignacio5, 2023-04-18:
When you try to use another location, what error do you receive ? Can you describe the steps you are following ?

roblatour, 2023-04-18:
Here is a screenshot of the error I get:

[image]

Here too is a video of the install process (I had to zip it to be able to upload it)
[vid2.zip](https://github.com/espressif/vscode-esp-idf-extension/files/11261945/vid2.zip)

brianignacio5, 2023-04-28:
I just tested the extension using spaces in PATH and the setup went through. Which version of the extension are you using ? Did you update to latest version ?

roblatour, 2023-04-28:
As shown in the video (inside the zip file) of my previous post I was using v1.6.1 of the extension, trying to install v5.0.1 of esp-idf

Having that said, this morning I gave it another try, creating a folder on my c: drive called 
c:\esp
and using that for the container name and c:\esp\.espressif for the tool directory and it seemed to work fine.

However, I wasn't paying attention to the version of the IDF I install (v5.0 or v5.0.1) so I uninstalled the visual studio extension and deleted the c:\esp folders contents and tried again trying to install 5.0, then 5.0.1, the the master branch.  In all cases I got the same error about the spaces in the path name (as from my original post). 

Of note, I have tried, reloading after the extention uninstall, exiting visual studio code, and even rebooting - but the problem remains.  I assume  something is being cached as a temp file someplace??

roblatour, 2023-05-02:
I finally found a work-around by creating a directory on my c: drive, and putting both the container and tool set directories in it as follows:

[image]

Eplankton, 2023-05-03:
Exactly the same problem with me.

roblatour, 2023-05-20:
stale but unresolved, would like to see a fix for the future.

Manoshanmugam, 2023-07-31:
Even I am facing the same issue while installing , can someone guide me to install espidf with vs code extension?

lmatter, 2024-01-30:
This is not necessarily an esp-idf issue.  I sadly have a space in my windows username and had this problem building Marlin 3D printer firmware - I think it's a VSCode issue.
Anyway, change the folders in configuration to ones without spaces.

Teunderbolt, 2024-02-09:
Installation goes well (installed in a dedicated folder directly on C://) tested it with version 1.6.5 and with the latest version 1.7.0. When building I get an error, because the project folder is in my user folder and my user folder has a space. A quick search learned me that this has been resolved with platformio but not with espressifs own vscode extension. I beg you please fix this.

brianignacio5, 2024-06-07:
Should be fixed by PR #1159

closed 2024-06-07 by brianignacio5