espressif/vscode-esp-idf-extension #941: I have some problems to build the project

hdpklm, 2023-04-17:
### OS

Windows

### Operating System version

10

### Visual Studio Code version

1.6.1

### ESP-IDF version

4.4, not sure from where I can find the version

### Python version

3.9

### Doctor command output

not use docker

### Description

I have some problems to build the project

```sh
# by useing idf
C:\hassan\programs\esp\esp-idf>idf.py -p com24 flash
Please use idf.py only in an ESP-IDF shell environment.
```

```sh
# by using build project in left-bottom of vscode
 *  Executing task: c:\hassan\programs\.espressif\python_env\idf4.3_py3.8_env\Scripts\python.exe c:\hassan\programs\esp\esp-idf\tools\idf_size.py c:\hassan\tools\bootloader-security\build\esp32-s3.map 

c:\hassan\programs\.espressif\python_env\idf4.3_py3.8_env\Scripts\python.exe: No module named esp_idf_size

 *  The terminal process "C:\WINDOWS\System32\cmd.exe /d /c c:\hassan\programs\.espressif\python_env\idf4.3_py3.8_env\Scripts\python.exe c:\hassan\programs\esp\esp-idf\tools\idf_size.py c:\hassan\tools\bootloader-security\build\esp32-s3.map" terminated with exit code: 1. 
```

### Debug Message

```plain
*  Executing task: c:\hassan\programs\.espressif\python_env\idf4.3_py3.8_env\Scripts\python.exe c:\hassan\programs\esp\esp-idf\tools\idf_size.py c:\hassan\tools\bootloader-security\build\esp32-s3.map 

c:\hassan\programs\.espressif\python_env\idf4.3_py3.8_env\Scripts\python.exe: No module named esp_idf_size

 *  The terminal process "C:\WINDOWS\System32\cmd.exe /d /c c:\hassan\programs\.espressif\python_env\idf4.3_py3.8_env\Scripts\python.exe c:\hassan\programs\esp\esp-idf\tools\idf_size.py c:\hassan\tools\bootloader-security\build\esp32-s3.map" terminated with exit code: 1.
```

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

brianignacio5, 2023-04-18:
Follow the documentation to [install and configure the extension here](https://github.com/espressif/vscode-esp-idf-extension/blob/master/docs/tutorial/install.md)

closed 2023-04-18 by brianignacio5

hdpklm, 2023-04-18:
> I think it has to do with commit 3e0f612a7e9ada9299e5b9f469ac0cba4c939b3e of esp-idf.
> you need to execute ./install.bat and export.bat located in your esp-idf folder (c:\hassan\programs\esp\esp-idf) to get the esp-idf-size package 

hi, @stephaistos I already execute the install.bat and export.bat

the problem begin after install the las version of idf, i remove the old version and install the new one, and restart vscode and the terminal, but nothing change

hdpklm, 2023-04-18:
> Please share the required troubleshooting information as described in the troubleshooting guide and describe your configuration.
> 
> Follow the documentation to [install and configure the extension here](https://github.com/espressif/vscode-esp-idf-extension/blob/master/docs/tutorial/install.md)

hi @brianignacio5 
I already have the extension of esp in vscode, but I install the new version of idf, I have to remove the extension, esp-idf, and .espressif. then install the extension only ???

hdpklm, 2023-04-18:
hi @brianignacio5 
no I´m not use docker, I install the extension and the esp-idf in the local computer.
I restart the computer and reexecute the install.bat and export.bat, but nothing change.

It sounds to me like I have to activate the conda environment or something like that

hdpklm, 2023-04-18:
I don't know if it affects but I have installed the vscode extension for arduino and platform

the arduino extension is much easier to use, but only supports up to version 1.6
and platformio is for the 3d printer, the framework is only compiled with platformio

but before updating the idf it worked very well, but it did not support the version of ESP32-S3

hdpklm, 2023-04-18:
I try to reinstall extension in vscode, but nothing change

but when I run C:\hassan\programs\esp\esp-idf\export.bat in the same terminal its work, but when I open a new terminal cant find the idf.py

this is the output of export.bat

```sh
Microsoft Windows [Versión 10.0.19044.2846]
(c) Microsoft Corporation. Todos los derechos reservados.

C:\hassan>idf
Please use idf.py only in an ESP-IDF shell environment.

C:\hassan>idf.py
Please use idf.py only in an ESP-IDF shell environment.

C:\hassan>C:\hassan\programs\esp\esp-idf\export.bat
Checking Python compatibility
Setting IDF_PATH: C:\hassan\programs\esp\esp-idf

Adding ESP-IDF tools to PATH...
Not using an unsupported version of tool xtensa-esp32-elf found in PATH: esp-2021r1.
Not using an unsupported version of tool cmake found in PATH: 3.24.2.
    c:\hassan\programs\esp\.espressif\tools\xtensa-esp-elf-gdb\12.1_20221002\xtensa-esp-elf-gdb\bin
    c:\hassan\programs\esp\.espressif\tools\riscv32-esp-elf-gdb\12.1_20221002\riscv32-esp-elf-gdb\bin
    c:\hassan\programs\esp\.espressif\tools\xtensa-esp32-elf\esp-12.2.0_20230208\xtensa-esp32-elf\bin
    c:\hassan\programs\esp\.espressif\tools\xtensa-esp32s2-elf\esp-12.2.0_20230208\xtensa-esp32s2-elf\bin
    c:\hassan\programs\esp\.espressif\tools\xtensa-esp32s3-elf\esp-12.2.0_20230208\xtensa-esp32s3-elf\bin
    c:\hassan\programs\esp\.espressif\tools\riscv32-esp-elf\esp-12.2.0_20230208\riscv32-esp-elf\bin
    c:\hassan\programs\esp\.espressif\tools\esp32ulp-elf\2.35_20220830\esp32ulp-elf\bin
    c:\hassan\programs\esp\.espressif\tools\cmake\3.24.0\bin
    c:\hassan\programs\esp\.espressif\tools\openocd-esp32\v0.12.0-esp32-20230313\openocd-esp32\bin
    c:\hassan\programs\esp\.espressif\tools\ninja\1.10.2\
    c:\hassan\programs\esp\.espressif\tools\idf-exe\1.0.3\
    c:\hassan\programs\esp\.espressif\tools\ccache\4.8\ccache-4.8-windows-x86_64
    c:\hassan\programs\esp\.espressif\tools\dfu-util\0.11\dfu-util-0.11-win64
    C:\hassan\programs\esp\.espressif\python_env\idf4.3_py3.9_env\Scripts\Scripts

Checking if Python packages are up to date...
Constraint file: c:\hassan\programs\esp\.espressif\espidf.constraints.v5.1.txt
Requirement files:
 - C:\hassan\programs\esp\esp-idf\tools\requirements\requirements.core.txt
Python being checked: C:\hassan\programs\esp\.espressif\python_env\idf4.3_py3.9_env\Scripts\Scripts\python.exe
Python requirements are satisfied.

Done! You can now compile ESP-IDF projects.
Go to the project directory and run:

  idf.py build

C:\hassan>idf.py
Usage: idf.py [OPTIONS] COMMAND1 [ARGS]... [COMMAND2 [ARGS]...]...

  ESP-IDF CLI build management tool. For commands that are not known to idf.py an attempt to execute it as a build system target will be
  made. Selected target: None

Options:
  --version                       Show IDF version and exit.
  --list-targets                  Print list of supported targets and exit.
  -C, --project-dir PATH          Project directory.
  -B, --build-dir PATH            Build directory.
  -w, --cmake-warn-uninitialized / -n, --no-warnings
                                  Enable CMake uninitialized variable warnings for CMake files inside the project directory. (--no-
                                  warnings is now the default, and doesn't need to be specified.) The default value can be set with the
                                  IDF_CMAKE_WARN_UNINITIALIZED environment variable.
  -v, --verbose                   Verbose build output.
  --preview                       Enable IDF features that are still in preview.
  --ccache / --no-ccache          Use ccache in build. Disabled by default. The default value can be set with the IDF_CCACHE_ENABLE
                                  environment variable.
  -G, --generator [Ninja]         CMake generator.
[... 42 lines cut ...]
  monitor                      Display serial output.
  openocd                      Run openocd from current path
  pack-component               (Deprecated) Deprecated! New CLI command: "compote component pack". Create component archive and store it
                               in the dist directory.
  partition-table              Build only partition table.
  partition-table-flash        Flash partition table only.
  post-debug                   Utility target to read the output of async debug action and stop them.
  python-clean                 Delete generated Python byte code from the IDF directory
  read-otadata                 Read otadata partition.
  reconfigure                  Re-run CMake.
  save-defconfig               Generate a sdkconfig.defaults with options different from the default ones
  set-target                   Set the chip target to build.
  show-efuse-table             Print eFuse table.
  size                         Print basic size information about the app.
  size-components              Print per-component size information.
  size-files                   Print per-source-file size information.
  uf2                          Generate the UF2 binary with all the binaries included
  uf2-app                      Generate an UF2 binary for the application only
  upload-component             (Deprecated) Deprecated! New CLI command: "compote component upload". Upload component to the component
                               registry. If the component doesn't exist in the registry it will be created automatically.
  upload-component-status      (Deprecated) Deprecated! New CLI command: "compote component upload-status". Check the component uploading
                               status by the job ID.

C:\hassan>
```

brianignacio5, 2023-04-18:
IF you want to open a terminal with the extension settings use the `ESP-IDF: Open ESP-IDF Terminal` command from the extension. The extension doesn't configure any terminal being open but just the one with this command.

hdpklm, 2023-04-18:
ok thank you