espressif/vscode-esp-idf-extension #942: non-root user devcontainer has no permission to access ttyUSB*

dremsol, 2023-04-17:
### OS

Windows

### Operating System version

Windows 11 - WSL2

### Visual Studio Code version

1.77.3

### ESP-IDF version

ESP-IDF v5.1-dev-4557-gf404fe96b1

### Python version

3.8.10

### Doctor command output

-

### Extension

-

### Description

non-root devcontainer user has no access to /dev/ttyUSB*

### Debug Message

```plain
--- idf_monitor on /dev/ttyUSB0 115200 ---
--- Quit: Ctrl+] | Menu: Ctrl+T | Help: Ctrl+T followed by Ctrl+H ---
Exception in thread Thread-2:
Traceback (most recent call last):
  File "/opt/esp/python_env/idf5.1_py3.8_env/lib/python3.8/site-packages/serial/serialposix.py", line 322, in open

    self.fd = os.open(self.portstr, os.O_RDWR | os.O_NOCTTY | os.O_NONBLOCK)
PermissionError: [Errno 13] Permission denied: '/dev/ttyUSB0'
''
```

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

brianignacio5, 2023-04-18:
Please share the troubleshooting information as requested in the issue template.

Did you configure the extension like in [WSL tutorial](https://github.com/espressif/vscode-esp-idf-extension/blob/master/docs/tutorial/wsl.md) ?

dremsol, 2023-04-18:
```
esp@f269daa4f011:/home/willem/projects/remsol/esp32-blink/pwm$ stat /dev/ttyUSB0 
  File: /dev/ttyUSB0
  Size: 0               Blocks: 0          IO Block: 4096   character special file
Device: 100019h/1048601d        Inode: 153         Links: 1     Device type: bc,0
Access: (0660/crw-rw----)  Uid: (    0/    root)   Gid: (   20/ dialout)
Access: 2023-04-18 15:14:36.682840195 +0000
Modify: 2023-04-18 15:14:36.682840195 +0000
Change: 2023-04-18 15:14:36.682840195 +0000
 Birth: -
```

adding esp user to dialout group by following line in dockerfile solves it
```
RUN groupadd --gid $USER_GID $CONTAINER_USER \
    && adduser --uid $USER_UID --gid $USER_GID --disabled-password --gecos "" ${CONTAINER_USER} \
    && usermod -a -G dialout $CONTAINER_USER
```

not sure if this is the preferred way though

brianignacio5, 2023-04-19:
Thanks for the information @dremsol Will update the template docker file.

brianignacio5, 2023-04-19:
The fix is in the linked PR. here the link to the [vsix installer](https://nightly.link/espressif/vscode-esp-idf-extension/actions/artifacts/654272050.zip)

PR #944 (2023-04-28): add container user to dialout group
## Description

Add container user to dialout to allow serial port access.

Fix #942 

## Type of change

- Bug fix (non-breaking change which fixes an issue)

## Steps to test this pull request

Provide a list of steps to test changes in this PR and required output

1. Start VS Code, run the Dev Containers: Open Folder in Container... command from the Command Palette (F1) or quick actions Status bar item, and select the project folder you would like to set up the container for.
2. Run `ESP-IDF Monitor` and see if it works.

- Expected behaviour: Monitor sta [... cut]
closed 2023-04-28 by brianignacio5