espressif/vscode-esp-idf-extension #949: Ubuntu 23.04 has adopted PEP 668 and this is preventing the extension from configuring

moefear85, 2023-04-22:
### OS

Linux

### Operating System version

Ubuntu 23.04

### Visual Studio Code version

1.77.3

### ESP-IDF version

master

### Python version

3.11.2

### Doctor command output

---------------------------------------------- ESP-IDF Extension for Visual Studio Code report ---------------------------------------------
OS linux x64 6.2.0-20-generic 
System environment variable IDF_PYTHON_ENV_PATH 
 undefined 
System environment variable PATH 
 <HOMEPATH>/.local/bin:<HOMEPATH>/software/miniconda/bin:<HOMEPATH>/.local/bin:/usr/local/sbin:/usr/local/bin:/usr/sbin:/usr/bin:/sbin:/bin:/usr/games:/usr/local/games:/snap/bin:<HOMEPATH>/software/espressif/tools/xtensa-lx106-elf/bin:<HOMEPATH>/scripts:/media/moefear/BACKUP/scripts:<HOMEPATH>/software/espressif/tools/xtensa-lx106-elf/bin:<HOMEPATH>/scripts:/media/moefear/BACKUP/scripts 
System environment variable PYTHON 
 undefined 
Visual Studio Code version 1.77.3 
Visual Studio Code language en 
Visual Studio Code shell /usr/bin/bash 
ESP-IDF Extension version 1.6.1 
Workspace folder /tmp/t1 
---------------------------------------------------- Extension configuration settings ------------------------------------------------------
ESP-ADF Path (idf.espAdfPath) <HOMEPATH>/source/esp-adf
ESP-IDF Path (idf.espIdfPath) <HOMEPATH>/source/esp-idf
ESP-MDF Path (idf.espMdfPath) ${env:MDF_PATH}
ESP-Matter Path (idf.espMatterPath) ${env:ESP_MATTER_PATH}
Custom extra paths (idf.customExtraPaths) <HOMEPATH>/software/espressif/tools/xtensa-esp-elf-gdb/12.1_20221002/xtensa-esp-elf-gdb/bin:<HOMEPATH>/software/espressif/tools/riscv32-esp-elf-gdb/12.1_20221002/riscv32-esp-elf-gdb/bin:<HOMEPATH>/software/espressif/tools/xtensa-esp32-elf/esp-12.2.0_20230208/xtensa-esp32-elf/bin:<HOMEPATH>/software/espressif/tools/xtensa-esp32s2-elf/esp-12.2.0_20230208/xtensa-esp32s2-elf/bin:<HOMEPATH>/software/espressif/tools/xtensa-esp32s3-elf/esp-12.2.0_20230208/xtensa-esp32s3-elf/bin:<HOMEPATH>/software/espressif/tools/riscv32-esp-elf/esp-12.2.0_20230208/riscv32-esp-elf/bin:<HOMEPATH>/software/espressif/tools/esp32ulp-elf/2.35_20220830/esp32ulp-elf/bin:<HOMEPATH>/software/espressif/tools/openocd-esp32/v0.12.0-esp32-20230313/openocd-esp32/bin:<HOMEPATH>/software/espressif/tools/esp-rom-elfs/20230320
Custom extra vars (idf.customExtraVars)
    OPENOCD_SCRIPTS: <HOMEPATH>/software/espressif/tools/openocd-esp32/v0.12.0-esp32-20230313/openocd-esp32/share/openocd/scripts
    ESP_ROM_ELF_DIR: <HOMEPATH>/software/espressif/tools/esp-rom-elfs/20230320/
Virtual env Python Path (idf.pythonBinPath) python
Serial port (idf.port) /dev/ttyUSB1
OpenOCD Configs (idf.openOcdConfigs) board/esp32-wrover-kit-3.3v.cfg
ESP-IDF Tools Path (idf.toolsPath) <HOMEPATH>/software/espressif
Git Path (idf.gitPath) git
-------------------------------------------------------- Configurations access -------------------------------------------------------------
Access to ESP-ADF Path (idf.espAdfPath) false
Access to ESP-IDF Path (idf.espIdfPath) true
Access to ESP-MDF Path (idf.espMdfPath) false
Access to ESP-Matter Path (idf.espMatterPath) false
Access to ESP-IDF Custom extra paths
Access to <HOMEPATH>/software/espressif/tools/xtensa-esp-elf-gdb/12.1_20221002/xtensa-esp-elf-gdb/bin: true
Access to <HOMEPATH>/software/espressif/tools/riscv32-esp-elf-gdb/12.1_20221002/riscv32-esp-elf-gdb/bin: true
Access to <HOMEPATH>/software/espressif/tools/xtensa-esp32-elf/esp-12.2.0_20230208/xtensa-esp32-elf/bin: true
Access to <HOMEPATH>/software/espressif/tools/xtensa-esp32s2-elf/esp-12.2.0_20230208/xtensa-esp32s2-elf/bin: true
Access to <HOMEPATH>/software/espressif/tools/xtensa-esp32s3-elf/esp-12.2.0_20230208/xtensa-esp32s3-elf/bin: true
Access to <HOMEPATH>/software/espressif/tools/riscv32-esp-elf/esp-12.2.0_20230208/riscv32-esp-elf/bin: true
Access to <HOMEPATH>/software/espressif/tools/esp32ulp-elf/2.35_20220830/esp32ulp-elf/bin: true
Access to <HOMEPATH>/software/espressif/tools/openocd-esp32/v0.12.0-esp32-20230313/openocd-esp32/bin: true
Access to <HOMEPATH>/software/espressif/tools/esp-rom-elfs/20230320: true
Access to Virtual env Python Path (idf.pythonBinPath) false
Access to CMake in environment PATH true
Access to Ninja in environment PATH true
Access to ESP-IDF Tools Path (idf.toolsPath) true
-------------------------------------------------------- Configurations has spaces -------------------------------------------------------------
Spaces in system environment Path false
[... 185 lines cut ...]

× This environment is externally managed
╰─> To install Python packages system-wide, try apt install
    python3-xyz, where xyz is the package you are trying to
    install.
    
    If you wish to install a non-Debian-packaged Python package,
    create a virtual environment using python3 -m venv path/to/venv.
    Then use path/to/venv/bin/python and path/to/venv/bin/pip. Make
    sure you have python3-full installed.
    
    If you wish to install a non-Debian packaged Python application,
    it may be easiest to use pipx install xyz, which will manage a
    virtual environment for you. Make sure you have pipx installed.
    
    See /usr/share/doc/python3.11/README.venv for more information.
```

### Other Steps to Reproduce

I tried using a python virtual environment, but the extension detects this, and for some reason, complains and doesn't accept it. I feel this is an unnecessary constraint.

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

moefear85, 2023-04-22:
note that virtualenv is already installed on my system.

mickeyl, 2023-04-26:
This has little to do with ESP-IDF or this extension (although it's ­– among hundreds of other packages ­also ­– affected). Please also refer to https://github.com/pypa/pip/issues/11776 and https://discuss.python.org/t/pep-668-marking-python-base-environments-as-externally-managed/10302 where workarounds / solutions are discussed. I'm not sure whether this extension can do anything to work around it.

moefear85, 2023-04-26:
here's the beef: the extension insists on the system-wide python when its unnecessary, which would have avoided the problem. the guys at ubuntu atleast know what their talking about, and any script that needs to install packages should allow the specification of an arbitary venv for that

brianignacio5, 2023-04-28:
Added the fix in this [vsix installer](https://nightly.link/espressif/vscode-esp-idf-extension/actions/artifacts/670362036.zip)

ankit-thealchemist, 2023-05-04:
I am still facing the error after installing the above vsix installer.
idf_versions.txt has (1) KB
idf_versions.txt progress: 100.00% (0.08 / 0.08) KB
/bin/sh: 1: /home/ankit/.espressif/python_env/idf4.4_py3.11_env/: Permission denied
Command failed: "/home/ankit/.espressif/python_env/idf4.4_py3.11_env/" "/media/ankit/Win_D/IOTCode/framework/esp-idf/tools/check_python_dependencies.py" -r "/media/ankit/Win_D/IOTCode/framework/esp-idf/tools/requirements/requirements.core.txt"
/bin/sh: 1: /home/ankit/.espressif/python_env/idf4.4_py3.11_env/: Permission denied
Installing ESP-IDF extension python packages in undefined ...

/bin/sh: 1: undefined: not found
Command failed: "undefined" -m pip install --upgrade --constraint "/home/ankit/.vscode/extensions/espressif.esp-idf-extension-1.6.1/espidf.constraints.txt" --no-warn-script-location  -r "/home/ankit/.vscode/extensions/espressif.esp-idf-extension-1.6.1/requirements.txt"
/bin/sh: 1: undefined: not found
idf_versions.txt has (1) KB
idf_versions.txt progress: 100.00% (0.08 / 0.08) KB
/bin/sh: 1: /home/ankit/.espressif/python_env/idf4.4_py3.11_env/: Permission denied
Command failed: "/home/ankit/.espressif/python_env/idf4.4_py3.11_env/" "/media/ankit/Win_D/IOTCode/framework/esp-idf/tools/check_python_dependencies.py" -r "/media/ankit/Win_D/IOTCode/framework/esp-idf/tools/requirements/requirements.core.txt"
/bin/sh: 1: /home/ankit/.espressif/python_env/idf4.4_py3.11_env/: Permission denied

brianignacio5, 2023-05-04:
You can see the issue is that your user has no permission to write in folder `/home/ankit/.espressif/python_env/idf4.4_py3.11_env/`

ankit-thealchemist, 2023-05-04:
> You can see the issue is that your user has no permission to write in folder `/home/ankit/.espressif/python_env/idf4.4_py3.11_env/`

could you let me know how to enable it!

brianignacio5, 2023-05-04:
In a terminal write `sudo chmod 777 /home/ankit/.espressif/python_env/idf4.4_py3.11_env`

ankit-thealchemist, 2023-05-04:
problem solved by putting the idf python(python virtual environment) in the settings.
[image]

PR #956 (2023-05-04): remove virtualenv install from extension
## Description

Remove virtualenv install from extension setup process.

Fixes #919 
Fix #949

## Type of change

- Bug fix (non-breaking change which fixes an issue)

## Steps to test this pull request

1. Click on "Configure ESP-IDF extension" command.
2. Execute setup process using Python 3.11.
3. Observe results. Install setup should work.

## How has this been tested?

Manual test of setup install

**Test Configuration**:
* ESP-IDF Version: 5.0
* OS (Windows,Linux and macOS): macOS

## Checklist
- [ ] PR Self Reviewed
- [ ] Applied Code formatting
 [... cut]
closed 2023-05-04 by brianignacio5