espressif/vscode-esp-idf-extension #952: Difficulty observing watched variables in real-time

zeynepdicle, 2023-04-25:
### OS

Windows

### Operating System version

windows 10

### ESP-IDF version

5.0.1

### Python version

3.8.7

### Doctor command output

none

### Description

What do I need to do to constantly observe the variable I am watching in watch? Currently, I can only observe it when I set a breakpoint or when the debugger stops.

### Debug Message

```plain
none
```

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

brianignacio5, 2023-04-26:
Could you describe a bit further your issue ? Can't understand the issue you are facing

zeynepdicle, 2023-04-26:
During debugging process, I am utilizing the watch window to monitor the state of variables. However, this is only possible when I hit a breakpoint or pause the program execution. I would like to continuously monitor the variables while in debug mode. How can I achieve this?

brianignacio5, 2023-04-26:
Well the question is why can't you see them in the Variables tab instead ?

https://code.visualstudio.com/docs/editor/debugging#_data-inspection

Any minimal example to reproduce ?

zeynepdicle, 2023-04-26:
I can't watch variables while debugging

This is my code
```
#include <stdio.h>
#include "freertos/FreeRTOS.h"
#include "freertos/task.h"
#include "driver/gpio.h"
#include "esp_log.h"
#include "led_strip.h"
#include "sdkconfig.h"

static const char *TAG = "example";

/* Use project configuration menu (idf.py menuconfig) to choose the GPIO to blink,
   or you can edit the following line and set a number here.
*/
#define BLINK_GPIO 10
#define BUTTON_GPIO 9
int counter = 0;

static uint8_t s_led_state = 0;

static led_strip_handle_t led_strip;

static void blink_led(void) {
  /* If the addressable LED is enabled */
  if (s_led_state) {
    /* Set the LED pixel using RGB from 0 (0%) to 255 (100%) for each color */
    led_strip_set_pixel(led_strip, 0, 0, 200, 200);
    /* Refresh the strip to send data */
    led_strip_refresh(led_strip);
  } else {
    /* Set all LED off to clear all pixels */
    led_strip_clear(led_strip);
  }
}

static void configure_led(void) {
  gpio_set_direction(BUTTON_GPIO, GPIO_MODE_INPUT);
  gpio_set_pull_mode(BUTTON_GPIO, GPIO_PULLUP_ONLY);

  ESP_LOGI(TAG, "Example configured to blink addressable LED!");
  /* LED strip initialization with the GPIO and pixels number*/
  led_strip_config_t strip_config = {
    .strip_gpio_num = BLINK_GPIO,
    .max_leds = 1,  // at least one LED on board
  };
  led_strip_rmt_config_t rmt_config = {
    .resolution_hz = 10 * 1000 * 1000,  // 10MHz
  };
  ESP_ERROR_CHECK(led_strip_new_rmt_device(&strip_config, &rmt_config, &led_strip));
  /* Set all LED off to clear all pixels */
  led_strip_clear(led_strip);
}

void app_main(void) {

  /* Configure the peripheral according to the LED type */
  configure_led();

  while (1) {
    counter++;
    ESP_LOGI(TAG, "Turning the LED %s!", s_led_state == true ? "ON" : "OFF");
    blink_led();
    /* Toggle the LED state */
    s_led_state = !s_led_state;
    vTaskDelay(1000 / portTICK_PERIOD_MS);
  }
}
```

While debugging is ongoing
[image]

While debugging is pause
[image]

zeynepdicle, 2023-04-26:
Actually, what I want to do is to monitor variables by live watching them.

closed 2023-05-25 by zeynepdicle