espressif/vscode-esp-idf-extension #954: [Feature Request]: Running the size task after the build task succeeds should not be so interruptive

henryshunt, 2023-04-26:
After the build task completes without errors, the size task runs. This is very annoying because it opens a new terminal tab when I'm trying to read the output of the build task. If I catch that the build task has something I wish to look at (e.g. warnings), the size task quickly switches in and I have to click to get back to the build task. Also, I often start the build task and then close the terminal, but the size task inevitably causes it to open back up again unannounced. Finally, I personally don't really have any need to see the size info, certainly not after every single build.

The contents of the size task should run as part of the build task (so it doesn't open a separate tab), or there should be an option to disable the size task from automatically running after the build task.

brianignacio5, 2023-04-27:
Hi @henryshunt thank you for your feedback. Will add an extension setting to disable this size task after build and share the vsix installer for you to try and see if it's okay.

I'll add a link in a bit from the PR

henryshunt, 2023-04-27:
Much appreciated, thank you!

brianignacio5, 2023-05-16:
Could you try this [vsix installer](https://nightly.link/espressif/vscode-esp-idf-extension/actions/artifacts/682972274.zip) to see if the issue is fixed ?

vsh-ektos, 2023-05-18:
@brianignacio5 It seems to work well, but the build task (not the build itself) does not complete
[image]

PR #965 (2023-05-23): add enableSizeTaskAfterBuildTask setting
## Description

Add `idf.enableSizeTaskAfterBuildTask` to enable or disable the IDF Size task after the build task.

Fixes #954

## Type of change

Please delete options that are not relevant.
- Bug fix (non-breaking change which fixes an issue)
- New feature (non-breaking change which adds functionality)
- Breaking change (fix or feature that would cause existing functionality to not work as expected)
- This change requires a documentation update

## Steps to test this pull request

Provide a list of steps to test changes in this PR and required output [... cut]
closed 2023-05-23 by brianignacio5