espressif/vscode-esp-idf-extension #958: Cannot add breakpoint if running.

AndrewCapon, 2023-04-30:
### OS

MacOS

### Operating System version

macOS 12.6.5

### Visual Studio Code version

 1.77.3

### ESP-IDF version

5.0.1

### Python version

3.9.12

### Doctor command output

---------------------------------------------- ESP-IDF Extension for Visual Studio Code report ---------------------------------------------
OS darwin x64 21.6.0 
System environment variable IDF_PYTHON_ENV_PATH 
 undefined 
System environment variable PATH 
 /usr/local/bin:/usr/bin:/bin:/usr/sbin:/sbin:/usr/local/share/dotnet:~/.dotnet/tools:/Library/Apple/usr/bin:/Library/Frameworks/Mono.framework/Versions/Current/Commands:/Applications/ARM/bin:/Applications/CMake.app/Contents/bin:/Applications/STM32CubeIDE.app/Contents/Eclipse/plugins/com.st.stm32cube.ide.mcu.externaltools.openocd.macos64_2.1.0.202110201500/tools/bin: 
System environment variable PYTHON 
 undefined 
Visual Studio Code version 1.77.3 
Visual Studio Code language en 
Visual Studio Code shell /bin/zsh 
ESP-IDF Extension version 1.6.1 
Workspace folder /Volumes/T7/Development2/esp/VS/espidf/blink 
---------------------------------------------------- Extension configuration settings ------------------------------------------------------
ESP-ADF Path (idf.espAdfPath) 
ESP-IDF Path (idf.espIdfPath) /Volumes/T7/Development2/esp/VS/esp-idf
ESP-MDF Path (idf.espMdfPath) ${env:MDF_PATH}
ESP-Matter Path (idf.espMatterPath) ${env:ESP_MATTER_PATH}
Custom extra paths (idf.customExtraPaths) /Volumes/T7/Development2/esp/VS/espressif/tools/xtensa-esp-elf-gdb/11.2_20220823/xtensa-esp-elf-gdb/bin:/Volumes/T7/Development2/esp/VS/espressif/tools/riscv32-esp-elf-gdb/11.2_20220823/riscv32-esp-elf-gdb/bin:/Volumes/T7/Development2/esp/VS/espressif/tools/xtensa-esp32-elf/esp-2022r1-11.2.0/xtensa-esp32-elf/bin:/Volumes/T7/Development2/esp/VS/espressif/tools/xtensa-esp32s2-elf/esp-2022r1-11.2.0/xtensa-esp32s2-elf/bin:/Volumes/T7/Development2/esp/VS/espressif/tools/xtensa-esp32s3-elf/esp-2022r1-11.2.0/xtensa-esp32s3-elf/bin:/Volumes/T7/Development2/esp/VS/espressif/tools/riscv32-esp-elf/esp-2022r1-11.2.0/riscv32-esp-elf/bin:/Volumes/T7/Development2/esp/VS/espressif/tools/esp32ulp-elf/2.35_20220830/esp32ulp-elf/bin:/Volumes/T7/Development2/esp/VS/espressif/tools/openocd-esp32/v0.11.0-esp32-20221026/openocd-esp32/bin:/Volumes/T7/Development2/esp/VS/espressif/tools/esp-rom-elfs/20220823
Custom extra vars (idf.customExtraVars)
    OPENOCD_SCRIPTS: /Volumes/T7/Development2/esp/VS/espressif/tools/openocd-esp32/v0.11.0-esp32-20221026/openocd-esp32/share/openocd/scripts
    ESP_ROM_ELF_DIR: /Volumes/T7/Development2/esp/VS/espressif/tools/esp-rom-elfs/20220823/
Virtual env Python Path (idf.pythonBinPath) /Volumes/T7/Development2/esp/VS/espressif/python_env/idf5.0_py3.9_env/bin/python
Serial port (idf.port) /dev/cu.usbmodem4
OpenOCD Configs (idf.openOcdConfigs) board/esp32s3-builtin.cfg
ESP-IDF Tools Path (idf.toolsPath) /Volumes/T7/Development2/esp/VS/espressif
Git Path (idf.gitPath) git
-------------------------------------------------------- Configurations access -------------------------------------------------------------
Access to ESP-ADF Path (idf.espAdfPath) false
Access to ESP-IDF Path (idf.espIdfPath) true
Access to ESP-MDF Path (idf.espMdfPath) false
Access to ESP-Matter Path (idf.espMatterPath) false
Access to ESP-IDF Custom extra paths
Access to /Volumes/T7/Development2/esp/VS/espressif/tools/xtensa-esp-elf-gdb/11.2_20220823/xtensa-esp-elf-gdb/bin: true
Access to /Volumes/T7/Development2/esp/VS/espressif/tools/riscv32-esp-elf-gdb/11.2_20220823/riscv32-esp-elf-gdb/bin: true
Access to /Volumes/T7/Development2/esp/VS/espressif/tools/xtensa-esp32-elf/esp-2022r1-11.2.0/xtensa-esp32-elf/bin: true
Access to /Volumes/T7/Development2/esp/VS/espressif/tools/xtensa-esp32s2-elf/esp-2022r1-11.2.0/xtensa-esp32s2-elf/bin: true
Access to /Volumes/T7/Development2/esp/VS/espressif/tools/xtensa-esp32s3-elf/esp-2022r1-11.2.0/xtensa-esp32s3-elf/bin: true
Access to /Volumes/T7/Development2/esp/VS/espressif/tools/riscv32-esp-elf/esp-2022r1-11.2.0/riscv32-esp-elf/bin: true
Access to /Volumes/T7/Development2/esp/VS/espressif/tools/esp32ulp-elf/2.35_20220830/esp32ulp-elf/bin: true
Access to /Volumes/T7/Development2/esp/VS/espressif/tools/openocd-esp32/v0.11.0-esp32-20221026/openocd-esp32/bin: true
Access to /Volumes/T7/Development2/esp/VS/espressif/tools/esp-rom-elfs/20220823: true
Access to Virtual env Python Path (idf.pythonBinPath) true
Access to CMake in environment PATH true
Access to Ninja in environment PATH true
Access to ESP-IDF Tools Path (idf.toolsPath) true
-------------------------------------------------------- Configurations has spaces -------------------------------------------------------------
Spaces in system environment Path false
[... 148 lines cut ...]
TypeError: unhashable type: 'dict'
```

### Other Steps to Reproduce

Tried with S3Box, ProS3 and XIAO ESP32S3. All have same problem.

Eclipse esp-idf plugin works perfectly.

VSCode platformIO works perfectly.

To replicate:

1. Clean installation of esp-idf plugin
2. Install esp-idf and tools via plugin.
3. Create project using example
4. Set up jtag, flashing via jtag works fine.
5. Add breakpoint somewhere in loop code and Launch debugger, breakpoint is hit.
6. Continue execution, breakpoint is hit again.
7. Remove breakpoint and continue execution.
8. Try to add breakpoint while it is now running, breakpoint not hit as it is not added due to exception.

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

brianignacio5, 2023-05-04:
Could you provide more information about the error? What is the breakpoint request being sent to debug backend ? What kind of breakpoint are you trying to insert and where? What kind of value is being set ?

brianignacio5, 2023-05-04:
Change the `logLevel` to 4 in launch.json to see more output from debug adapter. There is also debug log file in your project workspace folder. Before the error there should some debug adapter messages and also the request to GDB from debug adapter

AndrewCapon, 2023-05-04:
Adding working breakpoint while paused:

```
[Debug Adapter]
2023-05-04 09:46:05,634 - Debug Adapter (ReaderThread) - DEBUG - read line: >>b'Content-Length: 223\\r\\n'<<

[Debug Adapter]
2023-05-04 09:46:05,634 - Debug Adapter (ReaderThread) - DEBUG - read line: >>b'\\r\\n'<<

2023-05-04 09:46:05,634 - Debug Adapter(Command Processor) - DEBUG - Got json: {
    "arguments": {
        "breakpoints": [
            {
                "line": 86
            }
        ],
        "lines": [
            86
        ],
        "source": {
            "path": "/Volumes/T7/Development2/esp/VS/espidf/blink/main/blink_example_main.c"
        },
        "sourceModified": false
    },
    "command": "setBreakpoints",
    "seq": 25,
    "type": "request"
}
2023-05-04 09:46:05,634 - Debug Adapter (WriterThread) - DEBUG - Writing: {"type": "event", "event": "output", "body": {"output": "Got json: {\n    \"arguments\": {\n        \"breakpoints\": [\n            {\n                \"line\": 86\n            }\n        ],\n        \"lines\": [\n            86\n        ],\n        \"source\": {\n            \"path\": \"/Volumes/T7/Development2/esp/VS/espidf/blink/main/blink_example_main.c\"\n        },\n        \"sourceModified\": false\n    },\n    \"command\": \"setBreakpoints\",\n    \"seq\": 25,\n    \"type\": \"request\"\n}\n", "source": {}}, "seq": 88}

[Debug Adapter]
2023-05-04 09:46:05,634 - Gdb - DEBUG - MI->: -break-insert -i 0 /Volumes/T7/Development2/esp/VS/espidf/blink/main/blink_example_main.c:86
[Debug Adapter]
2023-05-04 09:46:05,637 - Gdb - DEBUG - MI<-:
[{'message': 'done',
  'payload': {'bkpt': {'addr': '0x42007bc4',
                       'disp': 'keep',
                       'enabled': 'y',
                       'file': '/Volumes/T7/Development2/esp/VS/espidf/blink/main/blink_example_main.c',
                       'fullname': '/Volumes/T7/Development2/esp/VS/espidf/blink/main/blink_example_main.c',
                       'func': 'app_main',
                       'line': '86',
                       'number': '6',
                       'original-location': '/Volumes/T7/Development2/esp/VS/espidf/blink/main/blink_example_main.c:86',
                       'thread-groups': ['i1'],
                       'times': '0',
                       'type': 'breakpoint'}},
  'stream': 'stdout',
  'token': None,
  'type': 'result'}]
[Debug Adapter]
2023-05-04 09:46:05,637 - Gdb - DEBUG - RESULT: done {'bkpt': {'addr': '0x42007bc4',
          'disp': 'keep',
          'enabled': 'y',
          'file': '/Volumes/T7/Development2/esp/VS/espidf/blink/main/blink_example_main.c',
          'fullname': '/Volumes/T7/Development2/esp/VS/espidf/blink/main/blink_example_main.c',
          'func': 'app_main',
          'line': '86',
          'number': '6',
          'original-location': '/Volumes/T7/Development2/esp/VS/espidf/blink/main/blink_example_main.c:86',
          'thread-groups': ['i1'],
          'times': '0',
          'type': 'breakpoint'}}
2023-05-04 09:46:05,637 - Debug Adapter(Command Processor) - DEBUG - Processed command: setBreakpoints
[Debug Adapter]
2023-05-04 09:46:05,637 - Debug Adapter (WriterThread) - DEBUG - Writing: {"type": "response", "request_seq": 25, "success": true, "command": "setBreakpoints", "body": {"breakpoints": [{"line": 86, "verified": "true", "source": {"path": "/Volumes/T7/Development2/esp/VS/espidf/blink/main/blink_example_main.c"}}]}, "seq": 89}

2023-05-04 09:46:05,637 - Debug Adapter (WriterThread) - DEBUG - Writing: {"type": "event", "event": "output", "body": {"output": "Processed command: setBreakpoints\n", "source": {}}, "seq": 90}
```

brianignacio5, 2023-05-05:
@AndrewCapon It is not broken the debug adapter. The thing is that the user is trying to add a breakpoint while the program is running and not interrupting the execution. In the terminal you would usually run the interrupt command but the debug adapter when you set a breakpoint doesn't interrupt the program execution. Need to add the workflow of interrupting the program if executing before adding breakpoint.

AndrewCapon, 2023-05-05:
Hi @brianignacio5 

Thanks, this used to work, something must have changed in the last few weeks?

AndrewCapon, 2023-05-05:
Hi @brianignacio5 

This was working a few weeks ago, I know as I was using it.

Are you saying that adding a breakpoint in the debugger when code is running has never worked?

Wouldn't you have had loads or reports if this was the case?

PR #997 (2023-06-29): Enhance/debug pause breakpoint
## Description

Pause program execution when trying to set a breakpoint during execution

Fixes #958

## Type of change

Please delete options that are not relevant.
- Bug fix (non-breaking change which fixes an issue)

## Steps to test this pull request

Provide a list of steps to test changes in this PR and required output

1. Click on F5 to start debugging a esp-idf application
2. When the program is running, try to set a breakpoint. The program will pause.
3. Observe results.

- Expected behaviour:

After breakpoint is set when program is running, the program [... cut]
closed 2023-06-29 by brianignacio5