espressif/vscode-esp-idf-extension #961: ESP32C3 Failed to start debug adapter (OCD-741)

dalkvenk, 2023-05-04:
### Development Kit

ESP32C3 Dev Kit M1

### Module or chip used

ESP32C3

### Debug Adapter

Internal JTAG

### OpenOCD version

v0.11.0-esp32-20220706

### Operating System

MacOS

### Using an IDE ?

VS Code

### OpenOCD command line

openocd -f board/esp32c3-builtin.cfg

### JTAG Clock Speed

default

### ESP-IDF version

v5.0

### Problem Description

1. Use blink example from SDK
2. Flash the binary using internal JTAG using the command 
`openocd -f board/esp32c3-builtin.cfg -c "program_esp /Users/k.venkatraman/Cue-Hub/ESP-Projects/blink/build/blink.bin 0x10000 verify exit"`
3. **The internal JTAG debugger flashes the program successfully**.
4. Set up launch.json file as below
```
{
    "version": "0.2.0",
    "configurations": [
      {
        "type": "espidf",
        "name": "ESP-IDF OpenOcd",
        "request": "launch",
        "debugPort": 43474,
        "logLevel": 5,
        "mode": "auto",
        "tmoScaleFactor": 2,
      }
    ]
  }
```

5. Press F5 to start debugging.
6. Encounter the error: "Failed to connect to "remote 127.0.0.1:3333"

### Debug Logs

```shell
Logs from the Output console on VS Code:

[OpenOCD]
❌ Open On-Chip Debugger  v0.11.0-esp32-20220706 (2022-07-06-15:49)
[... 205 lines cut ...]
    "arguments": {
        "restart": false
    },
    "command": "disconnect",
    "seq": 2,
    "type": "request"
}
2023-04-24 10:43:17,024 - Debug Adapter (main) - DEBUG - Adapter is Stopping...
2023-04-24 10:43:17,024 - Debug Adapter (main) - DEBUG - Stopping target poller
2023-04-24 10:43:17,024 - Debug Adapter (main) - DEBUG - Closing the GDB process
2023-04-24 10:43:17,024 - Debug Adapter (main) - DEBUG - Closing the Socket
2023-04-24 10:43:17,024 - Debug Adapter (main) - DEBUG - Stopping of the Writer thread
2023-04-24 10:43:17,024 - Debug Adapter (WriterThread) - DEBUG - Writing: {"type": "event", "event": "output", "body": {"output": "Got json: {\n    \"arguments\": {\n        \"restart\": false\n    },\n    \"command\": \"disconnect\",\n    \"seq\": 2,\n    \"type\": \"request\"\n}\n", "source": {}}, "seq": 10}

2023-04-24 10:43:17,025 - Debug Adapter (main) - DEBUG - Stopping of the Reader thread
2023-04-24 10:43:17,025 - Debug Adapter (main) - DEBUG - Adapter is Stopped
DEBUG_ADAPTER_STOPPED
2023-04-24 10:43:17,025 - Debug Adapter (main) - CRITICAL - Debug adapter -> Extension: DEBUG_ADAPTER_STOPPED
2023-04-24 10:43:17,025 - Debug Adapter(Command Processor) - DEBUG - Processed command: disconnect
[Stopped] : ESP-IDF Debug Adapter
```

### Expected behavior

Internal JTAG debugger successfully launches the debug session.

brianignacio5, 2023-05-04:
Maybe the user needs to kill the openOCD used for flashing before, since it is probably still open ? Since the openocd is being executed outside the vscode extension the user needs to manually kill it. If the user wants to use the already existing openOCD session, it would need to launch the debug adapter manually too and change the mode to manual in launch.json

JTAG flashing is also possible from the extension and could reduce the issue of launch openocd again.

brianignacio5, 2023-05-15:
Closing this issue for lack of response from the user.

closed 2023-05-15 by brianignacio5