espressif/vscode-esp-idf-extension #963: Problem using monitoring - params in sdkconfig not used

Xav-Scheiber, 2023-05-05:
### OS

Windows

### Operating System version

Windows 10

### Visual Studio Code version

1.78.0

### ESP-IDF version

4.4.2

### Python version

3.9.1

### Doctor command output

---------------------------------------------- ESP-IDF Extension for Visual Studio Code report ---------------------------------------------
OS win32 x64 10.0.19045 
System environment variable IDF_PYTHON_ENV_PATH 
 undefined 
System environment variable PATH 
 C:\Program Files (x86)\VMware\VMware Player\bin;C:\Program Files (x86)\Common Files\Oracle\Java\javapath;C:\windows\system32;C:\windows;C:\windows\System32\Wbem;C:\windows\System32\WindowsPowerShell\v1.0;C:\windows\System32\OpenSSH;C:\Program Files (x86)\Cegid\ManufacturingPMI;C:\Program Files\TortoiseSVN\bin;C:\Raisonance\Ride\bin;C:\Raisonance\Ride\arm-gcc\bin;C:\Program Files (x86)\Microchip\xc8\v1.44\bin;%JAVA__HOME%\bin;C:\Program Files\Java\jdk1.8.0_101\bin;C:\Program Files (x86)\Inno Setup 5;C:\Program Files (x86)\WiX Toolset v3.11\bin;C:\Program Files\nodejs;C:\ProgramData\chocolatey\bin;C:\WINDOWS\system32;C:\WINDOWS;C:\WINDOWS\System32\Wbem;C:\WINDOWS\System32\WindowsPowerShell\v1.0;C:\WINDOWS\System32\OpenSSH;C:\Program Files (x86)\STMicroelectronics\STM32 ST-LINK Utility\ST-LINK Utility;C:<HOMEPATH>\AppData\Local\Packages\PythonSoftwareFoundation.Python.3.9_qbz5n2kfra8p0\LocalCache\local-packages\Python39\Scripts;C:<HOMEPATH>\.espressif\python_env\idf4.4-rc1_py3.8_env\Scripts;C:\Program Files\PuTTY;C:\Program Files\Maven\apache-maven-3.8.5\bin;%IDF_PATH%\tools;%ADF_PATH%\tools;C:\Program Files\TortoiseGit\bin;C:\Program Files\Microsoft SQL Server\150\Tools\Binn\;C:\flutter\flutter\nuget.exe;C:\Program Files\Git\cmd;C:<HOMEPATH>\AppData\Local\Microsoft\WindowsApps;C:<HOMEPATH>\AppData\Roaming\npm;C:<HOMEPATH>\AppData\Local\Programs\Microsoft VS Code\bin;C:<HOMEPATH>\.espressif\python_env\idf4.4-rc1_py3.8_env\Scripts;C:\Program Files (x86)\Actiona;C:<HOMEPATH>\.espressif\esp-idf\tools;C:\flutter\flutter\bin;C:<HOMEPATH>\AppData\Local\Microsoft\WinGet\Links 
System environment variable PYTHON 
 undefined 
Visual Studio Code version 1.78.0 
Visual Studio Code language en 
Visual Studio Code shell C:\WINDOWS\System32\WindowsPowerShell\v1.0\powershell.exe 
ESP-IDF Extension version 1.6.2 
Workspace folder c:\MICRO\GITHUB\amplificateur-connecte-bt-can 
---------------------------------------------------- Extension configuration settings ------------------------------------------------------
ESP-ADF Path (idf.espAdfPath) C:<HOMEPATH>\.espressif\esp-adf
ESP-IDF Path (idf.espIdfPath) C:<HOMEPATH>\Desktop\esp-idf
ESP-MDF Path (idf.espMdfPath) ${env:MDF_PATH}
ESP-Matter Path (idf.espMatterPath) ${env:ESP_MATTER_PATH}
Custom extra paths (idf.customExtraPaths) C:<HOMEPATH>\.espressif\tools\xtensa-esp32-elf\esp-2021r2-patch2-8.4.0\xtensa-esp32-elf\bin;C:<HOMEPATH>\.espressif\tools\xtensa-esp32s2-elf\esp-2021r2-patch2-8.4.0\xtensa-esp32s2-elf\bin;C:<HOMEPATH>\.espressif\tools\xtensa-esp32s3-elf\esp-2021r2-patch2-8.4.0\xtensa-esp32s3-elf\bin;C:<HOMEPATH>\.espressif\tools\riscv32-esp-elf\esp-2021r2-patch2-8.4.0\riscv32-esp-elf\bin;C:<HOMEPATH>\.espressif\tools\esp32ulp-elf\2.28.51-esp-20191205\esp32ulp-elf-binutils\bin;C:<HOMEPATH>\.espressif\tools\esp32s2ulp-elf\2.28.51-esp-20191205\esp32s2ulp-elf-binutils\bin;C:<HOMEPATH>\.espressif\tools\cmake\3.20.3\bin;C:<HOMEPATH>\.espressif\tools\openocd-esp32\v0.11.0-esp32-20211220\openocd-esp32\bin;C:<HOMEPATH>\.espressif\tools\ninja\1.10.2;C:<HOMEPATH>\.espressif\tools\idf-exe\1.0.3;C:<HOMEPATH>\.espressif\tools\ccache\4.3\ccache-4.3-windows-64;C:<HOMEPATH>\.espressif\tools\dfu-util\0.9\dfu-util-0.9-win64
Custom extra vars (idf.customExtraVars)
    OPENOCD_SCRIPTS: C:<HOMEPATH>\.espressif\tools\openocd-esp32\v0.11.0-esp32-20211220/openocd-esp32/share/openocd/scripts
    IDF_CCACHE_ENABLE: 1
Virtual env Python Path (idf.pythonBinPath) C:<HOMEPATH>\.espressif\python_env\idf4.4_py3.8_env\Scripts\python.exe
Serial port (idf.port) COM10
OpenOCD Configs (idf.openOcdConfigs) interface/ftdi/esp32_devkitj_v1.cfg,board/esp32-wrover.cfg
ESP-IDF Tools Path (idf.toolsPath) C:<HOMEPATH>\.espressif
Git Path (idf.gitPath) C:<HOMEPATH>\.espressif\tools\idf-git\2.30.1\cmd\git.exe
-------------------------------------------------------- Configurations access -------------------------------------------------------------
Access to ESP-ADF Path (idf.espAdfPath) true
Access to ESP-IDF Path (idf.espIdfPath) true
Access to ESP-MDF Path (idf.espMdfPath) false
Access to ESP-Matter Path (idf.espMatterPath) false
Access to ESP-IDF Custom extra paths
Access to C:<HOMEPATH>\.espressif\tools\xtensa-esp32-elf\esp-2021r2-patch2-8.4.0\xtensa-esp32-elf\bin: true
Access to C:<HOMEPATH>\.espressif\tools\xtensa-esp32s2-elf\esp-2021r2-patch2-8.4.0\xtensa-esp32s2-elf\bin: true
Access to C:<HOMEPATH>\.espressif\tools\xtensa-esp32s3-elf\esp-2021r2-patch2-8.4.0\xtensa-esp32s3-elf\bin: true
Access to C:<HOMEPATH>\.espressif\tools\riscv32-esp-elf\esp-2021r2-patch2-8.4.0\riscv32-esp-elf\bin: true
Access to C:<HOMEPATH>\.espressif\tools\esp32ulp-elf\2.28.51-esp-20191205\esp32ulp-elf-binutils\bin: true
Access to C:<HOMEPATH>\.espressif\tools\esp32s2ulp-elf\2.28.51-esp-20191205\esp32s2ulp-elf-binutils\bin: true
Access to C:<HOMEPATH>\.espressif\tools\cmake\3.20.3\bin: true
Access to C:<HOMEPATH>\.espressif\tools\openocd-esp32\v0.11.0-esp32-20211220\openocd-esp32\bin: true
Access to C:<HOMEPATH>\.espressif\tools\ninja\1.10.2: true
Access to C:<HOMEPATH>\.espressif\tools\idf-exe\1.0.3: true
Access to C:<HOMEPATH>\.espressif\tools\ccache\4.3\ccache-4.3-windows-64: true
Access to C:<HOMEPATH>\.espressif\tools\dfu-util\0.9\dfu-util-0.9-win64: true
Access to Virtual env Python Path (idf.pythonBinPath) true
Access to CMake in environment PATH undefined
Access to Ninja in environment PATH undefined
[... 122 lines cut ...]
                "limitSymbolsToIncludedHeaders": false
            },
            "configurationProvider": "ms-vscode.makefile-tools"
        }
    ],
    "version": 4
} 
--------------------------------------------------------------------------------------------------------------------------------------------

### Description

When starting monitoring my esp32 board powershell seems not to take into account the parameters present in the sdkconfig file of the project. For exemple I've a COM10 with baudrate set to 115200 but monitoring is start with 460800... This problem appeared on version 1.6.2 and does not exist in version 1.6.1. It seems related to functionality #951
Returning to version 1.6.1 resolve my problem.
This is a log from terminal:
[image]

### Debug Message

```plain
A powershell popup error appears with exit code 1.
```

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

brianignacio5, 2023-05-05:
Have you try using the new `idf.monitorBaudRate` configuration setting to set the monitoring baud rate ?

vsh-ektos, 2023-05-05:
Add a line to setting.json
"idf.monitorBaudRate": "115200"
The update 1.6.2 set the wrong default UART BoudRate

vsh-ektos, 2023-05-05:
@brianignacio5 Maybe it's better to change it to 115200?
https://github.com/espressif/vscode-esp-idf-extension/blob/f00a29da7c6dd3caaecc70f01f5f438dc62c2fd0/package.json#L467

Xav-Scheiber, 2023-05-05:
Ok it's solved by adding the line "idf.monitorBaudRate":"115200", thanks.
But it's very annoying to come across this kind of bug.
Now what's the use of the line CONFIG_ESPTOOLPY_MONITOR_BAUD_115200B=y in the sdkconfig file ?

brianignacio5, 2023-05-05:
From V5.0 is already not being used by idf.py monitor as described [here](https://docs.espressif.com/projects/esp-idf/en/latest/esp32/migration-guides/release-5.x/5.0/tools.html#idf-monitor)

Xav-Scheiber, 2023-05-05:
Ok, so if I understand correctly we accept a bug for those who use old version like 4.4. I can't upgrade to version 5 because I'm using esp-adf framework which is not supported with version 5 (see here in the readme: https://github.com/espressif/esp-adf ).

roblatour, 2023-05-05:
I have the same problem, it took me 40 minutes this morning to realize it was because of this update :-(

brianignacio5, 2023-05-06:
You're right @Xav-Scheiber I'll add esp-idf version validation to fix this issue

BlackGhostHD, 2023-05-07:
Quality assurance is our passion... :angry:

crackwitz, 2023-05-11:
These changes were introduced by merged pull request #951

Xav-Scheiber, 2023-05-12:
> 

Yes I agree. Errors happen to everyone! And luckily we have a temporary solution! Let's get things :)

PR #964 (2023-05-25): use monitor baud rate setting on version 5 or up
## Description

The `idf.monitorBaudRate` setting should be used for ESP-IDF version 5.0 or newer.

Fixes #963

## Type of change

- Bug fix (non-breaking change which fixes an issue)

## Steps to test this pull request

Provide a list of steps to test changes in this PR and required output

1. Click on "ESP-IDF Monitor your device" for ESP-IDF v.4.4
2. Check monitor baud rate being used is the same from sdkconfig value.
3. Click on "ESP-IDF Monitor your device" for ESP-IDF 5.0
4. Check monitor baud rate being used is the same from `idf.monitorB [... cut]
closed 2023-05-25 by brianignacio5