espressif/vscode-esp-idf-extension #966: DFU build failure is completely silent in the IDE

reinderien, 2023-05-07:
### OS

Windows

### Operating System version

Windows 11

### Visual Studio Code version

1.78.0

### ESP-IDF version

5.0.1

### Python version

3.8.7

### Doctor command output

The doctor command runs and then silently ends without telling me where the report went. Turns out it's on the clipboard... maybe you should tell someone about that.

```none
---------------------------------------------- ESP-IDF Extension for Visual Studio Code report ---------------------------------------------
OS win32 x64 10.0.22621 
System environment variable IDF_PYTHON_ENV_PATH 
 undefined 
System environment variable PATH 
 C:\Program Files\Python310\Scripts\;C:\Program Files\Python310\;C:\Program Files\Python38\Scripts\;C:\Program Files\Python38\;C:\Windows\system32;C:\Windows;C:\Windows\System32\Wbem;C:\Windows\System32\WindowsPowerShell\v1.0\;C:\Windows\System32\OpenSSH\;C:\Program Files (x86)\NVIDIA Corporation\PhysX\Common;C:\Program Files\NVIDIA Corporation\NVIDIA NvDLISR;C:\Program Files\Intel\WiFi\bin\;C:\Program Files\Common Files\Intel\WirelessCommon\;C:\WINDOWS\system32;C:\WINDOWS;C:\WINDOWS\System32\Wbem;C:\WINDOWS\System32\WindowsPowerShell\v1.0\;C:\WINDOWS\System32\OpenSSH\;C:\Program Files\dotnet\;C:\Program Files\Pandoc\;C:\Program Files\MiKTeX\miktex\bin\x64\;C:\Program Files\PuTTY\;C:\Program Files\GitHub CLI\;C:\Program Files\Git\cmd;C:\Program Files\Microsoft VS Code\bin;C:<HOMEPATH>\AppData\Local\Microsoft\WindowsApps;C:<HOMEPATH>\.dotnet\tools;C:\ProgramData\Miniconda3\Library\bin\graphviz;C:\Program Files\PostgreSQL\11\bin;C:\Program Files (x86)\WinMerge;C:\Program Files\jdk-17.0.1\bin;;%USERPROFILE%\AppData\Local\Microsoft\WindowsApps 
System environment variable PYTHON 
 undefined 
Visual Studio Code version 1.78.0 
Visual Studio Code language en 
Visual Studio Code shell C:\WINDOWS\System32\WindowsPowerShell\v1.0\powershell.exe 
ESP-IDF Extension version 1.6.2 
Workspace folder c:<HOMEPATH>\src\riptime\esp\firmware 
---------------------------------------------------- Extension configuration settings ------------------------------------------------------
ESP-ADF Path (idf.espAdfPath) ${env:ADF_PATH}
ESP-IDF Path (idf.espIdfPath) /ProgramData/esp-idf
ESP-MDF Path (idf.espMdfPath) ${env:MDF_PATH}
ESP-Matter Path (idf.espMatterPath) ${env:ESP_MATTER_PATH}
Custom extra paths (idf.customExtraPaths) /ProgramData/esp-idf-tools/tools/xtensa-esp-elf-gdb/11.2_20220823/xtensa-esp-elf-gdb/bin;C:/ProgramData/esp-idf-tools/tools/riscv32-esp-elf-gdb/11.2_20220823/riscv32-esp-elf-gdb/bin;C:/ProgramData/esp-idf-tools/tools/xtensa-esp32-elf/esp-2022r1-11.2.0/xtensa-esp32-elf/bin;C:/ProgramData/esp-idf-tools/tools/xtensa-esp32s2-elf/esp-2022r1-11.2.0/xtensa-esp32s2-elf/bin;C:/ProgramData/esp-idf-tools/tools/xtensa-esp32s3-elf/esp-2022r1-11.2.0/xtensa-esp32s3-elf/bin;C:/ProgramData/esp-idf-tools/tools/riscv32-esp-elf/esp-2022r1-11.2.0/riscv32-esp-elf/bin;C:/ProgramData/esp-idf-tools/tools/esp32ulp-elf/2.35_20220830/esp32ulp-elf/bin;C:/ProgramData/esp-idf-tools/tools/cmake/3.24.0/bin;C:/ProgramData/esp-idf-tools/tools/openocd-esp32/v0.11.0-esp32-20221026/openocd-esp32/bin;C:/ProgramData/esp-idf-tools/tools/ninja/1.10.2;C:/ProgramData/esp-idf-tools/tools/idf-exe/1.0.3;C:/ProgramData/esp-idf-tools/tools/ccache/4.6.2/ccache-4.6.2-windows-x86_64;C:/ProgramData/esp-idf-tools/tools/dfu-util/0.9/dfu-util-0.9-win64;C:/ProgramData/esp-idf-tools/tools/esp-rom-elfs/20220823
Custom extra vars (idf.customExtraVars)
    IDF_CCACHE_ENABLE: 1
    OPENOCD_SCRIPTS: /ProgramData/esp-idf-tools/tools/openocd-esp32/v0.11.0-esp32-20221026/openocd-esp32/share/openocd/scripts
    ESP_ROM_ELF_DIR: /ProgramData/esp-idf-tools/tools/esp-rom-elfs/20220823/
Virtual env Python Path (idf.pythonBinPath) /ProgramData/esp-idf-tools/python_env/idf5.0_py3.8_env/Scripts/python.exe
Serial port (idf.port) COM1
OpenOCD Configs (idf.openOcdConfigs) board/esp32c3-builtin.cfg
ESP-IDF Tools Path (idf.toolsPath) /ProgramData/esp-idf-tools
Git Path (idf.gitPath) /ProgramData/esp-idf-tools/tools/idf-git/2.30.1/cmd/git.exe
-------------------------------------------------------- Configurations access -------------------------------------------------------------
Access to ESP-ADF Path (idf.espAdfPath) false
Access to ESP-IDF Path (idf.espIdfPath) true
Access to ESP-MDF Path (idf.espMdfPath) false
Access to ESP-Matter Path (idf.espMatterPath) false
Access to ESP-IDF Custom extra paths
Access to /ProgramData/esp-idf-tools/tools/xtensa-esp-elf-gdb/11.2_20220823/xtensa-esp-elf-gdb/bin: true
Access to C:/ProgramData/esp-idf-tools/tools/riscv32-esp-elf-gdb/11.2_20220823/riscv32-esp-elf-gdb/bin: true
Access to C:/ProgramData/esp-idf-tools/tools/xtensa-esp32-elf/esp-2022r1-11.2.0/xtensa-esp32-elf/bin: true
Access to C:/ProgramData/esp-idf-tools/tools/xtensa-esp32s2-elf/esp-2022r1-11.2.0/xtensa-esp32s2-elf/bin: true
Access to C:/ProgramData/esp-idf-tools/tools/xtensa-esp32s3-elf/esp-2022r1-11.2.0/xtensa-esp32s3-elf/bin: true
Access to C:/ProgramData/esp-idf-tools/tools/riscv32-esp-elf/esp-2022r1-11.2.0/riscv32-esp-elf/bin: true
Access to C:/ProgramData/esp-idf-tools/tools/esp32ulp-elf/2.35_20220830/esp32ulp-elf/bin: true
Access to C:/ProgramData/esp-idf-tools/tools/cmake/3.24.0/bin: true
Access to C:/ProgramData/esp-idf-tools/tools/openocd-esp32/v0.11.0-esp32-20221026/openocd-esp32/bin: true
Access to C:/ProgramData/esp-idf-tools/tools/ninja/1.10.2: true
Access to C:/ProgramData/esp-idf-tools/tools/idf-exe/1.0.3: true
[... 107 lines cut ...]
{"message":"ESP Insights Project commit: 203a87f\r\n","stack":"Error: ESP Insights Project commit: 203a87f\r\n\n\tat Ot.printError (\\dist\\extension.js:2:1220603)\n\tat Socket.<anonymous> (\\dist\\extension.js:2:1220079)\n\tat Socket.emit (node:events:513:28)\n\tat addChunk (node:internal/streams/readable:315:12)\n\tat readableAddChunk (node:internal/streams/readable:289:9)\n\tat Readable.push (node:internal/streams/readable:228:10)\n\tat Pipe.onStreamRead (node:internal/stream_base_commons:190:23)","level":"error","timestamp":"2023-05-06T22:04:33.220Z"}
{"message":"ESP RainMaker Project commit: cc861bf\r\n","stack":"Error: ESP RainMaker Project commit: cc861bf\r\n\n\tat Ot.printError (\\dist\\extension.js:2:1220603)\n\tat Socket.<anonymous> (\\dist\\extension.js:2:1220079)\n\tat Socket.emit (node:events:513:28)\n\tat addChunk (node:internal/streams/readable:315:12)\n\tat readableAddChunk (node:internal/streams/readable:289:9)\n\tat Readable.push (node:internal/streams/readable:228:10)\n\tat Pipe.onStreamRead (node:internal/stream_base_commons:190:23)","level":"error","timestamp":"2023-05-06T22:04:33.547Z"}
{"user":true,"level":"warn","message":"The selected device target \"esp32c3\" is not compatible for DFU, as a result the DFU.bin was not created.","timestamp":"2023-05-06T23:07:57.813Z"}
{"user":true,"message":"Wait for ESP-IDF build or flash to finish","stack":"Error: One_Task_At_A_Time\n\tat l (\\dist\\extension.js:2:1398230)\n\tat l.next (<anonymous>)\n\tat \\dist\\extension.js:2:1398529\n\tat new Promise (<anonymous>)\n\tat No (\\dist\\extension.js:2:1398265)\n\tat \\dist\\extension.js:2:1526731\n\tat Generator.next (<anonymous>)\n\tat \\dist\\extension.js:2:1459406\n\tat new Promise (<anonymous>)\n\tat $s (\\dist\\extension.js:2:1459151)\n\tat \\dist\\extension.js:2:1526659\n\tat D.d (\\out\\vs\\workbench\\api\\node\\extensionHostProcess.js:98:45036)\n\tat D.withProgress (\\out\\vs\\workbench\\api\\node\\extensionHostProcess.js:98:44879)\n\tat Object.withProgress (\\out\\vs\\workbench\\api\\node\\extensionHostProcess.js:116:34519)\n\tat \\dist\\extension.js:2:1526555\n\tat Generator.next (<anonymous>)\n\tat \\dist\\extension.js:2:1459406\n\tat new Promise (<anonymous>)\n\tat $s (\\dist\\extension.js:2:1459151)\n\tat \\dist\\extension.js:2:1526503\n\tat Ie.perform (\\dist\\extension.js:2:1190841)\n\tat ba (\\dist\\extension.js:2:1526485)\n\tat \\dist\\extension.js:2:1460354\n\tat e.ExtHostCommands.h (\\out\\vs\\workbench\\api\\node\\extensionHostProcess.js:98:125315)\n\tat e.ExtHostCommands.$executeContributedCommand (\\out\\vs\\workbench\\api\\node\\extensionHostProcess.js:98:126218)\n\tat a.N (\\out\\vs\\workbench\\api\\node\\extensionHostProcess.js:110:11620)\n\tat a.M (\\out\\vs\\workbench\\api\\node\\extensionHostProcess.js:110:11338)\n\tat a.H (\\out\\vs\\workbench\\api\\node\\extensionHostProcess.js:110:10431)\n\tat a.G (\\out\\vs\\workbench\\api\\node\\extensionHostProcess.js:110:9412)\n\tat \\out\\vs\\workbench\\api\\node\\extensionHostProcess.js:110:8 [... cut ...]
The log explains most of it, but I believe there to be a hardware problem with my host board that prevents USB from showing up. The problem in the IDE is that it silently fails and doesn't tell me a single thing about this condition, and I need to hunt through the logs to see what's going on.

### Debug Message

```plain
N/A
```

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

reinderien, 2023-05-07:
Typo; that should read

`%USERPROFILE%\.vscode\`.

reinderien, 2023-05-07:
In the extension host log I'm also getting

```none
2023-05-06 20:28:30.469 [error] TypeError: Cannot read properties of undefined (reading 'start')
	at c:\Users\gtoom\.vscode\extensions\espressif.esp-idf-extension-1.6.2\dist\extension.js:2:1534619
	at Generator.next (<anonymous>)
	at s (c:\Users\gtoom\.vscode\extensions\espressif.esp-idf-extension-1.6.2\dist\extension.js:2:1459208)
```

radurentea, 2023-05-09:
hi @reinderien,

Thank you for the report. I will work on a fix.

I see that you are using a C3 board, but DFU is being supported only for the S2 and S3 boards.

reinderien, 2023-05-09:
@radurentea I appreciate that. And yes, selecting DFU was my mistake - the board is working quite well with JTAG now. The bug is about the errors above and the IDE's ability to report them in the UI. Thank you and let me know if you need any more information.

radurentea, 2023-05-09:
The issue is that the error is not being shown in the Output channel.
As a work around for now, I suggest you to activate the notification messages (they are hidden by default) by doing the following steps:
1) Open your project folder
2) Go into .vscode
3) Open settings.json
4) Add `"idf.notificationSilentMode": false, `

closed 2023-05-23 by radurentea